Yaskawa-Global/motoros2 #11: Certain group cfgs make JointState output stop working

ted-miller, 2023-05-30:
On an `R1+R2+S1` system, the `/joint_states` topic does not publish. 

I did check the size of the joint_states message with `R1+R2+S1`. It was somewhere around 1100 bytes. So I don't think the size of the message is a problem.

Tested `R1 S1 S2 S3` for other reasons. It worked ok. So it's not the number of groups. But it's still possible it's the number of axes.

I have not checked yet to verify if the publish API is returning an error code.

ted-miller, 2023-05-30:
Additionally, this issue occurs on Foxy, Galactic, and Humble. The ROS2 version is not the reason for this issue.

ted-miller, 2023-09-29:
I restored an old CMOS.bin for other reasons.

When trying to run my FJT script, it says that I didn't have velocity data for all axes. Well... scrolling up through the log, I see that all of the joint names are blank. 

So now I'm thinking that the JointStates publisher is broken because of the lack of joint-names.
[CMOS.zip](https://github.com/Yaskawa-Global/motoros2/files/12768240/CMOS.zip)

I will continue debug next week.

gavanderhoorn, 2023-09-29:
> When trying to run my FJT script, it says that I didn't have velocity data for all axes. Well... scrolling up through the log, I see that all of the joint names are blank.

if you don't configure custom ones, you should get the default ones generated:

https://github.com/Yaskawa-Global/motoros2/blob/6c5614b64f27adc40bff1958b63e1467baeffeae/src/ControllerStatusIO.c#L77-L102

ted-miller, 2023-09-29:
Yeah... I just now saw that too. 
So now I don't know what's up and my system is extra broken.

gavanderhoorn, 2023-09-29:
Would you have a sample of the `/joint_states` topic?

ted-miller, 2023-09-29:
/joint_states won't publish. `topic echo` just freezes until I `ctrl-c`. 

However, I am able to view the topics for each individual control-group, such as `/ctrl_groups/r1/joint_states`. That works fine.

gavanderhoorn, 2023-09-29:
Hm, wireshark? Any XRCE-DDS traffic?

ted-miller, 2024-12-02:
@jimmy-mcelwain I just had another thought. Rather than a max number of bytes, the uros allocator may be limiting the size of the sequence (array of joint names). 

In the FJT action, we're explicitly defining some allocation rules to that regard.

https://github.com/Yaskawa-Global/motoros2/blob/0075ad831c6463a7eda01ff59d801ecefb3bec30/src/ActionServer_FJT.c#L83-L84

I don't think that we explicitly define these elsewhere. (Though I'm not certain. I'm just using the web IDE right now.) Perhaps it's defaulting to a max of `10` and we need to override that in the initialization function for that topic/typesupport.

ted-miller, 2024-12-02:
> Then it successfully publishes all 13 joints because the packet size is reduced.

>  the uros allocator may be limiting the size of the sequence (array of joint names)

Or it could be limiting the size of the entire message in that same regard. It might not be the sequence size.

jimmy-mcelwain, 2024-12-02:
I've made some progress. I realized that `/tf` is able to properly publish a message above ~512 bytes by splitting it into 2 messages, but `/joint_states` isn't not publishing anything at all if the message is above 512. I thought that it might be a QoS profile difference since `/tf` uses `default` QoS and `/joint_states` uses `sensor` QoS. So I changed the QoS for `/joint_states` to default, and now `/joint_states` is being properly published with all 13 joints, even though the message is larger than 512 bytes. It seems like the serialization/deserialization for sensor messages larger than the `UCLIENT_UDP_TRANSPORT_MTU` isn't working here. I don't know if this is intended or not. I'll look into it more tomorrow. 

Changing [this line](https://github.com/Yaskawa-Global/micro_ros_motoplus_buildscripts/blob/913df72bff87b154ece462aff567c56ad82c0d80/client-colcon.meta#L49) to a higher number may temporarily solve the problem, but I understand the downsides to that, and that it may not be a true solution.

gavanderhoorn, 2024-12-03:
@ted-miller wrote:
> Rather than a max number of bytes, the uros allocator may be limiting the size of the sequence (array of joint names).

The code initialising the 'global joint state publisher' initialises the message:

https://github.com/Yaskawa-Global/motoros2/blob/0075ad831c6463a7eda01ff59d801ecefb3bec30/src/PositionMonitor.c#L66

it does configure the correct capacity for the names and the other arrays in the `JointState` message:

https://github.com/Yaskawa-Global/motoros2/blob/0075ad831c6463a7eda01ff59d801ecefb3bec30/src/PositionMonitor.c#L101-L104

We don't change the default 'max' string length, but we use `rosidl_runtime_c__String__assign(..)` ([here](https://github.com/Yaskawa-Global/motoros2/blob/0075ad831c6463a7eda01ff59d801ecefb3bec30/src/PositionMonitor.c#L79) fi) to store the names, which will `reallocate(..)` the `char*` if needed ([here](https://github.com/ros2/rosidl/blob/41be8d6ed987e589493ffaf24d5e2a74e2554a42/rosidl_runtime_c/src/string_functions.c#L109-L110)).

Given:

@jimmy-mcelwain wrote:
> I realized that `/tf` is able to properly publish a message above ~512 bytes by splitting it into 2 packets, but `/joint_states` isn't not publishing anything at all if the message is above 512. I thought that it might be a QoS profile difference since `/tf` uses `default` QoS and `/joint_states` uses `sensor` QoS. So I changed the QoS for `/joint_states` to `default`, and now `/joint_states` is being properly published with all 13 joints, even though the message is larger than 512 bytes. It seems like the serialization/deserialization for sensor messages larger than the `UCLIENT_UDP_TRANSPORT_MTU` isn't working here.

I would indeed also start to suspect either we configure something incorrectly, or there is (still) some issue with fragmentation/reassembly of larger/largish messages at the RMW level.

We've ran into serialisation issues related to fragmentation earlier in the development of MotoROS2. I'll try to look up the specific issue where we diagnosed it. In the end, the bug was in RMW Micro XRCE-DDS IIRC and was fixed there.

> Changing [this line](https://github.com/Yaskawa-Global/micro_ros_motoplus_buildscripts/blob/913df72bff87b154ece462aff567c56ad82c0d80/client-colcon.meta#L49) to a higher number may temporarily solve the problem, but I understand the downsides to that, and that it may not be a true solution.

it would indeed be a work-around, as increasing that constant would make us run into the same problem later, with either even more joints *or* longer joint names (fi).

There are definitely potential issues with a 'low' MTU and best effort QoS, but I would not expect those to already start affecting us here. Typical problems would be loss of fragments which then won't be retransmitted (because: best effort) causing loss of messages, but the messages we're discussing here are a couple hundred bytes, and at most 2 to 3 fragments (if I understand @jimmy-mcelwain correctly). Especially on a network segment that doesn't see too much contention -- and I would expect @jimmy-mcelwain to use such a setup -- that should not cause us to lose *all* messages.

gavanderhoorn, 2024-12-03:
@jimmy-mcelwain: just to make sure: which version of `eProsima/Micro-XRCE-DDS-Client` are you testing this against? [This](https://github.com/Yaskawa-Global/micro_ros_motoplus/blob/8e19dcf71c07bca6c0a6f9408a2b5ac453b27584/target_ws_pkgs.repos#L19-L26) one, or [this](https://github.com/Yaskawa-Global/micro_ros_motoplus/blob/1d9c582f49e7e436136cac37371273ed9d9418d2/target_ws_pkgs.repos#L19-L26) one?

jimmy-mcelwain, 2024-12-03:
I am using the former (`2ae3b964aef3bf1609d39dad1e188dbcadfeeeab`) with humble right now. I believe that I also tested on the latter on jazzy, and `/joint_states` wasn't publishing there either. Probably for the same reason, but I didn't verify that.

gavanderhoorn, 2024-12-03:
I just checked and both versions actually have the patch(es) I had in mind. So it wouldn't matter.

I might suggest we open an issue on `eProsima/Micro-XRCE-DDS-Client` again to start a discussion about this. The maintainers will probably ask for a MWE. We should be able to create one using just RCL and RCLC on a regular desktop (and setting `RMW_IMPLEMENTATION` to `rmw_microxrcedds`).

jimmy-mcelwain, 2024-12-04:
I've opened https://github.com/eProsima/Micro-XRCE-DDS-Client/issues/394

gavanderhoorn, 2024-12-05:
Summarising: this is expected / by design.

Configuring reliability QoS on a (micro-)ROS publisher influences two aspects of the communication 'chain':

1. QoS of the `Client<->Agent` channel
2. QoS of `Agent<->DDS` dataspace exchanges

If both are set to best effort, only the DDS part would support fragmentation / reassembly of messages.

XRCE-DDS does not support this over a best effort channel. So anything `> MTU` will not make it out to Agent. Whether `Agent<->DDS` then is configured for best effort or not doesn't matter anymore.

ted-miller, 2024-12-05:
So is the solution just to take the easy-route and increase the MTU?

Specifically for `joint_states`, I think our worst-case is 14336 bytes.
(`max groups` * `max axes`) + (`max joint name` + (`3 arrays` * `max axes` * `sizeof double`))

`/tf` could certainly be larger. But we default that to be reliable.

And frankly, the reality is that we're never going to see an eight-robot system. The support for >4 groups is for the sake of external/base axes.

gavanderhoorn, 2024-12-12:
> So is the solution just to take the easy-route and increase the MTU?

That would indeed seem to be the only way to make best effort QoS work with large messages, given the limitations currently present in micro-ROS / XRCE-DDS.

> Specifically for `joint_states`, I think our worst-case is 14336 bytes. (`max groups` * `max axes`) + (`max joint name` + (`3 arrays` * `max axes` * `sizeof double`))
>
> `/tf` could certainly be [...]

It seems it could / would be OK to document the limitation in micro-ROS / XRCE-DDS and explain that systems with a certain number of axes can't use best effort QoS profiles right now for these topics.

We could preallocate for the absolute worst case, but that seems rather wasteful.

ted-miller, 2024-12-12:
How about a compromise of increasing the MTU to accommodate at least 3 groups and document the limitation after that?

That would cover two robots and a positioner, or a robot with a positioner and a base track. 

I personally haven't seen any ROS applications more complex than that.

ted-miller, 2024-12-13:
I'd like to squeeze the MTU adjustment into 0.2.0.

I'm back on an R1+R2 system and it annoys me a lot that joint_states doesn't work.

closed 2025-03-31 by gavanderhoorn