Yaskawa-Global/motoros2 #8: Interoperability issues with different DDS implementations

ted-miller, 2023-05-30:
In Galactic, the default RMW has [changed from Fast DDS to Cyclone DDS](https://docs.ros.org/en/galactic/Concepts/About-Different-Middleware-Vendors.html?highlight=default).

During testing, communication failed. Once I [stopped the daemon](https://docs.ros.org/en/galactic/How-To-Guides/Working-with-multiple-RMW-implementations.html) and ran export `RMW_IMPLEMENTATION=rmw_fastrtps_cpp` everything worked fine.

After having discussed this over at https://github.com/ros2/rmw_cyclonedds/issues/184 it appears this is a much wider problem. It's not just Fast-DDS vs Cyclone, but it's very likely all currently available DDS-based RMWs are incompatible.

Labeling as a bug.

It's not a bug here, or something we can control, but it's something users don't expect, causes MotoROS2 to not work with anything other than Fast-DDS and we will get support requests related to this.

gavanderhoorn, 2023-05-31:
Added `micro-ROS` label as although it's not strictly micro-ROS itself, `rmw_microxrcedds` is partly *why* we run into this.

gavanderhoorn, 2023-07-06:
@destogl: the issue we discussed in Bordeaux.