Yaskawa-Global/motoros2 #9: Upper limit to publishing frequency

ted-miller, 2023-05-30:
The frequency for publishing topics is configurable in the `motoros2_config.yaml` configuration file. Initial testing has revealed that the MotoROS2 is limited to about 100 Hz.

`0.1.0` added features to attempt to improve the publish rates. While it did in fact improve things, there are still limits imposed by the QoS settings. Topics with a `reliable` quality seem to take ~10 ms to publish. Can this be improved upon?

gavanderhoorn, 2023-06-05:
As we've seen some similar behaviour in the past (`reliable` topics seemingly 'blocking' `best_effort` topics): [this discussion](https://discourse.ros.org/t/ros2-qos-reliability-issue/31700) on ROS Discourse goes into some detail as to what might be a cause.

Specifically, [this comment](https://discourse.ros.org/t/ros2-qos-reliability-issue/31700/23) points to some things we might want to check.

gavanderhoorn, 2024-07-01:
Just noticed: if MotoROS2 is started normally (ie: binary installed on the controller, cold start of controller), publishing frequency seems limited quite severely.

If I start it manually, I get a max of about 160 Hz.

Not sure yet what the difference is.