Yaskawa-Global/motoros2 #17: `JointState::velocity` values fluctuate, even when servos are off

gavanderhoorn, 2023-05-31:
Even when servos are off, values reported in `JointState::velocity` can fluctuate (as in: 'randomly' change).

This could be due to noise in the values reported by M+ APIs, or a conversion error on our end.

The fluctuations are relatively minor (ie: small fractional changes). In general the reported values appear to be OK.

ted-miller, 2023-05-31:
To elaborate on this issue, it is not due to a conversion error in motoros2. I have witnessed this behavior on the teach pendant. 

I'm not sure if it is due to the m-register-speed-feedback method that we're using, or something else.

If we see it happen again, we should look at the "Servo Monitor" screen on the pendant (not the M registers) to see what is being reported there.

BrunoB81HK, 2024-03-11:
Hi,

I think I encountered a similar bug, but I'm not sure it is exactly the same.

For some reason, the velocities published on the `/joint_state` topic flickers, but a lot more than fractions. I have multiple joint velocities that are more than 600 rad/sec for a timeframe.

Currently, it is not causing any issues, but it could in the future.

Here is a screenshot of one of my log files. It represent the yaml format of a `sensor_msgs/JointState`: 

[image]

gavanderhoorn, 2024-03-11:
Thanks for reporting.

Have you seen the values on the teach pendant as well? @ted-miller mentions something like that in his comment.

gavanderhoorn, 2024-03-11:
@ted-miller: could this be due to modification concurrent with reading the M-registers? Seeing as we are concatenating two M-registers to get the full value, if they get modified while a read is in progress, strange things could happen.

ted-miller, 2024-03-11:
I'm doubtful that it's a race condition with the mpReadIO function, but can't say with certainty.

I'd be interested to know if the values are shown on the teach pendant.

My first guess is that something else is writing to those same registers. Perhaps an external PLC or something in CIOPRG.LST.

ted-miller, 2024-03-11:
Separate email from @BrunoB81HK:

> We are currently connecting the robot controller to a PLC...

BrunoB81HK, 2024-03-12:
> could you clarify what "for a timeframe" means exactly?

By a timeframe, I mean the time between two messages published on the ´/joint_state´ topic. So if the frequency is 30 hz, the timeframe is ~33 ms.

BrunoB81HK, 2024-03-12:
> Separate email from @BrunoB81HK:
> 
> 
> 
> > We are currently connecting the robot controller to a PLC...

Great catch, but this happened before we started connecting to a PLC. 

I can try and get some data to help in the debugging. Let me know what you need and I'll do my best to provide it to you tomorrow.

BrunoB81HK, 2024-03-23:
@gavanderhoorn @ted-miller Let me know if I can provide anything to help you find the problem.

ted-miller, 2024-03-25:
@BrunoB81HK could you take a look at the teach pendant to see if the values are fluctuating on there? There are two places to look.

If you are using the Smart Pendant, please switch to the Classic Interface for this.

-----------------------

1) Navigate to `IN / OUT` > `Register`. Check the values of the following M-Registers to see if they are matching what is displaying on the `joint_states`.

[image]

-----------------------

2) Navigate to `ROBOT` > `SERVO MONITOR`. In the top `DISPLAY` menu, select `MONITOR ITEM 1`. Then select `SPEED INST`.

[image]

[image]

-----------------------

Lastly, could you please save a copy of the CIOPRG.LST (ladder program) from your pendant and post it here? I want to verify there aren't any other instructions writing to those registers.

BrunoB81HK, 2024-03-29:
Hi @ted-miller,

I finally found some time to look into it. The Servo monitor was not fluctuating, but the registers were. Register `M050` to `M057` are fluctuating between 0 (`0000_0000_0000_0000`) and 6000 (`0001_0111_0111_0000`).

[image]

---

According to your image, those registers are linked to the speed of `R1`'s `Axis 1` to `Axis 4`. This makes sense since those are the only links that fluctuate. Here is the `velocity` component of a received `JointState` message : 

```py
[ 686.3018489302283, 686.3018512946209, 686.3018530768039, 686.3018490205695, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0 ]
```
---

I looked up the user ladder program and found what looks like timers. Based on the line (`TMR M050,6000`), this is what is causing problem here. Let me know if you still need the `CIOPRG.LST` file.

[image]

---

The thing is, I don't think we modified the ladder for this controller, but I will talk to my colleagues to make sure. Do you know of any option that could need those timers?

Thanks a lot for your help!

gavanderhoorn, 2024-03-29:
Nice find @BrunoB81HK.

That would indeed seem to explain how those `600`-ish values end up in the `velocity` field.

ted-miller, 2024-04-01:
@BrunoB81HK 
The simple fix would be to modify the parameters listed in section 3.12 of https://wiki.ros.org/motoman_driver/Troubleshooting#Not_receiving_axis_speeds_from_robot_controller
This controls which registers get populated with the speed data. MotoROS2 will automatically detect this configuration and adjust accordingly. 

Just change these parameters to use any other register numbers that aren't being used in the CIO ladder program.

BrunoB81HK, 2024-04-08:
This solved the issue! Thanks a lot!

ted-miller, 2025-04-14:
We encountered this issue again. It turns out that our standard Simple Connect ladder is using these registers for a series of timers.

Simple Connect is a very common option in the US. But it is not compatible with MotoROS2.

The solution is to either init the ladder, or change the parameters that control which registers are used. (https://github.com/Yaskawa-Global/motoros2/issues/17#issuecomment-2029720028)

closed 2025-04-14 by ted-miller

gavanderhoorn, 2025-04-16:
> It turns out that our standard Simple Connect ladder is using these registers for a series of timers.

isn't that what @BrunoB81HK also posted in https://github.com/Yaskawa-Global/motoros2/issues/17#issuecomment-2027357536? Is it mostly the discovery that this is Simple Connect doing this?

> Simple Connect is a very common option in the US. But it is not compatible with MotoROS2.

is there a way we could detect Simple Connect is installed, and raise an alarm in that case? Failing 'loudly' is often a good thing, as it provides a clear indication of why something isn't working.

ted-miller, 2025-04-17:
> Is it mostly the discovery that this is Simple Connect doing this?

Yes

> is there a way we could detect Simple Connect is installed

There is an input that comes on if the SC MotoPlus application is installed. But that bit is only reserved by YAI. It could be used for other things in different regions.

But in this particular case, it wouldn't have helped anyway. We didn't have the MotoPlus app installed. But we did have the ladder modifications from a previous test.