Yaskawa-Global/motoros2 #18: Add support for DX200

gavanderhoorn, 2023-05-31:
We'll expand this issue to track status of sub issues where we work on adding support for DX200 controllers to MotoROS2.

ted-miller, 2023-06-01:
Known issues at this time:

- Need to read robot calibration data (obfuscated in PlatformList)
- The command to create the node is returning an error code `RCL_RET_NODE_INVALID_NAME`. (I haven't done any debug on it yet.)
- Setup build system for libmicroros
- Update ParameterExtraction
- Update PlatformList
- Setup VS project configurations and related support files (mps)

gavanderhoorn, 2023-06-02:
@smith-doug @robinsonmm

gavanderhoorn, 2023-06-08:
@smith-doug @robinsonmm: would the robot you could potentially use to test happen to be mounted on a rail/track? If it isn't, we might be able to provide you with a test-build sooner.

acbuynak, 2023-06-08:
If helpful as a test case, I can offer a 12-axis (2 x 6R arms) system controlled by two DX200's in a master-slave configuration.

ted-miller, 2023-06-09:
hmm... I looks like `Ros_CtrlGroup_GetEncoderTemperature` is not actually used. I suspect this was copied from MotoROS1 in anticipation of future functionality. We should just remove it for now.

gavanderhoorn, 2023-06-10:
I'm pretty sure `DN3.19.00a` does have `mpGetEncoderTemp(..)`, and that's the version I verified symbols against.

> hmm... I looks like `Ros_CtrlGroup_GetEncoderTemperature` is not actually used. I suspect this was copied from MotoROS1 in anticipation of future functionality. We should just remove it for now.

We (I) added that about a year ago. Preparation for publishing controller statistics.

robinsonmm, 2023-06-15:
@gavanderhoorn our DX200 is NOT on a track.

gavanderhoorn, 2023-06-16:
@robinsonmm: then it would be great if you could perhaps also test the `alpha1` binary linked in https://github.com/Yaskawa-Global/motoros2/pull/49#issuecomment-1584968247.

It's expected to *not* work, but the test would help figure out at which point it breaks.

ted-miller, 2023-06-26:
wrt https://github.com/Yaskawa-Global/motoros2/pull/49#issuecomment-1587956034

I did some more "debugging" and got some weird results. 

My initial assumption was that there is an problem with the semaphore routines in `microxrcedds_client`. My thought was that the `UXR_LOCK_SESSION` was failing in `uxr_run_session_until_all_status`. So I added some debug broadcasts if those fail. Those broadcasts never happened, so I have to assume that the sems are ok. But at this point, `rclc_support_init_with_options` is still failing at startup.

So just to get a look at what is failing, I put a broadcast [here](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/a2937bc5af606b2434e9a841adbf5c003fc1ec38/src/c/core/session/session.c#L465), where the status-code is checked. When running this build, `rclc_support_init_with_options` is no longer failing. MotoROS2 initialization completes successfully. (I didn't change anything other than the Ros_Debug_Broadcast.)

However, when I attempt to use any of the CLI utilities, they just freeze up and hang. But I'm not sure whether that's due to failed communication with the robot or because I'm using a Windows Agent and CLI. Frankly, I've seen some oddities with my Windows setup and don't fully trust it. 

When I get my Ubuntu machine back, I'll try again to see if it works any better. 

But I'm struggling to understand how adding the debug broadcast changed the behavior. I guess it might affect timing a _little_... but it shouldn't be significant.

ted-miller, 2023-06-26:
Oh, also `rcutils_get_error_string` doesn't have any information. It just reports `error not set`.

gavanderhoorn, 2023-06-27:
> So just to get a look at what is failing, I put a broadcast [here](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/a2937bc5af606b2434e9a841adbf5c003fc1ec38/src/c/core/session/session.c#L465), [...]

hm. Maybe some optimisation the compiler now can no longer apply?

Which M+ `libmicroros` version are you using? The one I shared in https://github.com/Yaskawa-Global/motoros2/issues/18#issuecomment-1573451897, or something else?

ted-miller, 2023-06-27:
> hm. Maybe some optimisation the compiler can now no longer apply?

Possible. There was _once_ a long time ago that compiler optimizations broke my application. It took me a LONG time to debug that and that's why I always use `-O0` for everything.

> Which M+ libmicroros version are you using? 

I had to build from source to insert my debug broadcasts.

> When I get my Ubuntu machine back, I'll try again to see if it works any better.

Yep... my Ubuntu machine is able to list/echo topics just fine. So now I _really_ don't trust my Windows build and I have no idea why the DX200 build is actually working.

smith-doug, 2023-06-29:
Is it possible to get a Foxy version for the DX200?  I don't have the DX200 SDK anymore.  Our robot is currently set up for a project using Foxy.

ted-miller, 2023-06-29:
> > When I get my Ubuntu machine back, I'll try again to see if it works any better.
> 
> Yep... my Ubuntu machine is able to list/echo topics just fine. So now I really don't trust my Windows build and I have no idea why the DX200 build is actually working.

I just did a foxy build (unmodified) and it worked immediately with my Ubuntu agent. I'm wondering if the humble errors were entirely due to my windows agent. I'll need to test a vanilla humble build again tomorrow. (and/or redo my windows setup)

gavanderhoorn, 2023-07-10:
#61 introduced use of `gettimeofday(..)` which does not appear to be (directly) supported on DX200.

I've updated the OP.