Yaskawa-Global/motoros2 #19: Add support for FS100

gavanderhoorn, 2023-05-31:
We'll expand this issue to track status of sub issues where we work on adding support for FS100 controllers to MotoROS2.

gavanderhoorn, 2023-05-31:
Added this to the *Untargeted* milestone, as this is `low-priority` for now.

gavanderhoorn, 2023-08-21:
@destogl FYI also.

gavanderhoorn, 2023-08-25:
Update: seems there are some more endianness issues to resolve: https://github.com/micro-ROS/micro-ROS-Agent/issues/197#issuecomment-1692789200.

gavanderhoorn, 2023-08-28:
Hi @saahu27.

> I want to understand, contribute and make use of fs100 controller using ros2.

great. We're always looking for more community participation.

> Can we please get in touch so that i can explain more about my project and work and discuss how can i contribute.

unless you feel what you're doing is (highly) sensitive, I'd suggest starting a discussion [on the Discussion forum](https://github.com/Yaskawa-Global/motoros2/discussions/categories/general).

At the moment I'm debugging some endianess issues, as you can read in my previous https://github.com/Yaskawa-Global/motoros2/issues/19#issuecomment-1693305876.

I expect basic MotoROS2 to work on FS100 once those get fixed (ie: publications and IO services, no motion yet). But no guarantees.

saahu27, 2023-08-28:
I have learnt a lot on how to use ros from your previous contributions for universal robots  and i feel very happy to have interacted with you.Thank you for your response.  I was not able to express myself clearly earlier. i wanted to request to drop your email so i can get in touch with you. 

Nothing that i do is classified  i wanted some help on how to go about controlling the motoman 100fs using ROS 2 and Move it2.
So, i am currently working on porting motoman ros1 drivers using ros1 to ros2 bridge.  i came across the new ros2 drivers you were contributing. Been trying your work on ROS1 Drivers and just wanted to ask some pointers on how to understand your new drivers, what kind of control can we expect a point to point control/Cartesian? Trajectory control?   with what baud rate so i can decide ros2  controller update frequency  etc which I will open a discussion in the page you mentioned.

gavanderhoorn, 2023-08-29:
> i wanted to request to drop your email so i can get in touch with you.

nothing personal, but I try to avoid discussing ROS topics in private emails if at all possible.

Most of my work is open-source, and I'd like to be able to reuse diagnostic sessions and discussions as much as possible as well.

> I will open a discussion in the page you mentioned.

great, looking forward to it.

gavanderhoorn, 2024-02-07:
I've got quite some parts of MotoROS2 disabled for this FS100 test and it shows: `/joint_states` reaches about 180 Hz. Probably related to what is discussed in #9.

gavanderhoorn, 2024-02-07:
Turns out the FS100 does support SRAM, so I'm going to update these lines:

https://github.com/Yaskawa-Global/motoros2/blob/10d8f00317c5d4ef7960627962337bf72e2ba817/src/ConfigFile.c#L788-L791

and some others to no longer treat the FS100 like the DX200.

gavanderhoorn, 2024-02-08:
Current state: https://github.com/Yaskawa-Global/motoros2/compare/main...gavanderhoorn:motoros2:fs100_support.

Works like the YRC1 version, including updating config file from USB and storing it in SRAM.

Number of patches needed is actually quite limited. About half of them add bits and pieces to the build scripts and project configuration for FS100 support. And the other half `#ifdef`s a bunch of stuff to set FS100 specific values.

Main difference: `FollowJointTrajectoryAction` server is not enabled currently.

I'm also seeing the `JointState` topic publishing at approx 180 Hz, even though `controller_status_monitor_period` is set to `10` (which should result in 100 Hz IIUC).

gavanderhoorn, 2024-02-08:
Test build: [20240208_mr2_fs1_h_9.9.9-FS100-alpha33.zip](https://github.com/Yaskawa-Global/motoros2/files/14209893/20240208_mr2_fs1_h_9.9.9-FS100-alpha33.zip).

NOTE: this is not production ready and for testing purposes only.

ted-miller, 2024-02-08:
> we'd most likely not be able to use the FollowJointTrajectoryAction server due to memory constraints.

Can you elaborate on this? (Perhaps you already discussed in another thread, but I can't remember.) 

I know the M+ docs explicitly state there are limits to static memory. But we've blatantly ignored that for DX2 and YRC1, and everything works. Have you experienced issues on the FS?

gavanderhoorn, 2024-02-08:
I seem to remember to have tried keeping the static buffer at its 'normal' size and seeing loading errors. But that was a long time ago when there were multiple other issues.

I just re-enabled 'everything' and the node still comes up. I can start the point queue mode, PP shows servo power is on (virtual, as I have no real hw) and the service returns success.

Need to try with the buffer at normal capacity and see if that still loads.

Controller does have enough memory free btw (about 16 MB if I interpret the stats correctly), just not in the M+ partition.

ted-miller, 2024-02-08:
I've got a guy who should be able to do some testing the week after next. He's completely new to all this, so it'll take a bit to get up to speed. But I'm hoping to have him start helping with the issues-list. (I know I've said that like 4 or 5 times now. So, who knows if it'll actually happen.)

gavanderhoorn, 2024-03-07:
@ted-miller: did you / your colleague get a chance to test the alpha I posted?

ted-miller, 2024-03-07:
> did you / your colleague get a chance to test the alpha I posted?

No.

> I know I've said that like 4 or 5 times now. So, who knows if it'll actually happen

... yeeeaaahhhh....

ted-miller, 2024-03-12:
@gavanderhoorn 
I've got power on my FS100. 
Can you provide a binary (or the supporting libraries) that has 'everything' re-enabled and also has `DUMMY_SERVO_MODE` defined?

gavanderhoorn, 2024-03-12:
I'll have to build that specifically, as my FS100 here only has a virtual robot 'connected'.

Probably tomorrow.

ted-miller, 2024-03-12:
> my FS100 here only has a virtual robot 'connected'

So does that mean you already have `DUMMY SERVO MODE` defined in the alpha-build above? 
I also don't have a physical robot arm on my FS.

gavanderhoorn, 2024-03-12:
No, there just isn't any manipulator connected. I can't check, but I guess it's configured for virtual hw (rotary).

gavanderhoorn, 2024-03-13:
@ted-miller: [20240313_mr2_fs1_h_9.9.9-FS100-alpha40.zip](https://github.com/Yaskawa-Global/motoros2/files/14587036/20240313_mr2_fs1_h_9.9.9-FS100-alpha40.zip).

I'm not near my FS100 right now, so can't test, but I believe this should at least start and let you interact with it.

Things missing are: FJT server, `start_traj_mode` and `select_motion_tool`.

ted-miller, 2024-03-13:
I'm getting immediate deserialization errors. But I think my agent might be old. I'll update in the morning.

gavanderhoorn, 2024-03-14:
Hm. How old is your Agent? I use `microros/micro-ros-agent:humble` from 7 months ago.

Just tested `alpha40` and `alpha41` (the one with the full 2MB static buffer ..). They both load and I can call services and `ros2 topic echo /joint_states`.

What sort of PC are you using to run the Agent? I'm making some assumptions about endianness, word size, etc which may not be true on your side.

---

`alpha42` loads, starts up but then fails to complete init as heap allocation fails (somewhat expected). Ignore that one.

ted-miller, 2024-03-14:
Turns out my agent was 18 months old. It worked fine after an update.

For the most part, I think you've got it working well. There's a few issues though.

1) As mentioned before, installing the .out file in maintenance mode is _really_ slow. I had `alpha40` loaded from yesterday. The first time that I tried to load `alpha44`, I got this:
[image]
I'm assuming this is a timeout due to the excessive time it takes to install the binary. So binary size might be an issue.

I was able to reboot and try again. It worked the second time.

2) For `joint_states`, I'm seeing this:
```
effort:
- 5.17293035e-316
- 2.124100552e-314
- 7.70200109095887e-310
- 1.69759663277e-313
- 5.172925e-316
- 1.3051418519317825e-278
```

I verified we are zeroing out the array before calling the API. So perhaps the API doesn't like dummy mode.

3) `joint_states` doesn't update when I move the robot. This is possibly due to dummy mode. 
The only FS100 that I have with a physical arm is a pair of deltas (MPP3H). I might be able to use those for testing next week. But this system has another M+ app installed and I don't know what kind of resources it consumes. I'm not sure if I can safely remove this app and get a copy to reload later.

4) `tf` doesn't update when I move the robot either. Same comment as 3.

I didn't have issues with any of the other services.

gavanderhoorn, 2024-03-14:
> Turns out my agent was 18 months old. It worked fine after an update.

:+1: 

> 1\. As mentioned before, installing the .out file in maintenance mode is _really_ slow. [..] I'm assuming this is a timeout due to the excessive time it takes to install the binary. So binary size might be an issue.

The flash on the FS100 is just *really* slow. Loading/running directly from USB (fi) is much, much faster.

(I'm not claiming 'this is fine', just identifying the most likely cause here)

There aren't many options for reducing the binary size I haven't considered I believe, but I do have one or two more ideas. It wouldn't be too pretty though.

> I was able to reboot and try again. It worked the second time.
>
> 2\. For `joint_states`, I'm seeing this:
>
> ```
> effort:
> - 5.17293035e-316
> - 2.124100552e-314
> - 7.70200109095887e-310
> - [...]

O, right. `DUMMY_SERVO_MODE`.

That might not actually be enabled in this build ..

re: `effort` values: those are some really small values. I've seen it as well here, and I believe I've even seen it with the YRC1 build.

I'll check again whether there is something not right with the (de)serialisation. I am doing some not-exactly-as-they-should-be-done things.

> 3\. `joint_states` doesn't update when I move the robot. This is possibly due to dummy mode.

hm. Not sure what that would be.

As I wrote above: this isn't dummy servo mode.

> The only FS100 that I have with a physical arm is a pair of deltas (MPP3H). I might be able to use those for testing next week. But this system has another M+ app installed and I don't know what [...]

Couldn't you put the FS100 you're testing with now in 'virtual hw' mode?

MotoROS2 would not know the difference in that case. Mine if configured like that, and it all just behaves as-if there is an actual manipulator connected.

> 4\. `tf` doesn't update when I move the robot either. Same comment as 3.

Might be the same cause.

Could you perhaps make a wireshark capture?

ted-miller, 2024-03-14:
> As I wrote above: this isn't dummy servo mode
> Couldn't you put the FS100 you're testing with now in 'virtual hw' mode?

I'm not sure we're on the same page.

My current FS100 is in 'virtual' (dummy) mode. I don't have an arm attached.

So, when MR2 attempts to read the feedback position, it will always return 0. I'm saying that I need to test again on a physical arm to see if that's indeed the problem. 

The `DUMMY_SERVO_MODE` flag in MR2 tells it to read command-position instead of feedback-position.

gavanderhoorn, 2024-03-14:
Ah, right. I conflated a couple of things.

Ok: [20240314_mr2_fs1_h_9.9.9-FS100-alpha46.zip](https://github.com/Yaskawa-Global/motoros2/files/14606238/20240314_mr2_fs1_h_9.9.9-FS100-alpha46.zip).

`DUMMY_SERVO_MODE=1` now and I had this still disabled:

https://github.com/Yaskawa-Global/motoros2/blob/5f8c80011c64dc9152a4c72b2eb138760a9d649e/src/MotionControl.c#L1053

I suspect/hope this now makes things 'move' again.

ted-miller, 2024-03-14:
/tf is static:

[image]

[wireshark.zip](https://github.com/Yaskawa-Global/motoros2/files/14606283/wireshark.zip)

ted-miller, 2024-03-14:
Regarding /tf, I'm an idiot... my terminal was small and I wasn't seeing the whole message. I was just looking at flange > tcp0.

tf is working fine.

gavanderhoorn, 2024-03-14:
Nice. Have you tried the FJT action server?

ted-miller, 2024-03-14:
Yes. Both the FJT and Queue seemed to work. I didn't have a visualization of an arm setup. But I saw the correct axes moving on the pendant display.

gavanderhoorn, 2024-03-14:
I'll clean up the branch a little and open a PR tomorrow.

Seems like we have a sufficient level of functionality to consider adding FS1 support to mainline.

gavanderhoorn, 2024-03-15:
I've opened #227 as a first draft.

gavanderhoorn, 2025-03-19:
To give it more visibility: see https://github.com/Yaskawa-Global/motoros2/pull/227#issuecomment-2391626478 for the current state of the FS100 port.

To repeat it here: development of this port has been paused for now.

gavanderhoorn, 2025-11-05:
The current state is captured by the (later) comments on #227.

We've taken FS100 support off the roadmap in #454, which makes further development low priority and entirely on a best-effort basis.

While we don't expect to ever reach feature parity on FS100 (due to platform constraints), the most reported issue with the latest alpha tested is a complete hang when trying to execute trajectories through the `FollowJointTrajectory` action server.

Unfortunately we haven't been able to reproduce this, which makes debugging it very difficult.

If you have experience developing MotoPlus applications, have the SDK and are willing to spend the time, I would still be willing to assist debugging. But please note I don't expect it to be easy.