Yaskawa-Global/motoros2 #28: Improve JTA goal validation to also check for duplicate joint names

gavanderhoorn, 2023-06-01:
As per subject.

The current implementation of the `FollowJointTrajectory` action goal validation does not check whether incoming `JointTrajectory`s have duplicated joint names in the `joint_names` field.

This could allow malformed trajectories to be considered OK to execute, while they should be rejected.

gavanderhoorn, 2023-08-17:
In order to not delay releasing `0.1.1`, I propose we move this to the `0.1.2` milestone.

@ted-miller: thoughts?

ted-miller, 2023-08-17:
Yeah, I'm fine with bumping this out. 

Additionally, this type of feature would add more overhead to the message processing. I've been getting a lot of complaints lately about latency. So we should decide if this is really important. Personally, I don't think it's super needed. I don't envision a motion-planner outputting a joint that contradicts itself.

gavanderhoorn, 2023-09-27:
@ted-miller: would this be something we include in `0.1.2`? I'm thinking we should not delay releasing it too long, as it includes fixes for things like #144 and #55 which would be nice to release.

ted-miller, 2023-09-27:
Fix is in #162

PR #162 (2023-10-02): Check for duplicated joint names in a trajectory
Fixes #28 
Not yet tested due to https://github.com/Yaskawa-Global/micro_ros_motoplus/issues/132