Yaskawa-Global/motoros2 #29: Allow trajectory execution to be delayed (ie: when `trajectory.header.stamp != 0`)

gavanderhoorn, 2023-06-01:
The `FollowJointTrajectory` action server in `ros_control` ([joint_trajectory_controller](http://wiki.ros.org/joint_trajectory_controller)) allows clients to request trajectory execution to start at a specific time in the future instead of immediately upon reception of the goal.

This is done by setting `trajectory.header.stamp` to something other than `0`.

See [wiki/joint_trajectory_controller/UnderstandingTrajectoryReplacement](http://wiki.ros.org/joint_trajectory_controller/UnderstandingTrajectoryReplacement) for a (terse) discussion of this functionality (note: that page is mostly about how the `joint_trajectory_controller` performs trajectory *replacement* or blending, that's not what I link it for here).

As MotoROS2 and the ROS 2 host PC sync their clocks, we could consider supporting this.

The 'dwell' time this results in can be calculated as `goal.trajectory.header.stamp - current_controller_time`.

jimmy-mcelwain, 2025-01-06:
Should a movement be delayed if `time_from_start` has not been reached yet for a point in the middle of the trajectory? Or should this only be done for the start of a trajectory based on the timestamp?

gavanderhoorn, 2025-01-07:
> Should a movement be delayed if `time_from_start` has not been reached yet for a point in the middle of the trajectory? 

that would be reason to abort the goal completely I would say.

Trajectory processing is implemented such that MotoROS2 should follow the trajectory *very* closely, both in time and space, so any deviations above certain thresholds should be treated as errors and be reported as such.

> Or should this only be done for the start of a trajectory based on the timestamp?

the behaviour discussed in the OP would delay execution of an incoming trajectory goal if the stamp is set to the in the future.

'Details' (in quotes, as there isn't *that* much) should be available in the JTC implementation in `ros_control`.

Note: this isn't necessarily high-priority. I also don't know how much use it would see. IIRC, the primary use-case for this in ROS 1 (note: not MotoROS1) was to sync execution of multiple trajectories on remote hosts (or even the same host, but with multiple driver instances fi) with synced clocks.

By setting their starting time a little into the future, transmission and processing delays to and on each individual robot could be compensated/ignored (ie: synchronisation of motion would not be due to sending off the goals 'fast enough' or 'close enough together' (in time), but by virtue of their starting stamps being equal).

This could make synchronous motion possible across multiple MotoROS2 instances (running on multiple controllers) without anything 'special' -- except synced clocks.