Yaskawa-Global/motoros2 #34: Make status IO signals configurable

gavanderhoorn, 2023-06-01:
MotoROS2 currently has a hard-coded set of IOs it uses for signalling internal application status:

https://github.com/Yaskawa-Global/motoros2/blob/08329e5a1646315bffb5effdb40857eaea343523/src/ControllerStatusIO.h#L11-L27

These indices should be made configurable, to allow users with conflicting IO assignments to use MotoROS2 without having to rebuild the application.

gavanderhoorn, 2023-06-01:
While not necessarily required (as these signals are standard and reserved by Yaskawa for these purposes), it might be interesting to also allow configuration of the following signals:

https://github.com/Yaskawa-Global/motoros2/blob/08329e5a1646315bffb5effdb40857eaea343523/src/ControllerStatusIO.c#L268-L298

The same approach used for the `IO_FEEDBACK_*` signals could potentially be used.

ted-miller, 2023-07-28:
> While not necessarily required (as these signals are standard and reserved by Yaskawa for these purposes), it might be interesting to also allow configuration of the following `[IO_ROBOTSTATUS_]` signals

I disagree with this point.  Those signals are hardcoded in each robot controller. Given that they are vital and safety-related indicators, they should not be allowed to be easily change.

-----------------------
Working on this issue would also be a good opportunity to replace the tabs w/ spaces.

[image]

gavanderhoorn, 2023-09-26:
Looks like @SejalBehere ended up doing some work in this direction.

See https://github.com/Yaskawa-Global/motoros2/compare/main...SejalBehere:motoros2:Configurable-IO-Signals.