Yaskawa-Global/motoros2 #35: Possible memory leak (after dis-, then reconnect of Agent)

gavanderhoorn, 2023-06-01:
As mentioned in the *Known issues & limitations* section of the README ([here](https://github.com/Yaskawa-Global/motoros2/tree/08329e5a1646315bffb5effdb40857eaea343523#memory-leak)), MotoROS2 currently appears to suffer from a small memory leak.

Testing shows the following values for used and free memory after several consecutive connect and disconnect cycles with a Galactic micro-ROS Agent:

```
Initialization complete. Memory available: (962992) bytes. Memory in use: (85584) bytes
Initialization complete. Memory available: (962992) bytes. Memory in use: (85584) bytes
Initialization complete. Memory available: (962552) bytes. Memory in use: (86024) bytes
Initialization complete. Memory available: (962280) bytes. Memory in use: (86296) bytes
Initialization complete. Memory available: (962280) bytes. Memory in use: (86296) bytes
Initialization complete. Memory available: (962280) bytes. Memory in use: (86296) bytes
Initialization complete. Memory available: (962256) bytes. Memory in use: (86320) bytes
Initialization complete. Memory available: (962224) bytes. Memory in use: (86352) bytes
```

As a table:

|**#**|**Free**|**Delta**|**In use**|**Delta**|
|----:|-----:|----:|-----:|---:|
|    0|962992|     | 85584|    |
|    1|962992|    0| 85584|   0|
|    2|962552| -440| 86024| 440|
|    3|962280| -272| 86296| 272|
|    4|962280|    0| 86296|   0|
|    5|962280|    0| 86296|   0|
|    6|962256|  -24| 86320|  24|
|    7|962224|  -32| 86352|  32|

No changes to the config file between ~~reboots~~ dis-and-re-connects.

It's unclear what causes this at this point.

jimmy-mcelwain, 2024-11-11:
Just as an update to this, I am running iron, which includes the previously mentioned PR (https://github.com/micro-ROS/rmw_microxrcedds/pull/280). But it seems to still have the same memory issues. 

Here's a table showing the memory free and delta on my YRC1000 with Iron

|**#**|**Free**|**Delta**|
|----:|-----:|----:|
|    0|797368|     |
|    1|796848| -520|
|    2|796800|  -48|
|    3|796712|  -88|
|    4|796712|    0|
|    5|796752|  +40|
|    6|796752|    0|
|    7|796800|  +48|
|    8|796568| -232|
|    9|796448| -120|
|   10|796184| -264|
|   11|796072| -112|
|   12|796072|    0|
|   13|795976|  -96|
|   14|795920|  -56|
|   15|795928|   +8|
|   16|796000|  +72|
|   17|795888| -112|
|   18|795792| -96|
|   19|795744| -48|

It is still unclear to me where the problem is. I can try to track it down. But with how inconsistent it is, it is tough to tell what exactly is going wrong.

jimmy-mcelwain, 2024-11-13:
[On my very ugly branch](https://github.com/jimmy-mcelwain/motoros2/tree/memory_leak_wip), I have fixed seemingly all of the memory leaks from disconnect/reconnect cycles except for one. I believe that I have narrowed down the remaining leak to [`rcl_support_t`](https://github.com/Yaskawa-Global/motoros2/blob/225ed07df2e889c4688ede5b4276076de0576945/src/CommunicationExecutor.c#L206). When it gets initialized, it allocates 368 bytes, but when it is freed it only frees 344 bytes. I don't know what is happening to the other 24 bytes. But every single disconnect/reconnect cycle (except for the very first one, where some memory is permanently allocated), the total loss is exactly 24 bytes across the whole program, so I am almost certain that this is the only remaining problem. I don't know if this is a problem upstream or a problem with how we are using it. My branch is a branch off of `iron_wip`. There is no particular reason for that, I was just using iron at the time that I decided to try to fix this issue. The memory leak exists for both humble and iron, though. 

If anybody has any ideas on how to fix this last leak, please let me know.

jimmy-mcelwain, 2024-11-13:
I created a new branch off of main and copied over the relevant fix from my fork and made a PR. This will make it more clear what changes I made, and decrease the impact of the memory leak for the time being. The other leak that I mentioned in my previous comment is still present, though.

gavanderhoorn, 2024-11-15:
Yes, that is an issue upstream. Seeing as the same code is in `ros2/rcl`, I would [suggest posting there](https://github.com/ros2/rcl/issues) (if there isn't already an issue tracking this). Please post a comment here linking to it after you've opened it.

I would however suggest to check whether it's been perhaps already fixed in upstream's Humble, Jazzy or Rolling branches and we / micro-ROS is just lagging those. Just to avoid posting about an already resolved issue.

> I'm not suggesting that we merge the branch that I just made

indeed, I don't believe that'd be necessary. The leak is very small and it doesn't affect functionality in MotoROS2. I would suggest we wait until upstream patches it and then we can update to those versions.

jimmy-mcelwain, 2024-11-15:
I created an issue here https://github.com/ros2/rcl/issues/1198

jimmy-mcelwain, 2024-12-10:
Apparently it is the responsibility of the rmw to free that memory. So I opened a new issue here: https://github.com/micro-ROS/rmw_microxrcedds/issues/311.

jimmy-mcelwain, 2025-03-13:
The previously mentioned upstream issue is resolved now (https://github.com/micro-ROS/rmw_microxrcedds/issues/311)

Similar to https://github.com/Yaskawa-Global/motoros2/issues/376, I guess we're just going to wait until the next release of `rmw_microxrcedds` and package it in to the next build of libmicroros. 

The fix for humble/jazzy completely differs from the fix in rolling/that will be used in future distributions but we shouldn't have to think about that.

gavanderhoorn, 2025-03-19:
I would suggest we release `0.2.1` just to include the memory leak and the port parsing fixes. Even though they were fixed upstream, they fix problems in MotoROS2 and a new release wouldn't be too much work.

We wouldn't have any releases in `rmw_microxrcedds` to point to, but we can build `micro_ros_motoplus` against specific commits as well (we already do, just not for those packages).

As the fix is not included on the `foxy` nor the `galactic` branches, those builds of MotoROS2 will not get those fixes, but that'd be OK. Those versions of MotoROS2 are EOL already.

ted-miller, 2025-03-20:
> I would suggest we release 0.2.1 just to include the memory leak and the port parsing fixes.

Throw in https://github.com/Yaskawa-Global/motoros2/issues/11 and you've got a deal.

> As the fix is not included on the foxy nor the galactic branches, those builds of MotoROS2 will not get those fixes, but that'd be OK. Those versions of MotoROS2 are EOL already.

I don't want to continue including those binaries at all. (0.2.0 should be the last one.) Our releases have gotten way too cluttered.

ted-miller, 2025-03-20:
> I would suggest we release 0.2.1 just to include the memory leak and the port parsing fixes.

See https://github.com/Yaskawa-Global/micro_ros_motoplus/pull/19

> Throw in https://github.com/Yaskawa-Global/motoros2/issues/11 and you've got a deal.

See https://github.com/Yaskawa-Global/micro_ros_motoplus_buildscripts/pull/163

gavanderhoorn, 2025-03-31:
> I don't want to continue including those binaries at all. (0.2.0 should be the last one.) Our releases have gotten way too cluttered.

well it isn't really that much work.

Might be nice to just keep including them until they don't build anymore.

ted-miller, 2025-06-20:
TODO: Remove this issue from the README
https://github.com/Yaskawa-Global/motoros2/tree/9f8d26edecb8fd89abc9e852685e80a7c9f4f00b?tab=readme-ov-file#memory-leak

ted-miller, 2025-06-20:
This fix has been included in the `20250328` version of `libmicroros` and will be included in the `0.2.1` release

PR #425 (2025-06-20): Remove reference to memory leak (fixed)
Closes #35
closed 2025-06-20 by ted-miller