Yaskawa-Global/motoros2 #37: `start_traj_mode` fails if home position is not configured

gavanderhoorn, 2023-06-01:
If the home position is not set, or the home position needs to be checked, the controller refuses to start INFORM jobs.

For MotoROS2 specifically, this means that `start_traj_mode` & `start_point_queue_mode` fail with:

```shell
$ ros2 service call /start_traj_mode motoros2_interfaces/srv/StartTrajMode
requester: making request: motoros2_interfaces.srv.StartTrajMode_Request()

response:
motoros2_interfaces.srv.StartTrajMode_Response(result_code=motoros2_interfaces.msg.MotionReadyEnum(value=108), message='The INIT_ROS job has not started. Please call start_traj_mode service')
```

There is no indication on the pendant as to "why" the job failed to start. 

It is unclear whether this issue can be detected, or whether MotoROS2 can provide a better error message.

gavanderhoorn, 2023-06-01:
Note: this is also an issue with MotoROS1.

ted-miller, 2024-01-22:
We need to document this error in the readme.
https://github.com/Yaskawa-Global/motoros2/discussions/201#discussioncomment-8208576

They should be seeing a message on the pendant 
[image]

When attempting to start a job (using the pendant), they should get error 0380 Check Position

gavanderhoorn, 2024-01-22:
Should we instead not have MR2 check whether the home pos is set and raise an appropriate alarm if it isn't?

ted-miller, 2024-01-22:
It looks like we already are
https://github.com/Yaskawa-Global/motoros2/blob/9026a1d0b50b483f7c334bfd5ebbac8b328bfdb6/src/MotionControl.c#L1514

So that's apparently broken.

gavanderhoorn, 2024-01-22:
You mean we should be seeing:

https://github.com/Yaskawa-Global/motoros2/blob/9026a1d0b50b483f7c334bfd5ebbac8b328bfdb6/src/ErrorHandling.c#L27

?

ted-miller, 2024-01-22:
Yeah. That's what I would have thought.

Though perhaps 0x3040 doesn't apply if the home position has been registered and just needs to be validated. If that's the case, then I don't know of any mechanism to detect that this scenario.

gavanderhoorn, 2024-02-06:
Just ran into this when testing something for #19:

```
[...] [...]: Ros_MotionControl_StartMotionMode: enter
[...] [...]: Can't start 'INIT_ROS' because: 'The home position is not registered' (0x3040)
[...] [...]: Ros_MotionControl_StartMotionMode: exit
[...] [...]: Ros_ServiceStartPointQueueMode_Trigger: The INIT_ROS job has not started. Please call start_traj_mode or start_point_queue_mode service (108)
```

seems [this](https://github.com/Yaskawa-Global/motoros2/blob/9026a1d0b50b483f7c334bfd5ebbac8b328bfdb6/src/ErrorHandling.c#L27) isn't completely broken.

yai-rosejo, 2024-02-15:
I started looking into this issue. I found that the debug statement is occurring at:
https://github.com/Yaskawa-Global/motoros2/blob/10d8f00317c5d4ef7960627962337bf72e2ba817/src/MotionControl.c#L1503C1-L1516C6

The r.Data is not getting out of the BOOL Ros_MotionControl_StartMotionMode function.

This then gets put out into a failure to start in Ros_ServiceStartTrajMode_trigger:
https://github.com/Yaskawa-Global/motoros2/blob/10d8f00317c5d4ef7960627962337bf72e2ba817/src/ServiceStartTrajMode.c#L56C1-L78C6

Which then calls Ros_Controller_GetNotReadySubcode(), this function doesn't see the warning only that the job has not started via:
https://github.com/Yaskawa-Global/motoros2/blob/10d8f00317c5d4ef7960627962337bf72e2ba817/src/ControllerStatusIO.c#L447C1-L449C45

yai-rosejo, 2024-02-16:
This same problem also occurs in Issue #192 I believe

gavanderhoorn, 2024-09-11:
@jimmy-mcelwain @ted-miller: #265 and #297 have been linked as closing this issue, but I don't believe they address the suspected cause, do they?

jimmy-mcelwain, 2024-09-11:
So neither of those PRs would allow start_traj_mode to succeed if home position is not configured, but it would give a better response. I believe that the response if the home position is not configured would be `The controller has an active Error: Can't turn on servo because: 'The home position is not registered' (0x3040)`. I don't believe that jobs can be run while the home position is not registered, so I don't know if anything can be done besides giving a useful error message.

closed 2024-10-14 by ted-miller