Yaskawa-Global/motoros2 #43: Hesitation between trajectory points

ted-miller, 2023-06-01:
I'm commanding what should be a smooth continuous motion between two points. But, there is a slight (visible) hesitation between the points.

Happens on both streaming and FJT interfaces.

It's not always there, or at least it's not always noticeable, but I believe that's just because I'm a human, looking at a robot and not measuring stuff.

Possibilities:
- processing delay due to motoros2_fjt_pt_queue_proxy 'hammering' the point queue service server (I reduced the time between attempts to 0.05 sec)
- some emergent math behaviour in the code calculating the accelerations (and the velocities, etc)
- the code that actually processes the JointTrajectoryPoints (via the 'fake' single-point trajectories)

We don't have anything to back any of that up though right now.

ted-miller, 2023-06-30:
Just noticed this again. So I took a capture using the PP oscope function. Notice the dimples in what should be a sinusoidal wave.

[image]

I'm cycling through 4 points, moving back and forth.

```
switch(sequenceId)
        {
        case 0: 
        requestQueuePoint->point.positions[0] += 0.174533;
        requestQueuePoint->point.positions[4] += 0.174533;
        requestQueuePoint->point.velocities[0] = 0.174533;
        requestQueuePoint->point.velocities[4] = 0.174533;
        sequenceId += 1;
            break;
        case 1:
        requestQueuePoint->point.positions[0] += 0.174533;
        requestQueuePoint->point.positions[4] += 0.174533;
        requestQueuePoint->point.velocities[0] = 0;
        requestQueuePoint->point.velocities[4] = 0;
        sequenceId += 1;
            break;
        case 2:
        requestQueuePoint->point.positions[0] -= 0.174533;
        requestQueuePoint->point.positions[4] -= 0.174533;
        requestQueuePoint->point.velocities[0] = -0.174533;
        requestQueuePoint->point.velocities[4] = -0.174533;
        sequenceId += 1;
            break;
        case 3:
        requestQueuePoint->point.positions[0] -= 0.174533;
        requestQueuePoint->point.positions[4] -= 0.174533;
        requestQueuePoint->point.velocities[0] = 0;
        requestQueuePoint->point.velocities[4] = 0;
        sequenceId = 0;
            break;
        }
```

gavanderhoorn, 2023-06-30:
Would be interesting to have a recording of a regular trajectory execution as well. Just so we can compare.

gavanderhoorn, 2023-09-01:
I can't be sure, but I believe I noticed similar *hesitation* during a regular trajectory execution yesterday.

Should try the scope and plot the velocity.

gavanderhoorn, 2024-06-24:
I've just returned from ROSCon France and talked to some users of MotoROS2 which have also experienced the described *hesitation*. The videos and plots they shared showed a much more pronounced *start-stop* type of motion (basically: *really* start-stop-start-stop-...).

We agreed they will post their findings as (a) comment(s) on this issue.

gavanderhoorn, 2024-07-02:
@marip8 @Yolnan: as you are some of the users of the point queue proxy node with fairly long and complex trajectories: have you ever observed the *hesitation* described in the OP?

marip8, 2024-07-02:
I have noticed that usually our freespace motions for scan and plan look a little jerky even though they are being planned by `trajopt` and time-parameterized by ISP or TOTG. I hadn't thought to describe this as "hesitation" before, but the robot does seem to slow down slightly at what might be the trajectory waypoints. I don't really notice this behavior when our robots have been doing process motions, but that might be because they usually move more slowly and through a much denser trajectory than when doing freespace moves.

You could probably recreate this freespace behavior we are seeing by using the MoveIt Rviz motion planning panel to plan and execute arbitrary freespace moves around the workspace.

zakjeens, 2024-07-12:
Hi, as discussed during ROSConFR, I am posting here our 'hesitation' issue. 
We have two Yaskawas and have seen two behaviors that could be described as hesitation:
1. On a MA1440 used for welding we saw that the robot would make a full stop at each point if the points in the trajectory were far enough. For example: 1cm or 0.5cm between trajectory points = stops, yet 1mm or 0.5mm (or less) = fluid motion
The stopping can be clearly seen on the joint positions & velocities, as well as on the EE pose (acquired from TF)
[image]
[image]

[link to a repository of tests](https://github.com/zakjeens/yaskawa_debug_materials/tree/main)

2. On a GP7 we saw something that looks a lot more like vibration (or very rapid start&stop) having a 1mm distance between points.
[example of vibration](https://youtube.com/shorts/nuFha25ocz8?feature=share)

Thanks to some internal discussions and a discussion with @gavanderhoorn, we understood that we have been filling our FollowJointTrajectory and QueueTrajPoint messages incorrectly. As the driver wouldn't accept a message without velocity array correctly initialised, we have been resizing it to the correct size (automatically filled with 0.0). Therefore, if the trajectory is slow and the points are further apart, then the robot is capable of executing the trajectory exactly as requested. Meaning that it gets to each point and stops (because desired velocity=0). And when the points are too close, the velocity term is impossible to honor, so we get a fluid motion. 

So don't do like we did, and fill not only positions & timestamp, but also velocities of your trajectory messages! :)

gavanderhoorn, 2024-07-12:
Thanks @zakjeens.

> So don't do like we did, and fill not only positions & timestamp, but also velocities of your trajectory messages! :)

yes, indeed.

@ted-miller: we might want to improve/add to the documentation of the queuing service.

It's apparently unclear it's supposed to be used to "drip feed" a complete `JointTrajectory` with time, position, velocity *and* acceleration correctly set.

ted-miller, 2024-07-12:
I have tracked the issue down to how `Ros_MotionControl_AddToIncQueueProcess`. But I'm struggling to understand it. I'll pick this up again next week.

ted-miller, 2024-07-15:
So, I did the calculation of what the expected increment should be for the target speed of `0.174533`. On my axis[4], that should be `+/- 39` pulse. However, at the peak, I'm seeing `+- 55` pulse. So it seems that the "dimple" in the chart is the correct target. The robot's acceleration curve overshoots the target and comes back down.

[image]

I just had a discussion with @yai-rosejo and he believes that is normal/expected behavior, given the equations for the accCoef and jerk. Though, he's unsure of a couple things, in particular, the [2 * startVel](https://github.com/Yaskawa-Global/motoros2/blob/f61a7238d78066aa565542b6fefd112bad42d0cb/src/MotionControl.c#L446). But he does think he could minimize the overshoot with some adjustments.

To me, the amount of overshoot seems excessive. I feel like it would interfere with any process-control application (welding, dispensing, grinding, etc). 

@EricMarcil @gavanderhoorn What do you think? I seems that no one has complained about this motion profile for over a decade. So, maybe I'm making a big deal out of nothing.

Is this a non-issue?

gavanderhoorn, 2024-07-16:
No, I don't believe it's a non-issue.

If clients specify a velocity to be attained at a particular point in the trajectory we should make sure to reach that velocity.

Does the regular FJT server also show this behaviour? I would expect it does, seeing as it uses the same `Ros_MotionControl_AddToIncQueueProcess(..)`.

Perhaps we can find/come up with a better (fi: higher order) velocity and acceleration profile generation implementation.

gavanderhoorn, 2024-07-16:
Maybe @EricMarcil still knows how the current equations were derived / where they come from?

ted-miller, 2024-07-16:
> If clients specify a velocity to be attained at a particular point in the trajectory we should make sure to reach that velocity

To clarify... The target-command velocity is the current velocity when the robot reaches the target-command point. But before reaching that point, the velocity overshoots the target before dropping back down. 

I don't know if this image is any better or not:

[image]

> Does the regular FJT server also show this behaviour?

Yes.

> Perhaps we can find/come up with a better (fi: higher order) velocity and acceleration profile generation implementation

John has some ideas on adjusting the accel/jerk scaling. Apparently he recently did some of this type of coursework for his Masters degree. 

> Maybe @EricMarcil still knows how the current equations were derived / where they come from?

John seemed to think these are pretty standard equations out of a textbook.

EricMarcil, 2024-07-18:
I developed those equations a long time ago. At that time the requirement was to have time, position, velocity at each control points. Acceleration was not required and therefore had to be calculated. Jerk was not considered. And we don't know what the next point position or velocity might be because we haven't received it yet. So, the velocity is only garantied at the control point. We were also not sure about the processing time and tried to keep it fairly simple.

It seemed to worked well enough at that time to get the project on track and was never revisited afterwards or closely monitored as we are doing above. So there might be some better way of doing this and optimizing it by adding new criteria.

Aneeer, 2025-01-28:
Know that nobody's posted on this thread for a while, but I'm seeing stuttering on a dx200 controller with an MH180-120. We are composing FJT actions with velocities and accelerations just fine, but we're seeing stuttering from point to point for larger freespace motions when planning with Trajopt. I'm wondering if anyone found a fix or a band aid for this? It's worth noting we don't see this in simulation with a mock robot state publisher using the same robot urdf/xacro.

ted-miller, 2025-01-29:
No, this issue has not been addressed since the last post. I'll try to dig back into this.

ted-miller, 2025-01-30:
> We are composing FJT actions with velocities and accelerations just fine, but we're seeing stuttering from point to point for larger freespace motions when planning with Trajopt

Would you be able to post a short video of the robot. I just want to make sure we're looking at the same thing.

--------------------------------

I've been working on this today. I don't understand the original accel calculations at all. (Looks like some sort of spline thing... I don't know.) So I've been trying to replace it with a constant linear acceleration. It's not working yet.

ted-miller, 2025-01-31:
So, through the process of debugging my attempt at using a constant linear acceleration, I have come to realize that my problem was self-inflicted. I have not been using a motion planner. Instead I just hardcoded some points that sounded good in my head.

It's not possible to achieve my target position and target velocity without overshooting during execution. I didn't properly take accel into account. But the targets I specified are being reached correctly.

We ran a few points through moveit and saw smooth accel/decel.

I'm going to close this issue. My original complaint is invalid. 

-------------------------------------

@Aneeer, if you're still having issues, please open a new discussion with some video and logs/graphs of your trajectory.

closed 2025-01-31 by ted-miller

jimmy-mcelwain, 2025-01-31:
I think that the equations that we have right now should work well for motions that are made using a motion planner. 

What we use now is the most simple equation that can meet all 4 constraints of the motion calculation (initial position, initial velocity, final position, and final velocity). We are using cubic interpolation. With cubic interpolation, given a pair of points with position and velocities, there is only 1 possible path (4 constraints, cubic function), and so that path is not optimized in any way. The way that we do it now has no guarantee that there won't be "dimples", but it really depends on the path.

If the given points have an unreasonable position, velocity, or most notably large time delta between points, it could give an unreasonable path. But it should work really well for trajectories that have reasonable position/velocity changes and have "small" time deltas. I don't think that it is really the responsibility of MotoROS2 to make sure that "unreasonable" trajectories get executed smoothly. The thing that I am most worried about is points with a large time delta, especially with "drip feed" point queueing. But I think that our current equation does what it needs to do if the trajectory is made with a motion planner, and if we focus too much on making it smoother it sort of gets into the territory of motion planning.