Yaskawa-Global/motoros2 #52: Support Iron Irwini

gavanderhoorn, 2023-06-06:
Largely depends on micro-ROS.

They are tracking their status at: https://github.com/micro-ROS/micro_ros_setup/pull/647.

gavanderhoorn, 2023-06-22:
If I'm not mistaken, Iron support has just been released upstream: [micro-ROS Iron: 4.1.0](https://github.com/micro-ROS/micro_ros_setup/releases/tag/4.1.0).

gavanderhoorn, 2023-08-23:
@ted-miller: I believe you suggested adding this to the `0.2.0` milestone in #120?

gavanderhoorn, 2023-12-13:
Something to be aware of: https://github.com/ros2/rmw_dds_common/pull/68.

Almost starts to pay off to implement some sort of CI that checks message compatibility between different ROS 2 distributions.

ted-miller, 2024-03-12:
Doesn't build yet. But I'm working on this:
https://github.com/Yaskawa-Global/motoros2/tree/iron_wip

ted-miller, 2024-03-12:
> update VS project with Iron targets for all supported controllers

This kinda sucks to do and the current method does not scale well. I'm assuming the `cmake-passthrough` mode of M+SDK would make compilation easier. But I'm not sure how intellisense would behave without the stuff that was manually added.

ted-miller, 2024-03-25:
Very quick/preliminary testing. But I wanted to write it down before I leave.

AFAICT, libmicroros seems OK. (More testing needed)

Other things I noticed using the Iron version
- During setup, I had to `set S2C1402=3 and reboot` twice. This was probably a mistake on my part, but I really thought I set it right on the first attempt.
- I got the `invalid job` alarm. Then I deleted the job, rebooted, and let it regen a new job. I then got the alarm again. Something fishy here.
- Looking at `joint_states`, the cross-axis-coupling compensation routine isn't perfect. I'm seeing slight fluctuations in T axis when jogging others. But this is likely due to rounding errors on our part and could probably be fixed with https://github.com/Yaskawa-Global/motoros2/issues/199.
- When attempting to use any of the motoros2_interfaces messages, I get `Segmentation fault (core dumped)`. As I'm typing this, I realize that I sourced a `humble` version of these messages and never rebuilt them for `iron`. So it's probably that.

ted-miller, 2024-03-26:
> I got the invalid job alarm. Then I deleted the job, rebooted, and let it regen a new job. I then got the alarm again. Something fishy here

I did it yet again, and the alarm didn't occur. So, it seems that I just failed to delete the job. But I'm confused, because I was 90% sure that I _did_ delete the job. Perhaps I just did a cpu-reset too quickly after the deletion???

------------
> When attempting to use any of the motoros2_interfaces messages, I get Segmentation fault (core dumped). As I'm typing this, I realize that I sourced a humble version of these messages and never rebuilt them for iron. So it's probably that.

Yeah, I rebuilt the messages for `iron` and it worked.

---------------
I had one terminal open that was echoing robot_status and forgot to stop the echo. It was just running in the background.

I opened another terminal and started making service calls. The majority of them "worked", but returned an error message.

```
ros2 service call /read_group_io motoros2_interfaces/srv/ReadGroupIO "{address: 7001}"
2024-03-26 13:14:07.410 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7411: open_and_lock_file failed -> Function open_port_internal
requester: making request: motoros2_interfaces.srv.ReadGroupIO_Request(address=7001)

response:
motoros2_interfaces.srv.ReadGroupIO_Response(result_code=0, message='Success', success=True, value=192)
```

But if I go back to the original terminal, I can make those service calls without any issue. 

It seems that fastrtps might not be threadsafe. More testing is needed to submit a ticket to eProsima.

A quick google search said that someone got a similar error with a version mismatch.

My docker container:
```
ros2 doctor --report | grep fastrtps
2024-03-26 13:17:01.132 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7411: open_and_lock_file failed -> Function open_port_internal
rosidl_typesupport_fastrtps_cpp           : latest=3.0.2, local=3.0.2
rmw_fastrtps_shared_cpp                   : latest=7.1.3, local=7.1.3
rosidl_dynamic_typesupport_fastrtps       : latest=0.0.2, local=0.0.2
rosidl_typesupport_fastrtps_c             : latest=3.0.2, local=3.0.2
rmw_fastrtps_cpp                          : latest=7.1.3, local=7.1.3
fastrtps_cmake_module                     : latest=3.0.2, local=3.0.2
middleware name    : rmw_fastrtps_cpp
```

I'm entirely not sure which of our packages we would want to compare that too. I guess it would be the Agent, since it's doing the translation between fastrtps and microxrcedds. 

The Agent container:
```
ros2 doctor --report | grep fastrtps
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/__init__.py: 154: UserWarning: Fail to call QoSCompatibilityReport class functions.
rosidl_typesupport_fastrtps_cpp           : latest=2.2.2, local=2.2.2
rmw_fastrtps_shared_cpp                   : latest=6.2.6, local=6.2.5
rosidl_typesupport_fastrtps_c             : latest=2.2.2, local=2.2.2
rmw_fastrtps_cpp                          : latest=6.2.6, local=6.2.5
fastrtps_cmake_module                     : latest=2.2.2, local=2.2.2
middleware name    : rmw_fastrtps_cpp
```

Definitely some differences. Don't know if they're important.

---------------------------
I was surprised that `control_msgs` wasn't included by default in the Iron docker image. I feel like it was included in Humble.

I had to manually clone and build this.

--------------------------
After running the FJT script, I got this message in my client.

`control_msgs.action.FollowJointTrajectory_Result(error_code=-500301, error_string='Final position was outside tolerance. Check robot safety-limits that could be inhibiting motion. [group_1/joint_1: 0.0000 deviation] [group_1/joint_5: 0.0000 deviation]')`

This is very confusing. 

Looking at the logging script, I think it might be due to the time tolerance rather than position.

```
[1711461114.98510480] [192.168.1.31:50724]: 2024-03-26 13:51:54.983911 Trajectory complete
[1711461114.98533487] [192.168.1.31:50724]: 2024-03-26 13:51:54.983911 FJT using DEFAULT goal time tolerance: 500000000 ns
[1711461114.98539233] [192.168.1.31:50724]: 2024-03-26 13:51:54.984111 FJT action failed
[1711461114.98543978] [192.168.1.31:50724]: 2024-03-26 13:51:54.984111 Final position was outside tolerance. Check robot safety-limits that could be inhibiting motion. [group_1/joint_1: 0.0000 deviation] [group_1/joint_5: 0.0000 deviation]
```

I might need to open a new issue for this.

ted-miller, 2024-03-26:
Except for the misc notes above, it seems to be working. I was able to run the topics, services, and both types of motion.

gavanderhoorn, 2024-03-26:
> I did it yet again, and the alarm didn't occur. So, it seems that I just failed to delete the job. But I'm confused, because I was 90% sure that I _did_ delete the job. Perhaps I just did a [...]

I seem to remember having run into this as well, and then we added this to the [Alarm: 8014[1]](https://github.com/Yaskawa-Global/motoros2/blob/b066b5fe6ac0f7e750254e986153d236ada6052e/doc/troubleshooting.md#alarm-80141) FAQ:

> If the alarm is posted again after restarting the controller, make sure to allow sufficient time for the controller to properly delete the job (flushing changes to the file system may take some time). Allow for at least 20 seconds between deleting the job and restarting the controller.

Could be what you experienced.

> > When attempting to use any of the motoros2_interfaces messages, I get Segmentation fault (core dumped). As I'm typing this, I realize that I sourced a humble version of these messages and never rebuilt them for iron. So it's probably that.
> 
> Yeah, I rebuilt the messages for `iron` and it worked.

I would've expected that yes.

Key things have changed between Humble and Iron. Especially https://github.com/ros2/rmw_dds_common/pull/68 breaks *everything* between those versions.

> I had one terminal open that was echoing robot_status and forgot to stop the echo. It was just running in the background.
> 
> I opened another terminal and started making service calls. The majority of them "worked", but returned an error message.
> 
> [..]
> 
> A quick google search said that someone got a similar error with a version mismatch.
> 
> [..]
> 
> Definitely some differences. Don't know if they're important.

The set of packages you've used to built M+ `libmicroros` is *old*. It's from around July last year. I would suspect that to be the cause here before anything else.

It's also slightly surprising to see the SHM transport mentioned in the error message. AFAIK, that's explicitly disabled in the Agent Docker image.

> I was surprised that `control_msgs` wasn't included by default in the Iron docker image. I feel like it was included in Humble.
> 
> I had to manually clone and build this.

Didn't `apt install ros-iron-control-msgs` work?

Slightly off-topic, but:

> Looking at the logging script, I think it might be due to the time tolerance rather than position.
> 
> ```
> [1711461114.98510480] [192.168.1.31:50724]: 2024-03-26 13:51:54.983911 Trajectory complete
> [..]
> ```

is your debug listener out-of-date? That formatting (the date duplication) was fixed a long time ago.

ted-miller, 2024-03-26:
> The set of packages you've used to built M+ libmicroros is old. It's from around July last year. I would suspect that to be the cause here before anything else.

That's fair.

> > I had to manually clone and build this.
> 
> Didn't apt install ros-iron-control-msgs work?

I didn't know that had an apt package.

> is your debug listener out-of-date?

Extremely. I keep forgetting to update it.

ted-miller, 2024-03-26:
> The set of packages you've used to built M+ libmicroros is old. It's from around July last year. I would suspect that to be the cause here before anything else.

Do you have a trick to updating this? I'm just going through the repos list and manually checking for newer versions.
I'm assuming the following repos are applicable:
- eProsima/Micro-CDR - _No changes_
- eProsima/Micro-XRCE-DDS-Client (with private modifications) - **Update needed**
- micro-ROS/rmw-microxrcedds.git - _Inconsequential changes_

- micro-ROS/micro-ROS-Agent - _There is a newer version of this involving "thread". But it really seems inconsequential._

gavanderhoorn, 2024-03-26:
I have a tool which checks our `.repos` against upstream. But part of it is manual work, yes.

I'll take a look tomorrow.

ted-miller, 2024-03-26:
I pulled the latest Micro-XRCE-DDS-Client and applied the custom changes for a quick test.

I was not able to reproduce the original issue after updating. But I can't say with any certainty that the problem was fixed since I didn't reproduce the original issue multiple times. (hindsight being 20/20 and all...)

gavanderhoorn, 2024-03-28:
Combined with the latest test build of Iron M+ `libmicroros` (shared with you in the other thread), https://github.com/Yaskawa-Global/motoros2/compare/main...gavanderhoorn:iron_no_remap_rules should build.

ted-miller, 2024-06-20:
@gavanderhoorn 
Is Iron still of any benefit? I've asked @jimmy-mcelwain to start looking at this. But I see that microros now supports Jazzy.

gavanderhoorn, 2024-06-24:
Yes. We should support Humble, Iron and Jazzy.

Humble and Jazzy are LTS, so will be around for quite some time.

Iron is not an LTS, but should prepare us for Jazzy.

gavanderhoorn, 2024-07-01:
Updates to `micro_ros_motoplus`: Yaskawa-Global/micro_ros_motoplus#3.

ted-miller, 2024-09-06:
> find solution for broken CLI argument parsing (MotoROS2 does not have a CLI) due to upstream micro-ROS / rcl changes

I'm of the opinion that we just preprocessor it out for the time being. I think this issue has been sitting here too long.

> update documentation:
> Extracting the files

I'm not sure what this means

-----------------

Jimmy and I have both tested the current state. Everything is working well (with remap rules not supported).

jimmy-mcelwain, 2024-10-30:
I tried doing a full build (`RCL_MICROROS=OFF`). When I do so and load it onto the controller, it is missing the definition for the symbol `strtoll`. If I add a definition, then I am able to properly load it onto the controller and boot. However, as soon as I try to connect with my desktop client, it errors out [at this line](https://github.com/Yaskawa-Global/motoros2/blob/225ed07df2e889c4688ede5b4276076de0576945/src/ControllerStatusIO.c#L153-L158). The `rcl_ret_t` here is just `RCL_RET_ERROR`, nothing descriptive. I don't know what the problem is. I did notice that this is the same as #292, which. It seems to often fail on this step if there is some sort of communication error. I am certain, though, that I am not running into the same network problem as the poster there. I can keep the code exactly the same [on iron wip](https://github.com/Yaskawa-Global/motoros2/tree/iron_wip) with the exception of the definition of `RCL_MICROROS_COMPLETE_IMPL`, and it will succeed with the libmicroros build with `RCL_MICROROS=ON` and fail with the build with `RCL_MICROROS=OFF`. Since it only errors out upon connection with the client, I thought it could be an rmw problem, but I am having trouble figuring it out. 

@ted-miller suggested that I manually perform the remapping through the config file (change the string that is passed in when the publisher etc. is initialized), so I will probably do that sometime since I haven't been able to figure out how to remap using the intended mechanism.

gavanderhoorn, 2024-10-30:
I'm only one voice here, but:

> @ted-miller suggested that I manually perform the remapping through the config file (change the string that is passed in when the publisher etc. is initialized), so I will probably do that sometime since I haven't been able to figure out how to remap using the intended mechanism.

please don't do this.

Using parameters for topic/service/action names is a bit of an anti-pattern.

The real solution would be to address this one: https://github.com/ros2/rcl/issues/998, or figure out why a full build doesn't boot.

jimmy-mcelwain, 2024-11-04:
I still haven't been able to figure out why a full build won't boot, but I did get a build that got remapping to work. I created a finer-grain selection and decoupled the argument parsing from `RCL_MICROROS`, so you can have `RCL_MICROROS=ON` and a new option, `RCL_REMAPPING_ENABLED=ON` at the same time. 

I have forks of [micro_ros_motoplus](https://github.com/jimmy-mcelwain/micro_ros_motoplus/tree/iron_update_20240327), [micro_ros_rcl](https://github.com/jimmy-mcelwain/micro_ros_rcl/tree/iron-wip-candidate-20240621), and [MotoROS2](https://github.com/jimmy-mcelwain/motoros2/tree/iron_wip). Forking is disabled for micro_ros_motoplus_buildscripts, so I just put it [another branch](https://github.com/Yaskawa-Global/micro_ros_motoplus_buildscripts/tree/iron_attempt2) based on one that Ted had access to that I didn't. 

I haven't tested this thoroughly, but remapping topics and services works as expected. I will look through the micro_ros_rcl in particular and check the flags and `#ifdef` statements and everything

I may spend some more time trying to figure out why a full build fails. It would be nice if we could get a full build, because it's going to be a pain to integrate upstream changes in the future with this new `RCL_REMAPPING_ENABLED` being decoupled from `RCL_MICROROS`. 

I've attached a build of my libmicroros

[libmicroros_yrc1000_iron.zip](https://github.com/user-attachments/files/17623483/libmicroros_yrc1000_iron.zip)

jimmy-mcelwain, 2024-11-08:
I have it all building and remapping now without a ton of preprocessor modifications. The reason that I was having problems with the full build was because the `rcl_node_type_cache` wasn't getting initialized, which I didn't realize. The `rcl_node_type_cache` is used when `RCL_MICROROS=OFF`. It uses dynamic memory allocation, which I know is limited on our controllers, but in our case I doubt that it will be a problem. I will do a more thorough check on how much memory it uses in the future. `rcl_node_type_cache` gets [automatically initialized in Jazzy](https://github.com/micro-ROS/rcl/blob/8f2fd763347d83b21e3b90760f283d4468b60b55/rcl/src/rcl/node.c#L290-L293), but in Iron, you have to deal with it manually. So whenever we upgrade to Jazzy we'll have to get rid of the [cache initialization that I added to motoros2](https://github.com/Yaskawa-Global/motoros2/blob/19a2ee41ce7e69c76a9f06359e4149d6b7e5a1ef/src/CommunicationExecutor.c#L183-L186).

I there are branches on [motoros2](https://github.com/Yaskawa-Global/motoros2/tree/iron_wip), [micro_ros_motoplus](https://github.com/Yaskawa-Global/micro_ros_motoplus/pull/3), [micro_ros_rcl](https://github.com/Yaskawa-Global/micro_ros_rcl/tree/iron-wip-candidate-20241108), and [micro_ros_rcl_buildscripts](https://github.com/Yaskawa-Global/micro_ros_motoplus_buildscripts/tree/iron_attempt2) that have all been updated.

I started to rename branches but I realized that could have negative impacts part of the way through, so the naming convention doesn't make much sense right now, but it's all linked here and once it gets merged, it should be alright. So if you can't automatically pull one of the branches, that may be the reason.

What I have tested has worked on Iron. I need more thorough testing though. Remapping works. I have only tested a YRC1000 at the moment. motoros2 humble builds, but I haven't tested it with the changes. The iron specific stuff is `#ifdef`'d away, so I don't expect problems. 

I'm also not positive how we want to handle merging everything into main. I did not update the `one_func_per_file` for `rosidl` in `micro_ros_motoplus`. Should I create a new branch on our fork of `rosidl` for iron? Right now I am referencing an iron commit from the official repo. And `micro_ros_motoplus_buildscripts` will need some changes if we want the main branch to build both humble and iron. That probably won't be too difficult, though.

ted-miller, 2024-11-15:
Iron Irwini will be skipped because it is EOL next month. The [work on this topic](https://github.com/Yaskawa-Global/motoros2/tree/iron_wip) will be used to implement Jazzy.

closed 2024-11-15 by ted-miller