Yaskawa-Global/motoros2 #55: Rclc reports invalid name at startup (it's the default name)

ted-miller, 2023-06-08:
Getting `MotoROS2: Fatal Error (subcode: 5)` at startup due to `rclc_node_init_with_options = 201`. According to [RCL](https://github.com/ros2/rcl/blob/c43c7abb0a254f4b60c42c4c2993917eb08cb0ec/rcl/include/rcl/types.h#LL59C38-L59C38) this is "invalid name". 

The config yaml does not have a custom name defined. And unfortunately #33 is not implemented.

[motoros2_config.zip](https://github.com/Yaskawa-Global/motoros2/files/11691620/motoros2_config.zip)

Last three of MAC is `0A:37:5F`

```
[1686246952.68823862] [192.168.1.31:49225]: MotoROS2 0.1.0 - boot
[1686246952.68832254] [192.168.1.31:49225]: M+ libmicroros version: 'humble-20221102-dbg'
[1686246952.68837357] [192.168.1.31:49225]: PlatformLib version: 0.2.2
[1686246952.69634867] [192.168.1.31:49225]: No new configuration file found on CN102 USB drive.
[1686246952.69658685] [192.168.1.31:49225]: Checking configuration file: motoros2_config.yaml
[1686246952.80940700] [192.168.1.31:49225]: Config: agent_ip_address = 192.168.1.50
[1686246952.80963087] [192.168.1.31:49225]: Config: agent_port_number = 8888
[1686246952.80969644] [192.168.1.31:49225]: Config: sync_timeclock_with_agent = 1
[1686246952.80986404] [192.168.1.31:49225]: Config: stop_motion_on_disconnect = 1
[1686246952.80994272] [192.168.1.31:49225]: Config: publish_tf = 1
[1686246952.81012607] [192.168.1.31:49225]: Config: namespace_tf = 1
[1686246952.81279826] [192.168.1.31:49225]: Config: log_to_stdout = 0
[1686246952.81306601] [192.168.1.31:49225]: Config: executor_sleep_period = 10
[1686246952.81313133] [192.168.1.31:49225]: Config: action_feedback_publisher_period = 20
[1686246952.81330657] [192.168.1.31:49225]: Config: controller_status_monitor_period = 10
[1686246952.81341267] [192.168.1.31:49225]: Config: robot_status = sensor_data
[1686246952.81348944] [192.168.1.31:49225]: Config: joint_states = sensor_data
[1686246952.81363964] [192.168.1.31:49225]: Config: tf = default
[1686246952.81368661] [192.168.1.31:49225]: Waiting for robot alarms to clear...
[1686246955.31549644] [192.168.1.31:49225]: RMW settings 1: 1000; 0; 1000; 256; 256; 512
[1686246955.31571889] [192.168.1.31:49225]: RMW settings 2: 1; 1; 30; 15; 30; 1; -1
[1686246955.31579876] [192.168.1.31:49225]: RMW settings 3: 60; 120; 100; 100
[1686246955.31585240] [192.168.1.31:49225]: Using ROS domain ID: 0
[1686246955.31591654] [192.168.1.31:49225]: Using client key: 0x790A375F
[1686246955.31598020] [192.168.1.31:49225]: Attempting to connect to Micro-ROS PC Agent (at udp://192.168.1.50:8888)
[1686246956.33997822] [192.168.1.31:49225]: Found Micro-ROS PC Agent
[1686246956.34515023] [192.168.1.31:49225]: rclc_support_init_with_options = 0
[1686246956.34523773] [192.168.1.31:49225]: remap_rules str: ''
[1686246956.34540486] [192.168.1.31:49225]: len(remap_rules str): 0
[1686246956.34547782] [192.168.1.31:49225]: num parsed remap rules: 0
[1686246956.34554648] [192.168.1.31:49225]: faux argc: 2
[1686246956.34561539] [192.168.1.31:49225]: rcl_parse_arguments = 0
[1686246956.34569979] [192.168.1.31:49225]: rclc_node_init_with_options = 201
[1686246956.36735678] [192.168.1.31:49225]: motoRosAssert: MotoROS2: Fatal Error (subcode: 5)

```

ted-miller, 2023-06-08:
After changing the name in the config file, and rebooting, we're getting `8003[102] Controller Config Invalid`. This indicates that `MOTOMAN DRIVER` option function is disabled. After enabling it, it booted ok. But, I'm not clear why `rclc_node_init_with_options` is failing when using the default name

gavanderhoorn, 2023-06-08:
if you revert the name change in the config file, does it fail again?

ted-miller, 2023-09-04:
This appears to be an instance of #144 and should be resolved with #145.

PR #145 (2023-09-04): Always output double digits when generating default node name
As per subject.

It's legal for MAC addresses to contain zeros. Before this change, those would be formatted as ` 0`. As the format specifies a two column field, whitespace would be used for the second column, ultimately leading to spaces in the node name.

Whitespace is illegal in ROS resource names (all ROS versions), leading to `RCL_RET_NODE_INVALID_NAME` errors being reported by `rclc_node_init_with_options(..)`.

To fix this, force `snprintf(..)` to use leading zeros instead of spaces.

This should fix #144, and probably also f [... cut]
closed 2023-09-04 by ted-miller