Yaskawa-Global/motoros2 #58: Debug log always broadcast on `LAN2`

gavanderhoorn, 2023-06-09:
With `motoros2_config.yaml` setup to use `LAN3` on a YRC1000 (or at least: using an IP address in the range configured on `LAN3`) it appears debug log messages are only sent to `LAN2`.

Without `LAN2` having a connection, no messages will be received.

gavanderhoorn, 2023-06-09:
We have it hard-coded to always only use `LAN2` (ie: `ROS_USER_LAN1`):

https://github.com/Yaskawa-Global/motoros2/blob/8c1a524e8cad8e45e865454bb01678cabc416e38/src/Debug.c#L18-L34

Note: this could be *by-design*, but I though to report it anyway.

ted-miller, 2023-06-09:
> Note: this could be by-design, but I though to report it anyway.

I don't think this is by-design. Just an oversight.

jimmy-mcelwain, 2024-09-10:
So would the preferred solution be to publish to all enabled LAN ports simultaneously? Or maybe to add an option in motoros2_config.yaml for the user to configure which LAN port to send the message?

ted-miller, 2024-09-10:
I vote for the first option

gavanderhoorn, 2024-09-10:
I'd actually suggest implementing the configuration option, as that would cover multiple use-cases:

1. not configuring anything: broadcast on all active interfaces
2. configuring a specific interface: broadcast on that
3. disable broadcasting completely

@ted-miller's prefers option 1.

Option 2 would be my choice.

Option 3 is something which I believe we've discussed before and would be nice to have: in some cases it would/could be preferable to disable the debug log completely -- for security reasons fi -- and this would be one way to implement that.

We'd have to come up with a nice way to encode for "don't use any of the available interfaces".

Or perhaps we should not combine these, and have two options: global (en|dis)able of debug logging and one which configures which interface to use.

Similar to the user lan monitoring: this would only really make sense for controllers with multiple interfaces. On anything else, the setting will be ignored. Except disabling logging completely of course.

jimmy-mcelwain, 2024-10-25:
Resolved by #309 and #321

closed 2024-10-25 by jimmy-mcelwain