Yaskawa-Global/motoros2 #66: Release `FollowJointTrajectory` -> point queue conversion node

gavanderhoorn, 2023-06-16:
During `beta1`, a `FollowJointTrajectory` action server was implemented that uses the `queue_traj_point` service offered by MotoROS2 ([docs](https://github.com/Yaskawa-Global/motoros2/blob/5d594ee0216ad46aa19a1b16d1467fd0168f1367/doc/ros_api.md#queue_traj_point)) to stream `JointTrajectory`s of "infinite" length to the controller.

This works around the `200` point maximum trajectory length limitation which currently exists in MotoROS2 ([known issue](https://github.com/Yaskawa-Global/motoros2/tree/5d594ee0216ad46aa19a1b16d1467fd0168f1367#maximum-length-of-trajectories)). Length of trajectories would then only be limited by the amount of memory available to the conversion node (the same limitation that applies to all ROS 2 nodes).

This node was never publicly released. It should be made available, to avoid duplication of effort and to avoid different implementations of such nodes offering different user experiences.

gavanderhoorn, 2023-06-26:
I'm looking into this (again), but I still run into the issue with increased CPU usage after every executed trajectory (this is an `rclpy` node).

Initially I thought I perhaps ran into https://github.com/ros2/rclpy/pull/1070 (or: the issue it describes), but with that patch applied CPU usage still keeps going up, at a rate of about 5% per completed goal (of course, this is relative to the CPU performance).

marip8, 2023-07-06:
Does this node exist somewhere on a fork where we could test it out?

gavanderhoorn, 2023-07-07:
No, it's currently only in a private repository.

I'm still trying to figure out where my "CPU leak" is coming from.

smith-doug, 2023-08-18:
@gavanderhoorn  Can you add me to the repo for this?  I'd like to take a look.  It has become an issue for a current project.

gavanderhoorn, 2023-08-22:
I'll see if I can easily do that.

I've not worked on it lately, and I need to check what the status is.

Might just make it a *public* repository.

gavanderhoorn, 2023-10-04:
@smith-doug: have you had any luck using the proxy node? I haven't had any opportunity to look into the "CPU leak" yet.

jimmy-mcelwain, 2025-02-12:
I found the CPU leak [here](https://github.com/gavanderhoorn/motoros2_fjt_pt_queue_proxy/blob/9279969b6eec46b82fd4867ebeabfe534d653b6f/motoros2_fjt_pt_queue_proxy/server.py#L110-L114)

The problem is that the helper threads in the thread pool all make it to the line above, and then hang/wait (see the dump below). From the comment describing the line that causes the leak, this is clearly a known problem in theory, but it seems like the assumption was that it would only hang if the server died/disappeared, but it is seemingly happening (only sometimes) even if the server doesn't die. I don't know why yet. I'll look into it more. 

Here is the `py-spy` dump that I used to find the problem. 

```
Thread 10242 (active+gil): "MainThread"
    _spin_once_impl (rclpy/executors.py:789)
    spin_once (rclpy/executors.py:794)
    spin (rclpy/executors.py:294)
    main (motoros2_fjt_pt_queue_proxy/server.py:305)
    <module> (server:33)
Thread 10259 (idle): "ThreadPoolExecutor-0_0"
    wait (threading.py:320)
    wait (threading.py:607)
    call (rclpy/client.py:97)
    _queue_point (motoros2_fjt_pt_queue_proxy/server.py:114)
    fjt_goal_callback (motoros2_fjt_pt_queue_proxy/server.py:227)
    _execute_callback (simple_actions/simple_server.py:72)
    await_or_execute (rclpy/executors.py:107)
    _execute_goal (rclpy/action/server.py:333)
    __call__ (rclpy/task.py:239)
    run (concurrent/futures/thread.py:58)
    _worker (concurrent/futures/thread.py:83)
    run (threading.py:953)
    _bootstrap_inner (threading.py:1016)
    _bootstrap (threading.py:973)
Thread 10260 (idle): "ThreadPoolExecutor-0_1"
    wait (threading.py:320)
    wait (threading.py:607)
    call (rclpy/client.py:97)
    _queue_point (motoros2_fjt_pt_queue_proxy/server.py:114)
    fjt_goal_callback (motoros2_fjt_pt_queue_proxy/server.py:227)
    _execute_callback (simple_actions/simple_server.py:72)
    await_or_execute (rclpy/executors.py:107)
    _execute_goal (rclpy/action/server.py:333)
    __call__ (rclpy/task.py:239)
    run (concurrent/futures/thread.py:58)
    _worker (concurrent/futures/thread.py:83)
    run (threading.py:953)
    _bootstrap_inner (threading.py:1016)
    _bootstrap (threading.py:973)
Thread 10261 (idle): "ThreadPoolExecutor-0_2"
    wait (threading.py:320)
    wait (threading.py:607)
    call (rclpy/client.py:97)
    _queue_point (motoros2_fjt_pt_queue_proxy/server.py:114)
    fjt_goal_callback (motoros2_fjt_pt_queue_proxy/server.py:227)
    _execute_callback (simple_actions/simple_server.py:72)
    await_or_execute (rclpy/executors.py:107)
    _execute_goal (rclpy/action/server.py:333)
    __call__ (rclpy/task.py:239)
    run (concurrent/futures/thread.py:58)
    _worker (concurrent/futures/thread.py:83)
    run (threading.py:953)
    _bootstrap_inner (threading.py:1016)
    _bootstrap (threading.py:973)
Thread 10262 (idle): "ThreadPoolExecutor-0_3"
    wait (threading.py:320)
    wait (threading.py:607)
    call (rclpy/client.py:97)
    _queue_point (motoros2_fjt_pt_queue_proxy/server.py:114)
    fjt_goal_callback (motoros2_fjt_pt_queue_proxy/server.py:227)
    _execute_callback (simple_actions/simple_server.py:72)
    await_or_execute (rclpy/executors.py:107)
    _execute_goal (rclpy/action/server.py:333)
    __call__ (rclpy/task.py:239)
    run (concurrent/futures/thread.py:58)
    _worker (concurrent/futures/thread.py:83)
[... 40 lines cut ...]
    await_or_execute (rclpy/executors.py:107)
    _execute_goal (rclpy/action/server.py:333)
    __call__ (rclpy/task.py:239)
    run (concurrent/futures/thread.py:58)
    _worker (concurrent/futures/thread.py:83)
    run (threading.py:953)
    _bootstrap_inner (threading.py:1016)
    _bootstrap (threading.py:973)
Thread 10266 (idle): "ThreadPoolExecutor-0_7"
    wait (threading.py:320)
    wait (threading.py:607)
    call (rclpy/client.py:97)
    _queue_point (motoros2_fjt_pt_queue_proxy/server.py:114)
    fjt_goal_callback (motoros2_fjt_pt_queue_proxy/server.py:227)
    _execute_callback (simple_actions/simple_server.py:72)
    await_or_execute (rclpy/executors.py:107)
    _execute_goal (rclpy/action/server.py:333)
    __call__ (rclpy/task.py:239)
    run (concurrent/futures/thread.py:58)
    _worker (concurrent/futures/thread.py:83)
    run (threading.py:953)
    _bootstrap_inner (threading.py:1016)
    _bootstrap (threading.py:973)

```

jimmy-mcelwain, 2025-02-12:
I realized that part of the problem (the reason that threads kept on hanging for me), was because I didn't use a namespace, so I believe that the request was being sent to both the FJT server on the controller and the FJT server on the script. After namespacing the script FJT and sending the request to the namespaced server, I haven't been able to get any threads to hang at all. It is still possible for it to hang/the CPU leak still could pop up. But that being said I have probably done 100+ (small, 30 point) FJT movements in the last few minutes and I am not seeing high CPU utilization or any signs of a leak. 

@gavanderhoorn, is there any chance that you had the script and controller FJT servers in the same namespace? 

Regardless, I will try to change the script so that the threads won't hang if the `queue_traj_point` server does not respond. 

It would be nice if we could get this out there to sort of get around #85.

gavanderhoorn, 2025-02-13:
@jimmy-mcelwain: could you please repost your findings on https://github.com/gavanderhoorn/motoros2_fjt_pt_queue_proxy/issues/5? That way we keep things on-topic.