Yaskawa-Global/motoros2 #73: Make ROS 2 compatible support packages available (URDF, etc)

gavanderhoorn, 2023-06-21:
We have a set of support packages available over at [gavanderhoorn/motoman@ros2](https://github.com/gavanderhoorn/motoman/tree/ros2).

More will be added in the future, and the changes will be merged upstream.

gavanderhoorn, 2023-06-21:
I've pinned this to increase visibility.

marip8, 2023-07-06:
Do you think it would be possible to support both ROS1 and ROS2 in a revised support package repository? It seems like this should be possible with a tool like `ros_industrial_cmake_boilerplate` to set up the packages independent of `catkin` and `ament` and to install artifacts to the correct locations. It would be nice not to have two equivalent but repositories to maintain

gavanderhoorn, 2023-07-07:
Yes, that should be possible.

I don't believe we need RICB for that though. In the end, I believe "all" that would be needed would be a marker for the resource index, perhaps some meta-data for colcon and indeed installing things in the correct places.

For support pkgs specifically though there is one thing that we'd need to check additionally: IIRC, `package://` URIs are not supported everywhere, or at least not in Gazebo/`gazebo_ros2_control`. We do use those, so unless one of the (currently) known work-arounds works for us as well, that might prevent a ROS1-ROS2 combined approach.

gavanderhoorn, 2024-02-11:
Related discussion: https://github.com/ros-industrial/motoman/discussions/584.

gavanderhoorn, 2024-08-22:
As a heads up: I intend to update https://github.com/gavanderhoorn/motoman/tree/ros2 soon.

It's currently behind `kinetic-devel` of upstream `motoman` and a set of PRs is also slated to get merged there which will have to be migrated as well.

Due to the way the fork/branch works, this update will be in the form of a `force-push`, meaning all existing clones will have to `reset` their checkouts.