Yaskawa-Global/motoros2 #74: Make example MoveIt 2 configuration package(s) available

gavanderhoorn, 2023-06-21:
MoveIt 2 is non-trivial to configure.

Especially when *not* using `ros2_control`, as almost all available documentation and examples assume `ros2_control` is being used.

We give some hint(s) in the `README` ([Usage - With MoveIt](https://github.com/Yaskawa-Global/motoros2/tree/2d2b62df3655078b93cfe77c1c133ff5df8670fd#with-moveit)), but that's really rather minimal, and assumes an already working MoveIt 2 configuration package has been created, and just needs to be modified for use with MotoROS2.

Providing an example, known good configuration package would help users set up something for their own / other robots.

ted-miller, 2023-06-21:
@aaron-levan, did you keep any of your previous work on this topic?

Aaron-Levan, 2023-06-21:
No, I had everything in a VM that is, unfortunately, no longer with us...

Aaron-Levan, 2023-06-21:
> @Aaron-Levan, did you keep any of your previous work on this topic?

No, I had everything in a VM that is, unfortunately, no longer with us...

gavanderhoorn, 2023-06-21:
I keep putting this on my TODO, but it's too long already ..

I have working package for my GP25 + TSL1000, but it's for / from a very old version of MoveIt and not really suited for distribution.

charleswilmot, 2023-06-23:
At my company, we are currently deciding whether we should order Yaskawa MOTOMAN HC30PL arms or Fanuc CRX20IA arms.

Being able to use MoveIt is a necessary condition for us to purchase the Yaskawa robots.

Is there someone that we could contact before we take a decision? We would need to have one robot ready to use with MoveIt before October 2023.

gavanderhoorn, 2023-06-23:
@charleswilmot: could I ask you to please repost your question on the Discussion forum [in the Q&A section](https://github.com/Yaskawa-Global/motoros2/discussions/categories/q-a)? Your comment is related to this issue, but we're using this to track the work we need to do, and not for general MoveIt support related enquiries.

drolleigh, 2023-09-15:
Hello,
I have a simple moveit 2 config package for a Motoman gp25-12. I have motoROS2 successfully installed on the YRC1000 controller, and it is publishing the joint states that match the prepended group name. And it is broadcasting TFs with R1 as the namespace. The TFs appear to be accurate to the physical state of the robot. 

How do I make the moveit 2 launch files talk to motoROS2 on the robot controller? 

Any help is appreciated,
-D

gavanderhoorn, 2023-09-15:
We have some info in [Usage - with MoveIt](https://github.com/Yaskawa-Global/motoros2?tab=readme-ov-file#with-moveit). Would that be enough?

akashjinandra, 2023-09-26:
We are more wondering, what moveit files need to be launched. With ROS 1 we had followed the example of planning execution launch that was provided by the ROS industrial group: http://wiki.ros.org/Industrial/Tutorials/Create_a_MoveIt_Pkg_for_an_Industrial_Robot?action=AttachFile&do=view&target=moveit_planning_execution.launch.txt. I'm wondering now if there is a similar setup for motoros2. The MSA(moveit setup assistant) generates many files, but we aren't sure which one to launch when running with real robots.

gavanderhoorn, 2023-09-28:
I keep saying I'll try to make something available, but due to time constraints, I never get to it.

So I can't / don't want to promise anything, but I'll again try to look into it.

As I wrote above: the majority of the output of the MSA, at least when it comes to configuring the the driver interface can be ignored, as it's all `ros_control` focused.

jimmy-mcelwain, 2024-10-10:
I've been working on this the past few days. I have a mostly functional package for my GP12. I will continue working on it and upload a draft hopefully early next week. Would that be something that should be pushed to a new branch on this repo? Or should I put it elsewhere? I am using the support package from [here](https://github.com/gavanderhoorn/motoman/tree/ros2_rebased_and_cleaned/motoman_gp12_support) with a few small edits (prefix and joint names)

gavanderhoorn, 2024-10-11:
> Would that be something that should be pushed to a new branch on this repo?

no, it's not related to the driver so we should not host it here.

I'd suggest creating a repository for it on your personal account for now. Then after we test it, we can transfer it here to `yaskawa-global` and reference it in our documentation.

jimmy-mcelwain, 2024-10-16:
I have a some questions for you @gavanderhoorn. 

Earlier I was running into a problem where MoveIt2 wasn't working because there was no defined transform between `world` and `base_link`. So I created a mock `/tf_static` publisher in the launch file that published all zeros for the `world -> base_link` transform and got it working, but I am trying to find a real solution before I post anything here. 

My problem is that in `/tf`, we are publishing `world -> base`, `base -> flange`, etc. and we have no way to locate `base_link` within MotoROS2 to my knowledge, particularly since the `urdf` files that define that transform are not being stored in an accessible place. 

According to your documentation, the frame hierarchy should be [like this](https://gavanderhoorn.github.io/rep/rep-0199.html#id41), `base` should be a child of `base_link` but I don't see any way to link `world -> base_link`. In reality, we are calculating everything from `base`, but having that be the parent of `base_link` breaks convention. 

Maybe more notably, [the documentation says here](https://gavanderhoorn.github.io/rep/rep-0199.html#id29) that `base` must be fixed to it's parent frame. But since we are publishing a non-fixed `world -> base` and not `world -> base_link` and `base_link -> base`, it seems like that rule is being broken already. 

I wish that I could have a hierarchy like `world -> base (non-fixed)`, `base -> base_link (fixed)`, `base_link -> link_1(non-fixed)`, etc., but that is apparently not allowed. What should I do here?

gavanderhoorn, 2024-10-17:
Just making sure we're talking about the same things: the `base` frame described in the candidate REP you link is not Yaskawa's *Base* frame.

As to the TF frames published by MotoROS2: those don't get used by MoveIt.

Where does `world` come from? Is it a `link` in a URDF / XACRO you're using? It's not part of any of the models I'm aware of.

> I wish that I could have a hierarchy like `world -> base (non-fixed)`, `base -> base_link (fixed)`, `base_link -> link_1(non-fixed)`, etc., but that is apparently not allowed. What should I do here?

can you clarify *why* you wish you could have that frame hierarchy?

jimmy-mcelwain, 2024-10-17:
I believe that I misunderstood the way that external axes are handled in MotoROS2, so you can ignore part of what I said. I had an unnecessary `world -> base_link` virtual joint in my `srdf`. I thought that we would need to read the `world -> base` TF that was being published in MR2 to position the robot (essentially a transform from Yaskawa base frame to Yaskawa robot frame). But now I have been told that external axes will be included in the `urdf` and essentially treated as part of the robot. But I still have concerns/I am uncertain how to do some things. 

I have an updated version of my package [here](https://github.com/jimmy-mcelwain/motoros2_moveit2_config_packages). I slightly changed the `urdf` from your repo to update the naming convention so `joint_1_s -> group_1/joint_1` and `link_1_s -> group_1/link_1` so it will be a bit easier for users who stick with the default joint names in MR2 to deal with, but otherwise it should work with the `motoman_gp12_support` and `motoman_resources` packages from [your repo](https://github.com/gavanderhoorn/motoman/tree/ros2_rebased_and_cleaned). I am able to move the robot using MoveIt2 by setting joint angles and executing motion in Rviz.  

My first problem is that I am getting a error in Rviz when I open it through my launch file. When I run

`ros2 launch motoman_gp12_moveit2_config demo.launch.py`

I get

`Frame [group_1/base_link] does not exist` under Global Status: Error > Fixed Frame

If I go to Global Options, the only available Fixed Frame settings are those published on TF (e.g. world, r1/base, r1/flange, etc.). If I create a static transform `world -> base_frame` in my launch file, the error goes away. [Here is an example of another launch file that does this](https://github.com/moveit/moveit_resources/blob/ef5836ddf4d6a7773cee895a675bfa852a3798f6/panda_moveit_config/launch/demo.launch.py#L81-L88). To be clear, it still moves properly if I set joint angles even if the error is active. 

Another concern is that if I look at the position of tool0 in Rviz, it is offset by 0.45 meters in the z direction (the distance between `base_link` and `base`) compared to how it appears on the pendant. Additionally the orientation of tool0 appears to be incorrect, though I may just be reading it wrong. I was wondering if that would be problematic if a cartesian position is commanded. 

Some of these problems may be Rviz configuration problems, I am not sure. I know that the `moveit.rviz` on my repo gets immediately "outdated" upon launching it. I'll keep on looking into the problems, but if you have any suggestions please let me know. Thanks.

jimmy-mcelwain, 2024-10-24:
I had a misunderstanding regarding `robot_state_publisher`. I found [this discussion comment by you](https://github.com/Yaskawa-Global/motoros2/discussions/251#discussioncomment-9353399) @gavanderhoorn which cleared things up for me. Like the author of that discussion, I thought that `robot_state_publisher` was unnecessary. I thought that it was part of `ros2_control`, and since we are publishing `robot_status` that we didn't need it. I should have looked more into it and realized that `robot_state_publisher` is unrelated to `robot_status`. 

I have since updated my repo and I now have a `robot_state_publisher` node in my launch file. This clears up many of my problems. I also added a `warehouse_ros_sqlite` configuration which at least stores robot states properly. 

My only real concern at this point is that the position of `tool0` in `rviz` is 0.45m off in the z-direction. I know that this is because rviz is describing the position relative to `base_link` and not `base`, but it could be a little confusing and could perhaps cause problems. 

[Here is my repo](https://github.com/jimmy-mcelwain/motoros2_moveit2_config_packages/tree/main). I'm going to write instructions on how to configure the repo for different robot models. I am also not planning on hosting the `motoman_gp12_support` or `motoman_resources` on that repo in the future. But for now I think that the contents of `motoman_gp12_moveit2_config` are ready.

jimmy-mcelwain, 2024-10-31:
I have an example package for both GP12 and HC10DTP and a pretty detailed setup guide at [jimmy-mcelwain/motoros_moveit2_config_packages](https://github.com/jimmy-mcelwain/motoros2_moveit2_config_packages). I have never written a MoveIt config package, so if somebody could take a look and see if it is good, I would appreciate it. 

Also, I know that the old [ros-industrial/motoman](https://github.com/ros-industrial/motoman) repo had the support and config packages in the same repo. Is that something that we want to have again? Or should we host the support and config packages in separate repos? If we want separate repos, do we want to just host one 'example' config package? Or should we host more config packages as they come up?

ted-miller, 2024-11-01:
@jimmy-mcelwain I'll plan to go through your readme on Monday.

> Also, I know that the old [ros-industrial/motoman](https://github.com/ros-industrial/motoman) repo had the support and config packages in the same repo. Is that something that we want to have again? Or should we host the support and config packages in separate repos?

Personally, I don't see any reason to separate them.

ted-miller, 2024-12-19:
We should also consider documenting what differences there would be between a single-group and multi-group system.

https://github.com/Yaskawa-Global/motoros2/pull/340/files/59475bd24717251887e8516b946a068fc0455beb#r1881829265

gavanderhoorn, 2024-12-19:
As I only now read this:

@ted-miller wrote:
> Personally, I don't see any reason to separate them.

I would vote for keeping them separate.

MoveIt is just one motion planning framework. It's not universally used -- although admittedly it's most likely the first one anyone tries / comes into contact with.

Building workspaces *without* having to either ignore the MoveIt configuration packages or ignoring their dependencies (during `rosdep install [..]` fi) is annoying and sometimes even impossible.

If possible, keeping things separated would be a good idea, and it would also make it clear where to find them: if the MoveIt configurations would be hosted next to the support/description packages, what should be the name of the repository?

ted-miller, 2024-12-19:
Since this post, we had discussed (verbally) not hosting full moveit configurations. Instead, we just host the robot descriptions (urdf) and provide sample instructions for making a moveit configuration for a basic installation. 

That is the motivation for this repo (not yet public). https://github.com/Yaskawa-Global/robot_descriptions

jimmy-mcelwain, 2025-02-05:
I agree that it has become pretty convoluted and that people will have problems setting stuff up if they try to follow the instructions that I wrote right now. 

I submitted a couple of issues just now https://github.com/moveit/moveit2/issues/3314 and https://github.com/moveit/moveit2/issues/3315 that should make the instructions much easier to complete. 

I don't have any expectation that those issues would be resolved any time soon, considering the number of outstanding issues there. But they seem straightforward enough that maybe I could look into fixing them myself.  

If those are completed, users would still need to go through the MSA and afterwards copy over the launch file and `package.xml` and change the name of the robot, which I recognize is a potential problem. But it is easier than the current process. Maybe a straightforward script could do that for them. 

I have some concerns about writing a script/package that does much more than that. With the wide variety of arm configurations, planning groups, passive joints, numbers of joints, group states, end effectors, virtual joints,  etc. etc. I feel like to do a proper job of generating the `srdf` and all of the other config files, it would end up being much more complicated than it seems at first. We could use [srdfdom](https://github.com/moveit/srdfdom/tree/ros2) to do the actual writing without too much difficulty, but we would either have to make a lot of assumptions (about planning groups, passive joints, end effectors, etc.) or massively complicate the package by taking a bunch of user input. I feel like to make a proper package, we would need to do a rewrite of a lot of the MSA stuff.

gavanderhoorn, 2025-02-05:
The goal is not to be able to create MoveIt configurations for every possible Yaskawa controller configuration. Just to generate a simple, minimalistic MoveIt package for a single robot, with the default joint names (or perhaps configurable joint names), default topics, QoS, etc.

Just to get users something they can use to test MotoROS2 with.

MoveIt configurations are 95% the same for all Yaskawa robots MotoROS2 supports. Templating should be possible.

For any other use, the MSA would still have to be used.

jimmy-mcelwain, 2025-02-05:
Okay that makes sense. I can look into that.

jimmy-mcelwain, 2025-04-11:
I made a quick and dirty proof of concept config generator [here](https://github.com/jimmy-mcelwain/motoman_moveit2_config_generator). I'm using Jinja2 to turn the template config into a usable package. It works for my gp12, but it's not complete. Is this the sort of thing that you're looking for @gavanderhoorn? If it's worthwhile, I can put more thought into it and improve on it. 

I will also probably update my config tutorial repo (I can simplify/remove a couple of steps). Fortunately https://github.com/moveit/moveit2/issues/3314 got addressed. But https://github.com/moveit/moveit2/issues/3315 still makes the process convoluted.

gavanderhoorn, 2025-04-23:
@jimmy-mcelwain: yes, this goes in the direction of that I was thinking of.

I'll try to play with it a little.

> But [moveit/moveit2#3315](https://github.com/moveit/moveit2/issues/3315) still makes the process convoluted.

well that's partly why I suggested using a generator for these packages based on a known-good configuration.