Yaskawa-Global/motoros2 #79: Consider publishing (controller) statistics

gavanderhoorn, 2023-06-30:
As part of monitoring a running system, statistics on MotoROS2 performance and controller state would be nice to have.

There are no concrete plans to implement this, but to start a list of what could be part of such a state or statistics publication:

 - controller temperature (at least YRC provides this)
 - any kind of duty-cycle information available (load on servos, etc)
 - connect/disconnect cycles micro-ROS Agent
 - action server statistics: goals received, executed, rejected, aborted, cancelled, etc
 - traj statistics: max length (time and nr of pts), last length (time and nr of pts), total duration (cumulative), validation failures (perhaps split out by type of failure: start state, limits, etc))

YORROY123, 2026-11-09:
Hi, we'd like to +1 this, specifically for **encoder temperature**, and share a concrete use case.

We're building a monitoring / predictive-maintenance system for Yaskawa arc-welding cells, using MotoROS2 (Jazzy) with the micro-ROS Agent running on an arm64 host. `joint_states` (position, velocity, effort) already covers most of what we need. The one missing piece is per-axis motor/encoder temperature.

We noticed MotoROS2 already contains `Ros_CtrlGroup_GetEncoderTemperature()` in `src/CtrlGroup.c`, which wraps `mpGetEncoderTemp()`. It doesn't appear to be called or published anywhere yet. (From #49, it also seems to be the reason DN2.44 is the minimum DX200 version.)

Would you consider publishing it? For example:
- a low-rate topic (e.g. 1 Hz is plenty for thermal monitoring), per control group, or
- as part of the state/statistics publication proposed in this issue.

A related question: as far as we understand, only one MotoPlus application can be loaded on a controller at a time. That would mean a separate MotoPlus app (or the OPC UA Server option) can't run alongside MotoROS2 to provide this data. Is that correct for current YRC1000 system software?

Thanks for all the work on MotoROS2!

jimmy-mcelwain, 2026-11-09:
@YORROY123 thanks for the feedback we will consider that. 

> A related question: as far as we understand, only one MotoPlus application can be loaded on a controller at a time. That would mean a separate MotoPlus app (or the OPC UA Server option) can't run alongside MotoROS2 to provide this data. Is that correct for current YRC1000 system software?

There is no limitation regarding running multiple MotoPlus apps at once. It only becomes a problem if the apps try to claim the same resources. E.g. MotoROS2 claims certain registers, I/O bits, motion control, etc. so another app that tries to claim those same resources would not be compatible. I will say that I frequently run MotoROS2 and OPC UA server at the same time and I do not run into any problems. Unfortunately, I cannot guarantee with 100% certainty that they are fully compatible because I do not have the source code for our OPC UA server, but I've never had a problem running both at the same time.

ted-miller, 2026-11-09:
> There is no limitation regarding running multiple MotoPlus apps at once

With `MOTOMAN DRIVER` enabled (part of the MotoROS2 setup procedure), the limit is `3` apps.