Yaskawa-Global/motoros2 #80: Check use of unsafe stdio functions (`sprintf(..)`, `vsprintf(..)`, etc)

gavanderhoorn, 2023-06-30:
MotoROS2 uses quite a few of the functions in `stdio.h` which are involved in manipulating in-memory buffers, and potentially unsafe IO operations.

Examples are: `printf(..)`, `sprintf(..)`, `snprintf(..)`, `strcpy(..)`, `strncpy(..)`, etc.

The use of these functions should be audited and checked for any unsafe patterns that may have unintentionally ended up in MotoROS2. If needed, safer versions of these functions should be used.

A complicating factor might be that M+ does not always support the safer variants of these functions.

gavanderhoorn, 2023-06-30:
Tools like `cppcheck` and other static analysis tools could potentially assist with this.

SejalBehere, 2023-07-12:
I tried using cppcheck and Visual Lint for the analysis of all the files but am not getting any kind of error/warning messages even after the analysisis complete. Cppcheck didn't display any errors/warnings. Then I used cppcheck in Visual Lint and though it did analyze all the files and showed that issues had been found, it still did not display any error/warning messages and instead said "A message database is not currently available for Cppcheck." 

I have posted a discussion thread to the cppcheck discussion board. Here is the link to the thread:

https://sourceforge.net/p/cppcheck/discussion/general/thread/7098576e8c/

gavanderhoorn, 2023-07-12:
Could you show the command you used to run `cppcheck`? I've ran it in the past and it required quite a few flags.