Yaskawa-Global/motoros2 #91: Implicit declaration of `gettimeofday(..)`

gavanderhoorn, 2023-07-13:
#61 introduced usage of `gettimeofday(..)` here:

https://github.com/Yaskawa-Global/motoros2/blob/9fe8cf237f38c79b76ffb57ab111522ffb80d217/src/Debug.c#L69

this is not a function supported/exposed by the M+ SDK, so the M+ compiler complains about an *implicit declaration*:

> ```
> Debug.c(69): warning : implicit declaration of function 'gettimeofday'
> ```

For YRC, builds will work as `gettimeofday(..)` is present on the controller, but other controllers may not have this function causing the `.out` to not load. DX2 fi doesn't appear to support it (https://github.com/Yaskawa-Global/motoros2/issues/18#issuecomment-1629317081).

Regardless, the warning should be addressed (as it's really an error).

gavanderhoorn, 2023-07-13:
@ted-miller: I can only easily find `gettimeofday(..)` on YRC (but not as part of M+). DX2 does not appear to have it.

I expect FS to also not (directly) support it.

Would you have any suggestion on how to get around this?

ted-miller, 2023-07-13:
I've used `clock_gettime` on DX, FS, and YRC

```
typedef struct _timespec
{
    time_t      tv_sec;         /* seconds */
    long        tv_nsec;        /* nanoseconds (0 -1,000,000,000) */
} timespec;
extern int clock_gettime (int clock_id /* clock ID (always 0 */,  timespec * tp /* where to store current time */);
```

But that's nanoseconds instead of microseconds. So we'd need to rework

gavanderhoorn, 2023-07-14:
Not sure why, but I believe this prevents me from loading the `.out` on my YRC.

SejalBehere, 2023-07-14:
I'll look into it and get it done.

ted-miller, 2023-07-14:
> Not sure why, but I believe this prevents me from loading the .out on my YRC.

Are you getting alarm 1020 [5]?

> But without an updated PlatformLib, I can't open a PR yet.

I don't understand that comment

gavanderhoorn, 2023-07-14:
> > Not sure why, but I believe this prevents me from loading the .out on my YRC.
> 
> Are you getting alarm 1020 [5]?

yes, and the two others it always comes with.

> > But without an updated PlatformLib, I can't open a PR yet.
> 
> I don't understand that comment

Ok. Technically I could open a PR, but we'd need an updated PlatformLib to provide the `clock_gettime(..)` prototype. It'd be nice to include that update in the PR that switches to using `clock_gettime(..)`.

PR #99 (2023-07-14): Switch to `clock_gettime(..)`
Fixes #91 (well .. avoids it by just not using `gettimeofday(..)`).

Tested on my YRC.
closed 2023-07-14 by ted-miller