Yaskawa-Global/motoros2 #102: Cryptic error codes from mpStartJob

ted-miller, 2023-07-31:
When enabling a motion mode, the software tries to start the INFORM job. If it fails to start that job, it returns a cryptic error code. You would need to reference the MotoPlus documentation to find the definition. 

Ex: `Can't start 'INIT_ROS' because: 0x3040`

We should map the error codes to a lookup table with human-readable strings.

0x0000 Normal end
0x2010 Robot is in operation
0x2030 In HOLD status (PP)
0x2040 In HOLD status (External)
0x2050 In HOLD status (Command)
0x2060 In error/alarm status
0x2070 In SERVO OFF status
0x2080 Wrong operation mode
0x3040 The home position is not registered
0x3050 Out of range (ABSO data)
0x3400 Cannot operate MASTER JOB
0x3410 The JOB name is already registered in another task.
0x4040 Specified JOB not found

aigallegos, 2023-08-01:
Hi I would like to be assigned this. I'm pretty sure I can fix the problem I am just new to open source. Thanks!

ted-miller, 2023-08-01:
Thanks @aigallegos. We appreciate the help!

When you have a solution worked up, please submit a pull request to the main branch. If you need any info, let us know.

gavanderhoorn, 2023-08-02:
This was actually explicitly left as a `TODO` a couple of versions ago. We might have been too hasty removing those `TODO`s.

One is still left, here:

https://github.com/Yaskawa-Global/motoros2/blob/c2453273e80dfe2841d5338b08d489af943c5214/src/MotionControl.c#L1253-L1256

which I believe should also be addressed.

> We should map the error codes to a lookup table with human-readable strings.

I believe we already have a function which does this:

https://github.com/Yaskawa-Global/motoros2/blob/c2453273e80dfe2841d5338b08d489af943c5214/src/ErrorHandling.c#L11-L33

AFAICT, it maps the status codes you list.

PR #106 (2023-08-03): Fixed Cryptic Error Codes in mpStartJob
fixes #102. I wrote a new function that kept the error string format as an f-string for mpStartJob errors in MotionControl.c by just adding a new function called Ros_ErrorHandling_Only_ErrNo_ToString in ErrorHandling.c that did not use strncpy and therefore did not need a buffer to get passed in, similar to Ros_ErrorHandling_MotionNotReadyCode_ToString it just returns a char* instead of having a destination for the chars and can be directly passed into the error f-string.
closed 2023-08-03 by ted-miller