Yaskawa-Global/motoros2 #103: `/select_motion_tool` indicates success when tool doesn't exist

ted-miller, 2023-07-31:
My robot does not have multiple tools enabled. There is only the one tool file. If I try to select tool 5, it returns success.

CLI command:
`ros2 service call /select_motion_tool motoros2_interfaces/srv/SelectMotionTool "{group_number: 0, tool_number: 5}"`

Response:
```
requester: making request: motoros2_interfaces.srv.SelectMotionTool_Request(group_number=0, tool_number=5)

response:
motoros2_interfaces.srv.SelectMotionTool_Response(result_code=motoros2_interfaces.msg.SelectionResultCodes(value=0), message='', success=True)
```

gavanderhoorn, 2023-08-02:
Is there any API available which would allow checking which toolfiles have been configured?

`select_motion_tool` really only updates some internal state of MotoROS2, which will always succeed, as long as the value passed is "in range" (ie: less than `MAX_VALID_TOOL_INDEX`).

ted-miller, 2023-08-02:
> Is there any API available which would allow checking which toolfiles have been configured?

`mpSetToolNo` might be able to indicate that. There is a return code for "Tool is not changed". We could get the current tool first to avoid changing the tool with mpSetToolNo. (I remember there was some reason we didn't want to change it. I don't remember why.)

gavanderhoorn, 2023-08-02:
> There is a return code for "Tool is not changed". We could get the current tool first to avoid changing the tool with mpSetToolNo. 

hm, so attempt to change it, then change it back to what it was?

Would that not interrupt active motion?

> I remember there was some reason we didn't want to change it. I don't remember why.

We've discussed this in the *Port select_tool from MotoROS1* issue on the internal tracker. @EricMarcil had some good arguments against calling it, so we changed it from the ROS 1 implementation.

jimmy-mcelwain, 2024-10-02:
I spent some time trying to figure this out. I talked to @ted-miller and he found out that even if tool switching is prohibited and there is only one tool file visible on the controller side (`TCP ` is set to `NOT USED` in maintenance mode and `TOOL NO. SWITCH` is set to `NOT USED`), the 64 files are still populated with the correct information. I tested it, and M+ can still read the tool information even if the tool info isn't accessible on the pendant, so this isn't actually an issue.

closed 2024-10-02 by jimmy-mcelwain

gavanderhoorn, 2024-10-02:
So is the premise of this issue incorrect? Because it states "if they don't exist", which apparently isn't the case: they do 'always' exist.

> even if tool switching is prohibited and there is only one tool file visible on the controller side (`TOOL NO. SWITCH` is set to `NOT USED` in maintenance mode (which can also be accomplished by setting `TOOL NO. SWITCH` to `PROHIBIT` in `SETUP->TEACH COND`) `TCP ` is set to `NOT USED` in maintenance mode), the 64 files are still populated with the correct information.

is "correct information" all-zeros, or whatever happened to have been entered before you make them inaccessible?

Would there be any way to detect a service client attempts to select a tool file which is inaccessible/uninitialised (if that's even possible)?

I'm just wondering whether we should still give some kind of feedback to service clients they requested something which may not actually work or be what they want.

jimmy-mcelwain, 2024-10-02:
> So is the premise of this issue incorrect? Because it states "if they don't exist", which apparently isn't the case: they do 'always' exist.

That's right, the premise of the issue is incorrect, the tool files always exist. 

> is "correct information" all-zeros, or whatever happened to have been entered before you make them inaccessible?

It is whatever was entered before they were made inaccessible from the pendant side. It is not necessarily all zeros. 
 
> Would there be any way to detect a service client attempts to select a tool file which is inaccessible/uninitialised (if that's even possible)?
For uninitialized files, all tool files are always "initialized" to all zeros, and can be set/used without any modifications. I can set the tool number to a certain value in INFORM. We could manually check if the tool file is all zeros using `mpGetToolData`, but that would be the only way to see. There is no flag for initialization or anything like that. 

As for detecting if a client attempts to select a tool file that is inaccessible, the tool files are always accessible through `mpGetToolData`, but if we want to prevent users from switching tool numbers if `TOOL NO. SWITCH` is prohibited, that setting has a parameter associated with it that we could read. We could prevent users from switching tools if we wanted to based on that setting, but @ted-miller  told me that we can ignore that setting for MR2. I guess since it would just be imposed by us and there is no real need. 

> I'm just wondering whether we should still give some kind of feedback to service clients they requested something which may not actually work or be what they want.

It would be easy enough to check if the tool file is all zeros and give users a warning if it is, and/or we could give a warning if `TOOL NO. SWITCH` is prohibited. But there is no technical reason that we can't just set the tool number to what they request. I don't know what the best path is.

gavanderhoorn, 2024-10-03:
> It is whatever was entered before they were made inaccessible from the pendant side. It is not necessarily all zeros.

hm.

So to me this sounds like it could result in unwanted/dangerous situations. Or at least unexpected behaviour.

If a user setup some/a number of tool files, they must have had a reason to do so.

If they then go and disable access to those tool files using the setup item you mention, that must also have been done for a reason. Why else make it all inaccessible.

Could this be used as a 'safety' feature, or for locking out operators? Can INFORM jobs still change tool files even if they can't be browsed/edited using the PP?

> `TOOL NO. SWITCH` [..] has a parameter associated with it that we could read. We could prevent users from switching tools if we wanted to based on that setting, [..]

From what I currently understand I would perhaps be in favour of indeed checking whether `TOOL NO. SWITCH` is active and make the service fail in that case as well.

It would at least be consistent -- depending on course of whether INFORM jobs can still switch tools even with that setting enabled.

Overall I would say our goal should be to make MotoROS2 behave as much as possible as regular functionality on the controller, such as INFORM job execution and the UX the PP provides.

> but @ted-miller told me that we can ignore that setting for MR2.

@ted-miller: could you perhaps clarify why you feel we can ignore it?

gavanderhoorn, 2024-10-03:
Thanks for the clarification.

Would you agree though it could/would be confusing for users to have a system where tool switching is prohibited using the PP, but then be able to 'switch tools' (even though that's not really what we're doing) using the MotoROS2 ROS API?

ted-miller, 2024-10-07:
From what I can gather, this is just for locking operators out. A maintenance tech sets things up, then the normal operator can only press a few buttons.

IMO, the MR2 API shouldn't abide by this restriction. It would be the responsibility of the ROS Developer to decide what data is accessible to an operator.