Yaskawa-Global/motoros2 #114: `/start_traj_mode` failed incorrectly

ted-miller, 2023-08-07:
I was in an error state created by an estop. I tried to start traj mode. It failed due to the error state. But, the job did start and the drives came on. 

Eventually servo-sleep mode kicked in with the job still running. 

When I call start_traj_mode, the servos woke back up and the job was restarted. But the service returned `result_code=motoros2_interfaces.msg.MotionReadyEnum(value=109), message='INFORM job is not on a WAIT command. Check the format of INIT_ROS')`

I'm wondering if it was due to the servo-sleep or my error-reset situation.

A second call to /start_traj_mode worked ok.

ted-miller, 2023-08-07:
> A second call to /start_traj_mode worked ok.

Well maybe not. It indicated success. But when I tried to submit a goal, I got `control_msgs.action.FollowJointTrajectory_Result(error_code=-100208, error_string='Must call start_traj_mode service.')`

yai-rosejo, 2024-02-16:
Found a sequence that makes this message to appear and is repeatable:
e-stop -> turn off e-stop -> send reset_error -> send start_traj_mode -> wait till energy save mode turns on -> send fjt command -> inform error
```
[2024-02-16 19:34:21.624265] [192.168.1.31:65119]: reset: attempting to reset controller
[2024-02-16 19:34:21.624265] [192.168.1.31:65119]: reset: success
[2024-02-16 19:34:30.722604] [192.168.1.31:65119]: Ros_MotionControl_StartMotionMode: enter
[2024-02-16 19:34:31.211004] [192.168.1.31:65119]: Robot job is ready for ROS commands.
[2024-02-16 19:34:31.311004] [192.168.1.31:65119]: Robot job is ready for ROS commands.
[2024-02-16 19:34:31.367204] [192.168.1.31:65119]: Ros_MotionControl_StartMotionMode: exit
[2024-02-16 19:34:31.367204] [192.168.1.31:65119]: Ros_MotionControl_ActiveTrajMode = 1
[2024-02-16 19:34:31.367204] [192.168.1.31:65119]: Ros_ServiceStartTrajMode_Trigger: activated
[2024-02-16 19:35:54.783605] [192.168.1.31:65119]: Trajectory contains 61 points
[2024-02-16 19:35:54.783805] [192.168.1.31:65119]: Energy saving function is active. Re-enabling the robot.
[2024-02-16 19:35:54.783805] [192.168.1.31:65119]: Ros_MotionControl_StartMotionMode: enter
[2024-02-16 19:35:55.753405] [192.168.1.31:65119]: Robot job is ready for ROS commands.
[2024-02-16 19:35:55.753605] [192.168.1.31:65119]: Ros_MotionControl_StartMotionMode: exit
[2024-02-16 19:35:55.753605] [192.168.1.31:65119]: Ros_MotionControl_ActiveTrajMode = 1
[2024-02-16 19:35:55.753605] [192.168.1.31:65119]: Ros_ServiceStartTrajMode_Trigger: activated
[2024-02-16 19:35:55.753805] [192.168.1.31:65119]:
[2024-02-16 19:35:55.855605] [192.168.1.31:65119]: Robot has been re-enabled.
[2024-02-16 19:35:55.855605] [192.168.1.31:65119]: FollowJointTrajectory - Goal request rejected
[2024-02-16 19:35:55.855605] [192.168.1.31:65119]: The trajectory will be accepted and then immediately aborted
[2024-02-16 19:35:55.855805] [192.168.1.31:65119]: INFORM job is not on a WAIT command. Check the format of INIT_ROS
[2024-02-16 19:35:55.870405] [192.168.1.31:65119]: Robot job is ready for ROS commands.
```
```
control_msgs.action.FollowJointTrajectory_Result(error_code=-100109, error_string='INFORM job is not on a WAIT command. Check the format of INIT_ROS')
```

closed 2024-02-23 by ted-miller