Yaskawa-Global/motoros2 #131: VS: clean target fails: include directory not found

gavanderhoorn, 2023-08-23:
As per subject really.

Whenever I try to build the `clean` target of the VS project, I get:

```
Clean started...
1>------ Clean started: Project: MotoROS2, Configuration: DX200_foxy Win32 ------
1>EXEC : error : Include directory not found - C:\...\MotoPlus for Visual Studio\DX200_foxy\inc
1>mpBuilder : error : Problem with mpBuilder - Invalid Include Directory
1>C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.MakeFile.Targets(38,5): error MSB3073: The command ""C:\...\MotoPlus for Visual Studio\mpBuilder.exe" -c DX200_foxy -p "C:\...\motoros2\src\\" -n "MotoROS2" -b "C:\...\motoros2\DX200_foxy\\" -o clean" exited with code 1.
1>Done building project "MotoROS2_AllControllers.vcxproj" -- FAILED.
========== Clean: 0 succeeded, 1 failed, 0 skipped ==========
```

this aborts the build.

PS: it would also seem the build command still uses `MotoROS2` for the `-n` option. Not sure whether that's a problem or not.

ted-miller, 2023-08-23:
Looks like a bug in M+SDK. 

Short-term, it shouldn't be a problem. Every `build` is technically a `clean-rebuild` since there is no incremental build system with mpbuilder. 

I'll get it fixed sometime in September. 

Note to self: This applies to v1.5.1 of M+SDK.

gavanderhoorn, 2024-03-07:
Testing the new SDK build, I now see a different error:

```
------ Clean started: Project: MotoROS2, Configuration: YRC1000u_humble Win32 ------
C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppClean.targets(77,5): error MSB4044: The "CppClean" task was not given a value for the required parameter "FoldersToClean".
Done building project "MotoROS2_AllControllers.vcxproj" -- FAILED.
```

less 'error-y', but still not quite working.

On another mpBuilder project it does work though (`MotoROS_PlatformLib`), so perhaps the VS project is missing some (required) configuration?