Yaskawa-Global/motoros2 #137: Setup procedure for Smart Pendant

ted-miller, 2023-08-25:
I happen to be using the SP right now for another project. While trying to test MotoROS2, I have no idea how I'm supposed to set this up w/o switching to the annoying Classic Interface. 

We should really document the process.

Or... better yet, the SP supports [YIP packages](https://developer.motoman.com/en/yip/smart-packager). We could greatly simplify the installation process by maintaining an installer.

gavanderhoorn, 2023-08-25:
> I have no idea how I'm supposed to set this up w/o switching to the annoying Classic Interface.

me neither, but don't we already have #59? :)

ted-miller, 2023-08-25:
Not anymore we don't

gavanderhoorn, 2023-08-25:
> Or... better yet, the SP supports [YIP packages](https://developer.motoman.com/en/yip/smart-packager). We could greatly simplify the installation process by maintaining an installer.

Is there a command line version of the YIP packaging tool?

If not, that would mean 9 more GUI workflows to go through on every release.

Or are `.yip`s really `.zip`s with a different extension?

ted-miller, 2024-01-25:
I just learned the mechanism that SP uses to install Motoplus applications. It's got a size limit of < 1 MB. 

So... no YIP for this. (Unless we use the desktop version of the YIP installer.)

ted-miller, 2025-01-27:
There was some release note about increasing some memory area to 4MB. I have reason to suspect it's related to this. Need to investigate.

jimmy-mcelwain, 2025-03-24:
The mechanism used to load these files seemingly still has a hard cap of 1MB, even with the memory area increasing to 4MB. I don't believe that there is any technical reason for this limit (anymore). There's an open issue on the private smart pendant repo (`#6634`, for anybody that has access to that repo) that I've pinged a couple of times. Hopefully it will get addressed.

Should we make a setup procedure for smart pendant right now? Or should we wait until that issue (hopefully) gets addressed and makes its way to a software update? That could take a while.

gavanderhoorn, 2025-03-24:
I would not spend time on this without confirmed support for binaries > 1MB in `.yip`s.