Yaskawa-Global/motoros2 #144: Invalid Node Name when MAC address ends in 00

acbuynak, 2023-09-02:
## Issue
My controller has a MAC ending of `:6b:c8:00`

When the motoros2_config.yaml `node_name` is set to default, the last MotoROS2 app loads the last byte (`00`) as a single zero with a space (` 0`)

With a default motoros2_config.yaml, this results in...
```
Config: node_name = 'motoman_6b_c8_ 0'
```

and the subsequent rclc error code [(rcl codes reference)](https://github.com/ros2/rcl/blob/cc9ea2810ff7c181ccdd43dc1f5c336f531fa8d8/rcl/include/rcl/types.h#L57-L58)
```
rclc_node_init_with_options = 201
```

## Operational Patch

My solution was to just manually configure my `node_name`. This allows everything to work as expected.

---
## State of system which produced issue

MotoROS2 `v0.1.1`

**Hardware**
`YRC1000`
Controller Version: `YAS4.73.00A`
Robot. 1 x `GP20HL`

**Error on Teach Pendant:**
```
ALARM 8011 - MotoROS2: Fatal Error [5]
```

**Files**
1 - [20230901T210604_motoros2_debug_log.txt](https://github.com/Yaskawa-Global/motoros2/files/12502170/20230901T210604_motoros2_debug_log.txt)
2 - [micro_ros_terminal_output.txt](https://github.com/Yaskawa-Global/motoros2/files/12502171/micro_ros_terminal_output.txt)
3 - [PANELBOX.LOG](https://github.com/Yaskawa-Global/motoros2/files/12502472/PANELBOX.LOG)

acbuynak, 2023-09-02:
Realizing this could be related to #55?

gavanderhoorn, 2023-09-02:
Thanks for reporting this @acbuynak. And thank you for the clear bug report.

I've opened #145 with what I believe would be the fix.

> Realizing this could be related to #55?

The MAC of that controller does contain a zero in the third octet (client key is based on the MAC and is: `0x790A375F`). That would indeed be formatted as ` A`, just as in your case.

gavanderhoorn, 2023-09-02:
Added YRC labels as well as this issue is not specific to any controller version.

PR #145 (2023-09-04): Always output double digits when generating default node name
As per subject.

It's legal for MAC addresses to contain zeros. Before this change, those would be formatted as ` 0`. As the format specifies a two column field, whitespace would be used for the second column, ultimately leading to spaces in the node name.

Whitespace is illegal in ROS resource names (all ROS versions), leading to `RCL_RET_NODE_INVALID_NAME` errors being reported by `rclc_node_init_with_options(..)`.

To fix this, force `snprintf(..)` to use leading zeros instead of spaces.

This should fix #144, and probably also f [... cut]
closed 2023-09-04 by ted-miller