Yaskawa-Global/motoros2 #148: Segmentation Fault when running MotoROS services

Aneeer, 2023-09-05:
Hi Ted + Team, 

Moving my issue here from the email thread. Can currently see joint states from the robot but running any of the MotoROS2 services results in segmentation fault. The issue persists with echoing robot status. 

[image]
[image]
[image]
[image]

closed 2023-09-06 by Aneeer

gavanderhoorn, 2023-09-06:
> I was able to remedy most everything.

*most* everything?

Could you clarify what still didn't work?

And please still note that officially, Rolling is not supported.

ROS 2 does not guarantee binary compatibility between different versions of ROS, and MotoROS2 is only built on-top of binaries for Foxy (EOL), Galactic (EOL) and Humble at the moment.

It's very much possible breaking changes will be introduced in Rolling that will make it incompatible with MotoROS2.

Aneeer, 2023-09-06:
Sorry let me clarify :) 

1. Most everything is everything related to the issue I opened, it was a poor choice of words 
2. I have taken your advice and switched to Humble

Hence I closed the issue. Does that answer your question?

gavanderhoorn, 2023-09-06:
Yes. Thank you.