Yaskawa-Global/motoros2 #152: ALARM 8011  MotoROS2: Fatal Error [5]

KanishkNavale, 2023-09-15:
Hello,

Even tough we follow the setup instructions carefully, we get the following error:

ALARM 8011
 MotoROS2: Fatal Error
[5]

We have checked all the steps repeatedly and not found any consistency. I am attaching the following details,
Controller: YCR1000
ROS: Humble
[CMOS.zip](https://github.com/Yaskawa-Global/motoros2/files/12619008/CMOS.zip)
[PANELBOX.LOG](https://github.com/Yaskawa-Global/motoros2/files/12619002/PANELBOX.LOG)

KanishkNavale, 2023-09-15:
Hi @ted-miller ,
[motoros2_config.zip](https://github.com/Yaskawa-Global/motoros2/files/12619592/motoros2_config.zip)

Thank you for replying as soon as possible. I have attached the yaml config file.

ted-miller, 2023-09-15:
Try changing the default node name like so:

[image]

If that doesn't work, we'll need to see the output from the debug-listener script.