Yaskawa-Global/motoros2 #155: Trajectory doesn't Complete with Speed Limit set on Robot FSU

Aneeer, 2023-09-18:
ROS2 Version: Humble
MotoROS2 Version: Latest (0.1.1)
MicroROS Version: Latest

Hi team. So I'm in a bit of predicament where I don't know the best way to slow my robot from a functional safety standpoint without the controller throwing me an error. Wondering the best approach to robustly slow down the robot's movements in certain regions of the workspace from a safety standpoint. 

[image]
[motoros2_config.txt](https://githu
[20230918T161836_motoros2_debug_log.txt](https://github.com/Yaskawa-Global/motoros2/files/12653208/20230918T161836_motoros2_debug_log.txt)
b.com/Yaskawa-Global/motoros2/files/12652737/motoros2_config.txt)

ted-miller, 2023-09-18:
Hi @Aneeer 
Unfortunately, this is known issue. https://github.com/Yaskawa-Global/motoros2/issues/50
At this time, there is no single solution that works for all use-cases.

From a safety perspective, you are correct to be using the FSU in such a way that it satisfies the risk-assessment for your application. But MotoROS2 cannot detect when a limit has been enacted. It can only see that the intended trajectory-end was not reached.

The most basic answer is to monitor the feedback position (`/joint_states`, `/tf`) during execution to ensure that your trajectory is being followed as expected. If a significant deviation is detected, you can abort the existing trajectory and submit a new one. 

If you are using the `FollowJointTrajectory` action server (as shown above), then aborting the trajectory will force the robot to come to a complete stop before another trajectory can be submitted. 

If you are using the `QueueTrajPoint` service, then you may be able to react to the deviation in real time without halting. But please note that any data which has already been submitted to the queue will still be executed and potentially limited. 

-------------------------------
@gavanderhoorn: I kept this as an `issue` instead of `discussion` due to the output in the log. 

`Final position was outside tolerance. Check robot safety-limits that could be inhibiting motion. [joint_1_s: 0.0000 deviation] [joint_2_l: 0.0000 deviation] [joint_4_r: 0.0000 deviation] [joint_5_b: 0.0000 deviation]`

It seems to indicate that there was no deviation on the final point. So, that makes me wonder if this isn't hitting the a time-tolerance rather than a positional-tolerance. 

This should be investigated.

gavanderhoorn, 2023-09-19:
@Aneeer: you have `log_to_stdout: true`. Unless you have a specific reason, I would *strongly* recommend to set that to `false`. The debug logger logs the exact same output, and incurs much less overhead.

@ted-miller: agreed. But perhaps a new issue to track it would be better in that case.

gavanderhoorn, 2023-09-19:
I've opened #156 to track the need to document the current state of PFL/FSU support more clearly.

ted-miller, 2023-09-19:
In case you haven't seen [this post](https://github.com/ros-industrial/motoman/pull/542#issuecomment-1725873984), someone has updated MotoROS2 to include better behavior with the FSU. 

Once we integrate those changes, then the robot should follow the trajectory properly at the safety-reduced speeds. Then the only concern would be MoveIt potentially aborting the trajectory due to a time-constraint.

iydv, 2023-09-20:
Hi, the first version of [FSU Speed Limit Handling](https://github.com/ros-industrial/motoman/pull/542) integrated to motoros2 is here #157.

closed 2023-10-20 by ted-miller