Yaskawa-Global/motoros2 #156: Document current state of support for PFL/FSU

gavanderhoorn, 2023-09-19:
As evidenced by #155, the fact the FSU/PFL can interfere with trajectory execution my MotoROS2 is not obvious to users.

We have an open issue tracking this (#50), but that's not user-facing documentation.

We should add a section to the [Known issues & limitations](https://github.com/Yaskawa-Global/motoros2/tree/bd95e65679d582b13b3887e5277d34f7406db69a#known-issues--limitations) section instead.

Refer to #50, to keep discussion centralised, but clarify FSU/PFL support in MotoROS2 in the text itself.

gavanderhoorn, 2023-10-21:
@ted-miller: could you address this now that you've merged #155?

In the absence of a ROS API for "FSU status" (#50), documentation is the next best thing, and #155 changed the default behaviour of MotoROS2.

ted-miller, 2023-10-23:
Sure. Just gimme a few days to finish up some other tasks.

ted-miller, 2023-11-01:
I removed PFL from the title of this issue.
We haven't done any PFL testing on an HC. I'm concerned because PFL can not only limit robot motion, but also generate motion.

PR #187 (2023-11-08): Document FSU limitations
Fixes #156
closed 2023-11-08 by gavanderhoorn