Yaskawa-Global/motoros2 #166: Generated INFORM job detected as invalid

rsaintobert, 2023-09-28:
ROS2 Version: Humble
MotoROS2 Version: Latest (0.1.1)
MicroROS Version: Latest
YRC1000u
GP7

Hi. I am having trouble to make the driver work. 
I followed the installation procedure and I let MotoROS2 generate the INIT_ROS job.
I started the micro ros agent using the docker image and executed the "ros2 node list" cmd into the container (docker exec -it) but no ros node was visible. 
After reboot of the controller, the INIT_ROS job is detected as invalid and the alarm 8014 is raised. No ROS node is visible.

The weird thing is that I turns out to work one time without changing anything apparently. I was able to see the node and listen to the /joint_states topic, but after a reboot the problem appeared again.

<details>
<summary>Here are the logs of the agent</summary>

```bash
local-micro-ros-agent-1  | [1695908233.380252] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 9999
local-micro-ros-agent-1  | [1695908233.380331] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
local-micro-ros-agent-1  | [1695908315.098871] info     | ProxyClient.cpp    | ProxyClient              | session hard timeout enabled | client_key: 0xB5258455, timeout: 10000 ms
local-micro-ros-agent-1  | [1695908315.098906] info     | Root.cpp           | create_client            | create                 | client_key: 0xB5258455, session_id: 0x81
local-micro-ros-agent-1  | [1695908315.098964] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0xB5258455, address: 192.168.255.1:42474
local-micro-ros-agent-1  | [1695908315.110142] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0xB5258455, participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.128229] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xB5258455, topic_id: 0x000(2), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.129441] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0xB5258455, publisher_id: 0x000(3), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.132151] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0xB5258455, datawriter_id: 0x000(5), publisher_id: 0x000(3)
local-micro-ros-agent-1  | [1695908315.234171] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xB5258455, topic_id: 0x001(2), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.236989] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0xB5258455, publisher_id: 0x001(3), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.240042] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0xB5258455, datawriter_id: 0x001(5), publisher_id: 0x001(3)
local-micro-ros-agent-1  | [1695908315.245597] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xB5258455, topic_id: 0x002(2), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.248058] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0xB5258455, publisher_id: 0x002(3), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.250550] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0xB5258455, datawriter_id: 0x002(5), publisher_id: 0x002(3)
local-micro-ros-agent-1  | [1695908315.253837] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xB5258455, topic_id: 0x003(2), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.256189] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0xB5258455, publisher_id: 0x003(3), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.259048] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0xB5258455, datawriter_id: 0x003(5), publisher_id: 0x003(3)
local-micro-ros-agent-1  | [1695908315.262554] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0xB5258455, requester_id: 0x000(7), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.266670] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0xB5258455, requester_id: 0x001(7), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.268991] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0xB5258455, requester_id: 0x002(7), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.271409] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xB5258455, topic_id: 0x004(2), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.274315] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0xB5258455, publisher_id: 0x004(3), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.276964] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0xB5258455, datawriter_id: 0x004(5), publisher_id: 0x004(3)
local-micro-ros-agent-1  | [1695908315.279497] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xB5258455, topic_id: 0x005(2), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.281887] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0xB5258455, publisher_id: 0x005(3), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.285010] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0xB5258455, datawriter_id: 0x005(5), publisher_id: 0x005(3)
local-micro-ros-agent-1  | [1695908315.334888] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0xB5258455, requester_id: 0x003(7), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.339744] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0xB5258455, requester_id: 0x004(7), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.342384] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0xB5258455, requester_id: 0x005(7), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.346158] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0xB5258455, requester_id: 0x006(7), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.350938] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0xB5258455, requester_id: 0x007(7), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.354202] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0xB5258455, requester_id: 0x008(7), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.358204] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0xB5258455, requester_id: 0x009(7), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.360912] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0xB5258455, requester_id: 0x00A(7), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.363761] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0xB5258455, requester_id: 0x00B(7), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.369887] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0xB5258455, requester_id: 0x00C(7), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.373540] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0xB5258455, requester_id: 0x00D(7), participant_id: 0x000(1)
local-micro-ros-agent-1  | [1695908315.376741] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0xB5258455, requester_id: 0x00E(7), participant_id: 0x000(1)
```

[... cut ...]
# then an alarm will be raised at startup.
# This flag indicates that the job is intentionally configured and will
# suppress the alarm.
#
# When this flag is set to 'true', then MotoROS2 will not validate whether the
# job is compatible. It is the responsibility of the user to make sure the
# custom job includes the required INFORM statements and in the expected order
# (refer to the provided INFORM job files for the general structure).
#
# DEFAULT: false
#allow_custom_inform: false
```
</details>

ted-miller, 2023-09-28:
Hi @rsaintobert 
Could you post a copy of the INIT_ROS.JBI file that was generated? Also, a copy of the PANELBOX.LOG file from your robot would be helpful. I'm wondering if your robot has some additional control group(s) configured.

gavanderhoorn, 2023-09-28:
Slightly off-topic, but on the FS100 branch I have https://github.com/gavanderhoorn/motoros2/commit/578a6101e71a6da140fb8ba17bdea3e07b848d21 which would print the 'offending' line during the check.

I might cherry-pick that to `main` and submit a PR. Would make debugging this a bit easier perhaps.

rsaintobert, 2023-09-29:
Hi, here are the files.

<details>
<summary>INIT_ROS.JBI</summary>

```bash
/JOB
//NAME INIT_ROS
//POS
///NPOS 0,0,0,0,0,0
//INST
///DATE 2023/09/28 15:36
///COMM AUTO-GENERATED BY MotoROS2
///ATTR SC,RW
///GROUP1 RB1
NOP
DOUT OT#(890) OFF
DOUT OT#(889) OFF
TIMER T=0.050
DOUT OT#(889) ON
WAIT OT#(890)=ON
DOUT OT#(890) OFF
DOUT OT#(889) OFF
END

```
</details>

<details>
<summary>PANELBOX.LOG</summary>

```bash
/SYSTEM 9
//DATE      : 2023/09/29 09:52:57
//SYSTEM NO : YBS2.46.00A(EN/FR)-00
//PARAM  NO : 1.42/1.42
//APPLI     : GENERAL         
//LANGUAGE  :  2.46-00-00, 2.46-00-00
//REVISION
MAIN     2.46.00,1.02-10
CA      --------,16092200
PP       2.10-00, 2.00-00, 1.07
SAFETY#0 2.23-00/ 2.11-00, 2.23-00/ 2.11-00, 2.00-00, 2.00-00,17021600,17021600
SERVO#0  1.18-00/ 1.18-00, 1.02E00/ 1.02P00, 2.00-00, 1.01-00,19090800,10000002
//ROBOT NAME
R1 : 1-06VX7-A0*(GP7)                             0011_1111
//CONTROL POWER
TOTAL : 2021/06/16 13:56,00718:35'39
//SERVO POWER
TOTAL : 2021/06/16 13:56,00174:03'37
R1    : 2021/06/16 13:56,00173:35'16
//PLAYBACK TIME
TOTAL : 2021/06/16 13:56,00119:30'17
R1    : 2021/06/16 13:56,00119:29'13
//MOVING TIME
TOTAL : 2021/06/16 13:56,00000:23'44
R1    : 2021/06/16 13:56,00000:23'44
//OPERATING TIME
TOOL1 : 2021/06/16 13:56,00000:00'00
//ENERGY TIME
TOTAL : 2021/06/16 13:56,00024:33'21
//MOTOPLUS APP
01: MotoROS2 (humble)                0.1.1            micro-ROS based ROS 2 interface 
//CONTROLLER SETTING
============================================================
LANGUAGE

------------------------------------------------------------
LANGUAGE1           ENGLISH            
LANGUAGE2           FRENCH             
[... 163 lines cut ...]
  RCxG                       ----/--/-- --:--:--
  RExG                       ----/--/-- --:--:--
  ROxG                       ----/--/-- --:--:--
  RS                         ----/--/-- --:--:--
  SxE                        ----/--/-- --:--:--
  S1CxG                      ----/--/-- --:--:--
  S2C                        ----/--/-- --:--:--
  S3C                        ----/--/-- --:--:--
  S4C                        ----/--/-- --:--:--
  S1D                        ----/--/-- --:--:--
  S2D                        ----/--/-- --:--:--
  S3D                        ----/--/-- --:--:--
  S4D                        ----/--/-- --:--:--
  SVxG                       ----/--/-- --:--:--
  SVCxB                      ----/--/-- --:--:--
  SVD                        ----/--/-- --:--:--
  SVMxG                      ----/--/-- --:--:--
  SVPxG                      ----/--/-- --:--:--
  SVSxB                      ----/--/-- --:--:--

```
</details>

I rebooted the controller several time today and so far I can see the ros node, but the alarm for invalid job still appears.
I still don't know why the ros node is sometimes visible and sometimes not (boot order between agent and controller ?)

gavanderhoorn, 2023-09-29:
There is a difference here:

job MotoROS2 expects/generates:

```
TIMER T=0.05
```

job it 'sees' on your YRC1000m @rsaintobert:

```
TIMER T=0.050
```

@ted-miller: this is a similar difference to what I saw on the FS100 and I worked around/fixed with https://github.com/gavanderhoorn/motoros2/commit/6a79f9bf9fe28e09c0c2f9f3e5d45eeadd0e07e1. Would something similar be needed for the YRC1000m?

rsaintobert, 2023-09-29:
I tried to update the job manually and I confirm the YRC1000m adds a trailing 0 when entering 0.05 value.

gavanderhoorn, 2023-09-29:
Yes, that's also why I didn't suggest editing it. It wouldn't have worked.

ted-miller, 2023-09-29:
> this is a similar difference to what I saw on the FS100 and I worked around/fixed with https://github.com/gavanderhoorn/motoros2/commit/6a79f9bf9fe28e09c0c2f9f3e5d45eeadd0e07e1. Would something similar be needed for the YRC1000m?

What if we simply change `strncmp` to `strstr`?
https://github.com/Yaskawa-Global/motoros2/blob/4ab13e96ab1047fe232068d22cb2aa1d66c5f18e/src/InformCheckerAndGenerator.c#L263

gavanderhoorn, 2023-09-29:
Hm, could do that.

It wouldn't be the same though.

I personally like being specific about things.

ted-miller, 2023-09-29:
@rsaintobert 
For an immediate remedy, you can uncomment `allow_custom_inform` in the config file and set it to `true`. This will prevent the alarm from occurring. 

https://github.com/Yaskawa-Global/motoros2/blob/4ab13e96ab1047fe232068d22cb2aa1d66c5f18e/config/motoros2_config.yaml#L270

To update the config file:
https://github.com/Yaskawa-Global/motoros2#updating-the-configuration

gavanderhoorn, 2023-09-29:
Thanks for reporting the issue @rsaintobert :+1: 

This should be fixed in `0.1.2`, which will be released in October.

PR #168 (2023-09-29): Add controller-based conditional around the Inform TIMER instruction
Fixes #166 
Based on https://github.com/gavanderhoorn/motoros2/commit/6a79f9bf9fe28e09c0c2f9f3e5d45eeadd0e07e1

Some controllers use three-digits for the `T` tag. Others use two.
closed 2023-09-29 by gavanderhoorn