Yaskawa-Global/motoros2 #170: Incorrect TF broadcast for robots-on-tracks

gavanderhoorn, 2023-10-04:
After some discussion in #169, it appears the current implementation of TF broadcasting for robots on tracks is not working as it should.

Jogging a robot on a track does not update the `world -> r1/base` transform. Only robot pose changes are reflected in the `r1/base -> r1/tool0` (and `r1/base -> r1/tcp_0`) transforms.

gavanderhoorn, 2023-10-04:
somehow I didn't notice this, even though I have a robot-on-a-track here.

Most of the tools we use (also to test) don't consume the TF broadcast by MotoROS2, which could explain this.

As to a possible solution: if the Yaskawa `Base` to Yaskawa `RF` transform could be included in the `world -> r1/base` transform (on the ROS side), that would seem to fix this issue.

gavanderhoorn, 2023-10-04:
Note that the FK performed by the `robot_state_publisher` (together with a correct URDF (including the robot and the track)) *is* correct, as it's based on joint angles.

ted-miller, 2023-10-04:
Now I'm _really_ confused. Now I think that it should actually work. 

Turns out that the `offsetFromBaseToRobotOrigin` is taken into account.
https://github.com/Yaskawa-Global/motoros2/blob/62ac8edda96ba0051fae57e7891aa3edaa7231d1/src/PositionMonitor.c#L306-L313

I'll take a detailed look at this later today.

gavanderhoorn, 2023-10-04:
Maybe something here is not happy with my setup?

https://github.com/Yaskawa-Global/motoros2/blob/62ac8edda96ba0051fae57e7891aa3edaa7231d1/src/CtrlGroup.c#L88-L109

ted-miller, 2023-10-05:
@gavanderhoorn, can you post a copy of your CMOS.BIN?

ted-miller, 2023-10-05:
> can you post a copy of your CMOS.BIN?

Nevermind

PR #172 (2023-10-07): Correct `/tf` calculation for base tracks
Fixes #170 

Two problems:
1) The wrong parameters were being read in the lib
2) Scaling was wrong on the coordinates. For example, `1.234 m` was getting converted to `1 um`.
closed 2023-10-07 by gavanderhoorn