Yaskawa-Global/motoros2 #174: Alarm `4229[40]` (`ETHERNET PROCESS ERROR`) with MR2 installed, but LAN not connected

gavanderhoorn, 2023-10-08:
This is a report *by proxy*, as I've only found this by reading [Multi-Robot Control with MotoROS2](https://ntnuopen.ntnu.no/ntnu-xmlui/handle/11250/3093888), an MSc thesis report.

According to the author of that report, the YRC1 used in his/her experiments raises `4229[40]` (`ETHERNET PROCESS ERROR`) in case MotoROS2 is installed but none of the network ports have an active connection (ie: have a plugged in Ethernet cable). Only after connecting a cable to the LAN port (I guess either, but there are no details in the report about this) does the alarm stop getting posted.

This is described in Section `8.7`, with the error shown in Figure `8.7`.

Note that this was with MotoROS2 `beta1` verion `0.0.15`, so quite some time ago.

I do believe this is something to check though and make sure is no longer a problem with MotoROS2.

Unfortunately no details about controller sw are present in the report, other than it's a 4 group system (2xGP25, TSL 600 and an MT1) with a Fronius TP400i and a MagicWave 3000.

gavanderhoorn, 2023-10-08:
From the alarm documentation I'm getting the impression this may have something to do with EtherNet/IP. That might also explain why the author reported that "functions like jogging" didn't work when that alarm was raised (safety might be configured to go over EtherNet/IP on that setup).

Perhaps @tingelst can add something: it's "his" setup and the author of the report was his MSc student.

tingelst, 2023-10-08:
@gavanderhoorn, I can confirm that this is still an issue with the latest release installed.

I'll have my current master's students do some debugging and report back.

gavanderhoorn, 2023-10-09:
Thanks @tingelst.

Could you please ask them to post a comment with some information about the controller used?

Software versions, how are the two Fronius devices are connected, etc?

Unfortunately we can't ask for the debug log, as that would require a working LAN connection.

gavanderhoorn, 2023-10-09:
@tingelst: I've updated the OP with the information from the manual.

From the possible cause description this sounds like a regular alarm where MotoROS2 would not affect anything (as in: if DHCP is supposed to be used, but there is no ethernet connection, it's not going to work and the controller is "correct" to complain).

It'd be valuable to know what happens without MR2 installed on the controller in case there is no connection to any of the LAN ports (although the alarm sub code description specifically mentions `LAN 2`).

gavanderhoorn, 2023-10-11:
Friendly ping @tingelst?

tingelst, 2023-10-11:
@gavanderhoorn We've been out of the lab this week. Will report back, Monday/Tuesday upcoming week.

tingelst, 2023-10-17:
> From the possible cause description this sounds like a regular alarm where MotoROS2 would not affect anything (as in: if DHCP is supposed to be used, but there is no ethernet connection, it's not going to work and the controller is "correct" to complain).
> 
I suspect you are right here. LAN2 is currently set to DHCP. Will test if changing that to a static IP addresses the error.

gavanderhoorn, 2023-10-17:
Would perhaps also be nice to know whether the same alarm is raised in case MotoROS2 is *not* installed, DHCP is still enabled for `LAN2` and there is no network connection.

gavanderhoorn, 2023-10-24:
Friendly ping again @tingelst :)

tingelst, 2023-10-26:
I was able to remove the alarm by setting LAN2 to manual and not DHCP on the said system. 

On a second cell without MotoROS2 installed, I set LAN2 to DHCP without network connection. This did not raise any alarm.

gavanderhoorn, 2023-10-26:
Thanks.

> On a second cell without MotoROS2 installed, I set LAN2 to DHCP without network connection. This did not raise any alarm.

seems there is something we're doing with MotoROS2 which makes the OS raise this specific alarm.

We'll need to investigate.

Could you perhaps attach a copy of the `PANELBOX.LOG` here of the controller which raised the alarm? If you feel there's any sensitive information in it (it's a plain text file), feel free to redact that, but be careful not to remove/alter important parts.

ted-miller, 2023-10-26:
> seems there is something we're doing with MotoROS2 which makes the OS raise this specific alarm

Likely the fact that MR2 is trying to access network features when there's no address.

tingelst, 2023-10-27:
> Could you perhaps attach a copy of the `PANELBOX.LOG` here of the controller which raised the alarm? If you feel there's any sensitive information in it (it's a plain text file), feel free to redact that, but be careful not to remove/alter important parts.

Here it is: [PANELBOX.LOG](https://github.com/Yaskawa-Global/motoros2/files/13187739/PANELBOX.LOG)