Yaskawa-Global/motoros2 #177: MH900 (and possibly GP4) axis configuration won't work

ted-miller, 2023-10-10:
So I just discovered that the axis configuration on an MH900 is really weird, due to the extra-large power converters on the lower three axes.

```
============================================================
CONNECT(STO)                                        
        -- AXIS --  -- BRK --   -- CV  --           
    SV <123456789> <123456789> <123456789> ON_EN  OT
------------------------------------------------------------
R1 :#1 [456---123] [123456---] [111---234] ON_EN1 OT1   
============================================================
```

1) The axis-order is not one that will be recognized by [Ros_CtrlGroup_ConvertSequentialJointOrderToMotoJointOrder](https://github.com/Yaskawa-Global/motoros2/blob/f6f3f2506167839deb373b8882ae6e9c7ca4072c/src/CtrlGroup.c#L635)
2) The MotoPlus API can't control the 9th axis in a single control group.

This could potentially be re-wired on the servo board to a supported configuration. But the local engineer in charge of MH900 wasn't immediately sure about the implications. More investigation would be required.

-----------------------------
And while discussing this issue, it was mentioned that the GP4 also has a really weird axis configuration. I don't have details beyond "really weird".

gavanderhoorn, 2023-10-10:
> The MotoPlus API can't control the 9th axis in a single control group.

Should we not assert on this model until this has been fixed?

Just to prevent late user-facing errors?

And document the fact that model is unsupported.

ted-miller, 2023-10-10:
I think that's a good idea. But we don't currently have true robot-model-detection. We just get the axis type (linear/rotational) and then make some assumptions based on number of axes.

Having said that, it's possible.

ted-miller, 2023-10-10:
GP4:

I don't see anything weird here.
[image]

Perhaps due to the fact this is from MotoSim and not a real arm. But given that https://github.com/ros-industrial/motoman/discussions/544 got everything working, I'm guessing it's fine.

gavanderhoorn, 2023-10-10:
> But we don't currently have true robot-model-detection

any M+ APIs we could use?

ted-miller, 2023-10-10:
Here's a couple HSES commands... I have to imagine there are M+ equivalents. 

[GetAxisConfiguration.pdf](https://github.com/Yaskawa-Global/motoros2/files/12861595/GetAxisConfiguration.pdf)

[SystemInformation.pdf](https://github.com/Yaskawa-Global/motoros2/files/12861596/SystemInformation.pdf)

ted-miller, 2023-10-11:
I suppose we could always just parse the PANELBOX.LOG file.
It's not ideal, but it wouldn't be too difficult.

ted-miller, 2023-10-11:
Or we could invoke the HSES command. It's easy enough to hard-code a binary packet (as opposed to implementing an entire interface).

gavanderhoorn, 2023-10-11:
Are we mostly worried about the active axis in the 9th slot, or about the unusual order?

ted-miller, 2023-10-11:
Both equally.

We need to detect unusual order for the sake of [Ros_CtrlGroup_ConvertSequentialJointOrderToMotoJointOrder](https://github.com/Yaskawa-Global/motoros2/blob/f6f3f2506167839deb373b8882ae6e9c7ca4072c/src/CtrlGroup.c#L635). Otherwise we won't process an incoming trajectory properly.

We need to detect the 9th axis to raise a fatal alarm, since we can't control the U axis in that configuration.