Yaskawa-Global/motoros2 #178: Unexpected `8011[63]` (`Invalid motion type`) (YRC, GP25, linear track)

gavanderhoorn, 2023-10-12:
It looks like #176 works a little too well, in that it immediately is raised on a YRC1, GP25 on a linear track -- which is a simple system that should-just-work (tm).

Have to investigate.

gavanderhoorn, 2023-10-12:
@ted-miller: is `MOTION_TYPE_NOT_USED` returned for axes that are not present in motion groups?

ted-miller, 2023-10-12:
Ahhh... Yeah, I merged w/o testing. Foolish.

> is MOTION_TYPE_NOT_USED returned for axes that are not present in motion groups?

Yes. There's 8 axes in EVERY group. Most are not used.

ted-miller, 2023-10-12:
> Should this just only look at axes which are actually configured:

Yes. But it's not terribly straight-forward. The current control-group doesn't have knowledge of the base-group's axis configuration.

So it'd be something like:
`g_Ros_Controller.ctrlGroups[group->baseTrackGroupIndex].axisType[i] != AXIS_INVALID`

This seems like an awful lot of clutter for something that _shouldn't_ happen and AFAICT _cannot_ happen. 

I'd be in favor of just adding a `default: break;` to the switch and removing the alarm. 
But I'll leave it up to you.

gavanderhoorn, 2023-10-12:
See #179.

PR #179 (2023-10-13): Correct check for 'invalid motion type' (issue 178)
Should fix #178 -- if I interpreted your https://github.com/Yaskawa-Global/motoros2/issues/178#issuecomment-1759760398 correctly @ted-miller.

Please do test this time :)

I'm not near a controller so can't test it myself, but I did want to fix the mistake I introduced in #176.
closed 2023-10-13 by ted-miller