Yaskawa-Global/motoros2 #199: Use `mpGetFBPulsePosEx` instead of `mpGetFBPulsePos`

ted-miller, 2023-12-19:
`mpGetFBPulsePos` returns raw feedback position of the encoders. This can be different than logical position of the joint. This is due to things such as [cross-axis couplings](https://github.com/Yaskawa-Global/motoros2/blob/9026a1d0b50b483f7c334bfd5ebbac8b328bfdb6/src/CtrlGroup.c#L345-L358) or gravity-compensation.

We already implement compensation for the cross-axis thing. And I'm told that gravity-compensation is a very specialized feature that's only available on one or two arms.

Now I'm hearing that there is a new feature on the next-gen controller that could also be affected by this.

The simple fix is to use mpGetFBPulsePos**Ex**, which should apply any required corrections to the output.

gavanderhoorn, 2023-12-20:
So would that allow us to remove the code we currently have for things like that cross-axis coupling you link to?

ted-miller, 2024-01-30:
> So would that allow us to remove the code we currently have for things like that cross-axis coupling you link to?

Yes, we could remove that chunk of code.

> Is there also an mpGetServoSpeedEx(..)?

Not that I'm aware of.

-------------------------

I'm seeing some contradictory information on `mpGetFBPulsePosEx` regarding the `absolute accuracy` feature of the next-gen controller. I don't yet have the ability to test myself.

**One report**:

_It looks that the axes correction is applied just after calling mpGetFBPulsePos(), but if other correcting features are enabled, the differences will be appered. For example, gravitiy compensation (this is for special customer) or absolute accuracy (this is only YNX1000 currently). 
I suggest to use mpGetFBPulsePosEx() instead of mpGetFBPulsePos(),_ 

**Another**:
(apologies for the screenshot)

[image]

gavanderhoorn, 2024-01-30:
Hm. So in any case this reads as-if we'd have to introduce another batch of controller-specific `#ifdef`s. Correct?

ted-miller, 2024-01-30:
Yeah, sounds that way

ted-miller, 2024-11-25:
At this time, we don't intend to build MotoROS2 for the YNX1000.

closed 2024-11-25 by ted-miller

gavanderhoorn, 2024-11-25:
Do the other benefits not make this interesting in any case, also for other controllers?

This fi:

> > So would that allow us to remove the code we currently have for things like that cross-axis coupling you link to?
> 
> Yes, we could remove that chunk of code.

?

ted-miller, 2024-11-25:
We already have code to handle all known cases.

The idea of transitioning to the `Ex` function was to handle a new feature. But since that feature isn't applicable to our existing controller generation, then I'd rather not risk breaking any existing logic.

We could make an argument for future-proofing the code. But I don't really think it's worth it.

gavanderhoorn, 2024-11-25:
Should we perhaps add a comment to the implementation in MotoROS2 that explains why we're not using `mpGetFBPulsePosEx(..)`? Could avoid discussing it again in the future.

re: we already handle all known cases: perhaps true, but don't we do things that complicate using features such as absolute accuracy?