Yaskawa-Global/motoros2 #203: Cycle-switch-mode during setup procedure

ted-miller, 2024-01-23:
Re: https://github.com/Yaskawa-Global/motoros2/discussions/201#discussioncomment-8224715

Need to add this to setup procedure. Perhaps we can also detect this in real time via CIO.

(I vaguely recall this being an issue in the past. But I can't find anything about it in the repos.)

ted-miller, 2024-01-23:
Looks like we can detect with these:

[image]

gavanderhoorn, 2024-03-15:
@ted-miller: did we forget to add something to the setup documentation about this?

ted-miller, 2024-03-18:
It looks like I added to the motoros1 wiki, but didn't do it here.

ted-miller, 2024-03-26:
It just occurred to me that there is an API for mpSetCycle. 

@yai-rosejo @gavanderhoorn 

I vote that instead of documenting the step of setting cycle, we should automate it. Right after we enable servos, we can set cycle and then start the INIT_ROS job. 

Thoughts?

gavanderhoorn, 2024-03-26:
Hm. Changing 'global' controller configuration for a single application in an automagic way?

Not sure.

ted-miller, 2024-03-26:
It's not global. It's instantaneous.

When the user switches the key-switch, the robot's "saved" configuration is restored.

(At least I _think_ that's true. Need to verify.)

gavanderhoorn, 2024-03-26:
If that's true (ie: it reverts) automating it seems like it would improve UX.

related btw: I've never had my YRC1 configured with continuous and never really ran into any issues. Was I just lucky?

PR #212 (2024-02-21): Issue 203 cycle switch mode
Fixes #203 
Added in a CIO read to check for continuous cycle mode being set
closed 2024-06-17 by ted-miller