Yaskawa-Global/motoros2 #206: Problem with starting a trajectory when "Absolute Accuracy Compensation" is valid

yeu-geissdoerfer, 2024-01-30:
We had a case where we set the *Absolute Accuracy Compensation* to valid on a YRC1000, using controller SW version `YAS 5.2`. Once we had taken all the necessary steps to enable the *Absolute Accuracy Compensation Function*, we were able to connect the microROS agent with the MotoROS2 driver. 

To test the accuracy of the motion, we used the last published position of the JointState.msg as the start position of our trajectory and defined a desired joint state as a goal. After the trajectory was sent to the driver, it immediately rejected it and sent the following Error Message
```
[2024-01-29 14:47:07.192195] [192.168.0.2:57915]: Ros_ServiceStartTrajMode_Trigger: activated
[2024-01-29 14:47:12.451865] [192.168.0.2:57915]: Trajectory contains 2 points
[2024-01-29 14:47:12.452465] [192.168.0.2:57915]: Initializing trajectory for group #0
[2024-01-29 14:47:12.452665] [192.168.0.2:57915]: ERROR: Trajectory start position doesn't match current position (MOTO joint order).
[2024-01-29 14:47:12.452665] [192.168.0.2:57915]:  - Requested start: -264, -1030, 121, -72, -135, -63, 0, 0
[2024-01-29 14:47:12.452665] [192.168.0.2:57915]:  - Current pos: 0, 0, 0, 1, 0, 1, 0, 0
[2024-01-29 14:47:12.452665] [192.168.0.2:57915]:  - ctrlGroup->prevPulsePos: 0, 0, 0, 1, 0, 1, 0, 0
[2024-01-29 14:47:12.452865] [192.168.0.2:57915]: FollowJointTrajectory - Goal request rejected
[2024-01-29 14:47:12.452865] [192.168.0.2:57915]: The trajectory will be accepted and then immediately aborted
[2024-01-29 14:47:12.453065] [192.168.0.2:57915]: The first point must match the robot's current position.
```

The failure is due to the fact, that the **first point doesn't match the robot's current position** which is checked in 
https://github.com/Yaskawa-Global/motoros2/blob/b71ced5ff7aba53184c24786056417df5021f082/src/MotionControl.c#L194-L212

Looking deeper into the code shows that there are two different MotoPlus functions used to determine the robots position

- `pulsePos` uses `mpGetFBPulsePose()`  (FB stands for feedback) -> current **actual** position
https://github.com/Yaskawa-Global/motoros2/blob/b71ced5ff7aba53184c24786056417df5021f082/src/CtrlGroup.c#L337-L338
- `curPos` uses `mpGetPulsePose()` -> current **commanded** position
https://github.com/Yaskawa-Global/motoros2/blob/b71ced5ff7aba53184c24786056417df5021f082/src/MotionControl.c#L189

When we look into the "YRC1000 Options Instructions for Absolute Accuracy Compensation Function" the following is stated there

## Absolute Accuracy Compensation Function
The absolute accuracy compensation function positions the TCP with high accuracy to a position specified in Cartesian coordinates.

### Restrictions
When absolute accuracy compensation function is set to valid, only the reference value is available in the robot current position output function. **The feedback value cannot be used!**

This means that the MotoPlus function `mpGetFBPulsePose()` doesn't take the absolute accuracy compensation correction into account!

The functions `Ros_PositionMonitor_UpdateLocation()` and `Ros_Controller_IsInMotion()` are also affected, since they also use `mpGetFBPulsePose()`.

## Question

Now the question arrises whether the `mpGetFBPulsePose()` function can be replaced by `mpGetPulsePose()` plus checking if the robot is idle?

gavanderhoorn, 2024-01-30:
I'm going to mention #199, as it mentions the same/similar functions.

gavanderhoorn, 2024-01-31:
@ted-miller: while there doesn't appear to be a direct *advantage* to using absolute accuracy together with MotoROS2, as we indeed currently don't have any Cartesian interfaces, nor do we use Cartesian motion primitives while executing motion, could we perhaps find a way to make MotoROS2 compatible with it?

As mentioned, we can't just wholesale switch to `mpGetPulsePose(..)`, as that would break functionality in MotoROS2 elsewhere.

@yeu-geissdoerfer: are you mostly interested in just being able to use MotoROS2 on a YRC1000 with Absolute Accuracy enabled -- but not necessarily absolute accurate motions, or do you really want/need to also have absolute accurate motions with MotoROS2?

The former could *potentially* be possible. The latter probably wouldn't be.

yeu-geissdoerfer, 2024-01-31:
I would be interested in **absolute accurate motions with MotoROS2** as this seems to be *particularly important for the HC series*.

gavanderhoorn, 2024-01-31:
> I would be interested in **absolute accurate motions with MotoROS2** 

Unless I've misunderstood, the absolute accurate motions will only improve Cartesian motions. MotoROS2 does not support those.

> as this seems to be _particularly important for the HC series_.

Your OP mentions a YRC1000 controller. Should that be a 1000 micro then?

yeu-geissdoerfer, 2024-01-31:
It is a YRC1000 controller, no micro.

yeu-geissdoerfer, 2024-01-31:
@gavanderhoorn what would be a scenario for you in which it is necessary to move the robot with MotoROS2 and Abs Acc Compensation activated?

gavanderhoorn, 2024-01-31:
I can think of two:

1. a hybrid application where MotoROS2 is used for non-process motions (fi: to get an EEF into position using collision free motion planning) and then starting an INFORM job to perform process motion (fi: welding, blending, etc)
2. a robot which is used both with ROS and with regular INFORM jobs, but not at the same time

basically it would include all motions performed by MotoROS2 which do not make use of / can't benefit from abs acc.

Which would be all of its motion, IIUC. We're only using pulse based joint motion at this time.

gavanderhoorn, 2024-01-31:
Created #207 to track the need to document compatibility with abs acc compensation.

gavanderhoorn, 2024-02-01:
@yeu-geissdoerfer: just making sure: would you still like to see a work-around / solution to the issue you raised, or have the previous comments clarified things sufficiently?

yeu-geissdoerfer, 2024-02-01:
I have the following question:

If our robots want to use Absolute Accuracy Compensation, they must first go through a special measurement procedure. This determines the actual DH parameters.

1. Could these *actual DH parameters* be used to update the existing URDF file of this robot, to compensate the geometrical error while planning the motion?
2. How could such a URDF file be updated automatically?

It would then be possible to compensate the geometric error without enabling the Absolute Accuracy Compensation on the robot controller.

gavanderhoorn, 2024-02-01:
While I'm not sure whether this is actually an X-Y problem: yes, that could very likely be done.

See the way [ur_calibration](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/tree/master/ur_calibration) does this, by converting their D-H parameters to a format compatible with URDF frames. These values can be stored in a `.yaml` file and then loaded at runtime to update a 'template' URDF.

ted-miller, 2024-02-02:
> Which information is currently possible to extract from the robot using MotoSDK

We do allow for extracting DH parameters from the robot.
https://github.com/Yaskawa-Global/motoros2/blob/10d8f00317c5d4ef7960627962337bf72e2ba817/src/CmosParameterExtraction.h#L205

But the DH parameters for the robot never get updated after a calibration. Normally, we only modify the encoder's zero-position when calibrating. 

> Is it possible to extract information of calibrated kinematics from Absolute Accuracy

I don't know enough about this function yet. It is very new. 

We'll need to investigate more.

gavanderhoorn, 2024-02-03:
Could I suggest we discuss calibrated URDFs and DH parameters in #210?

It's all related to Abs Acc Compensation, but at the same time we're starting to go off-topic.

gavanderhoorn, 2024-02-05:
@yeu-geissdoerfer: are you assisting @iydv with their request to make motions more accurate, and is that the context in which the question about abs acc compensation came up?

gavanderhoorn, 2024-02-06:
> We are now investigating the calibrated URDFs as this seems to be the most promising solution at the moment

could we discuss approaches in #210? I don't believe there are any standard or best practices at the moment, and this seems like a good opportunity to design something for Yaskawa support packages.

closed 2024-04-12 by yeu-geissdoerfer