Yaskawa-Global/motoros2 #221: `Must call queue_traj_point service` returned when motion mode is not 'streaming'

gavanderhoorn, 2024-03-09:
Context: gavanderhoorn/motoros2_fjt_pt_queue_proxy#2.

I would have expected the service server to return:

> Must call start_point_queue_mode

instead of mentioning `queue_traj_point`.

yai-rosejo, 2024-03-20:
Believe this message is coming from: https://github.com/Yaskawa-Global/motoros2/blob/5f8c80011c64dc9152a4c72b2eb138760a9d649e/src/ServiceQueueTrajPoint.c#L72-L86
So motoros2_interfaces__msg__QueueResultEnum__WRONG_MODE_STR will need to be updated from "Must call queue_traj_point service" to "Must call start_point_traj_point service"

ted-miller, 2024-03-25:
Closing. This issue is resolved with https://github.com/Yaskawa-Global/motoros2_interfaces/pull/15.

Note that source-builds will still use the wrong message until the next `motoros2_interfaces` is released.