Yaskawa-Global/motoros2 #250: Improve FJT goal validation (include `JointTolerance` checks fi)

gavanderhoorn, 2024-05-01:
As discussed in https://github.com/Yaskawa-Global/motoros2/pull/241#issuecomment-2079263653, it would be good to add more extensive `FollowJointTrajectoryAction` goal validation to the FJT action server.

The current implementation is too trustworthy and should be changed into a more defensive version.

jimmy-mcelwain, 2025-04-22:
Is this for `path_tolerance`, `goal_tolerance`, or both? I just ask because it would be easy to add a `goal_tolerance` check after the motion (we already have something like that), but I'm not sure exactly what a `path_tolerance` check would look like. Specifically, at what times/points would we check, and what we would check against? My understanding is that `path_tolerance` exists so the robot will abort the movement if it strays too far from the intended path in space. But like we talked about in  #405 and I mentioned in #415, the `error` in position at any given time is dominated by latency, not tracking error. Properly measuring tracking error while ignoring latency would require more complexity/ambiguity/processing time and maybe memory. So it seems like we could either go the straightforward route and just check `path_tolerance` at the `time_from_start` of each point, or we could go with something more complex that corrects for the error introduced by latency.

gavanderhoorn, 2025-04-23:
While I would like to add something like path tolerance support, the https://github.com/Yaskawa-Global/motoros2/pull/241#issuecomment-2079263653 linked in the OP is mostly about changing the order of operations in the current implementation:

> I'm beginning to wonder whether we shouldn't already parse the `JointTolerance` instances when first accepting the goal. If parsing fails -- due to unknown joints being referenced for instance -- we could refuse to execute the goal and provide a very clear error code and message to clients.
> 
> If we fail to parse at the end of an executed trajectory, we can't really do much anymore: failing the goal completely seems like a bit of a 'waste' of a good trajectory execution and failing so late for something almost 'unrelated' doesn't apply fail-early and least-surprise principles.

See also @ted-miller's follow-up comment.