Yaskawa-Global/motoros2 #259: Config file in releases outputs errors when processed through recommended linter

jimmy-mcelwain, 2024-06-10:
I was going through the setup instructions for motoros2 on my machine. I downloaded and unzipped the file [motoros2_0.1.2_yrc1000_humble.zip](https://github.com/Yaskawa-Global/motoros2/releases/download/0.1.2/motoros2_0.1.2_yrc1000_humble.zip) from the [releases page](https://github.com/yaskawa-global/motoros2/releases). When I ran the linter as suggested in the setup instructions, I got the following output. 

```
(venv_yamllint) jimmy@jimmy-ThinkPad-T580:~/Desktop/motoros2/config$ yamllint -s . && echo "all ok"
./motoros2_config.yaml
  1:4       error    wrong new line character: expected \n  (new-lines)
  8:2       warning  missing starting space in comment  (comments)
  14:2      warning  missing starting space in comment  (comments)
  15:2      warning  missing starting space in comment  (comments)
  21:2      warning  missing starting space in comment  (comments)
  28:2      warning  missing starting space in comment  (comments)
  30:2      warning  missing starting space in comment  (comments)
  34:2      warning  missing starting space in comment  (comments)
  36:2      warning  missing starting space in comment  (comments)
  40:2      warning  missing starting space in comment  (comments)
  42:2      warning  missing starting space in comment  (comments)
  63:2      warning  missing starting space in comment  (comments)
  65:2      warning  missing starting space in comment  (comments)
  73:2      warning  missing starting space in comment  (comments)
  88:2      warning  missing starting space in comment  (comments)
  90:2      warning  missing starting space in comment  (comments)
  124:2     warning  missing starting space in comment  (comments)
  126:2     warning  missing starting space in comment  (comments)
  133:2     warning  missing starting space in comment  (comments)
  140:2     warning  missing starting space in comment  (comments)
  151:2     warning  missing starting space in comment  (comments)
  163:2     warning  missing starting space in comment  (comments)
  165:2     warning  missing starting space in comment  (comments)
  173:81    error    line too long (81 > 80 characters)  (line-length)
  181:2     warning  missing starting space in comment  (comments)
  182:2     warning  missing starting space in comment  (comments)
  182:81    error    line too long (105 > 80 characters)  (line-length)
  184:2     warning  missing starting space in comment  (comments)
  197:2     warning  missing starting space in comment  (comments)
  203:81    error    line too long (88 > 80 characters)  (line-length)
  205:81    error    line too long (84 > 80 characters)  (line-length)
  225:2     warning  missing starting space in comment  (comments)
  230:81    error    line too long (81 > 80 characters)  (line-length)
  242:2     warning  missing starting space in comment  (comments)
  254:2     warning  missing starting space in comment  (comments)
  256:2     warning  missing starting space in comment  (comments)
  270:2     warning  missing starting space in comment  (comments)
```

This is without any modifications to the yaml file. I think that the file should probably be modified.

gavanderhoorn, 2024-06-11:
Those `.zip`s don't appear to contain the [yamllint config file](https://github.com/Yaskawa-Global/motoros2/blob/7df9e48883facde92073ce8abe2d33b4ff8f9a24/config/.yamllint).

That's most likely why `yamllint` isn't happy.

The [previous release](https://github.com/Yaskawa-Global/motoros2/releases/download/0.1.1/motoros2_0.1.1_yrc1000_humble.zip) did ship with that file included. Perhaps @ted-miller forgot to add it to the `.zip`?

ted-miller, 2024-06-11:
The release zips have been updated.

closed 2024-06-11 by ted-miller