Yaskawa-Global/motoros2 #263: Persistent ALARM 8014  Invalid MotoROS2 job detected [1]

luke-reframe, 2024-06-17:
Upon initiating the microros Docker client, there will always be an alarm 8014 [1]. This has been preventing proper communication with the FollowJointTrajectory Action Server though published topics from the bot are still functional. I have attempted to clear this first by removing all job files .DAT/.JBI from the connected USB (to remove any invalid jobs) and then again by setting allow_custom_inform to true.

In both cases, the Invalid MotoRos2 job is detected. At this point the controller is preventing new motoros2_config.yaml files to be loaded even with forced override. ROS_INIT.JBI file is not recreated despite controller restart attempts. Before deletion, I compared the ROS_INIT.JBI file to the examples in the repo and noted the only difference being the timer period being .5 seconds instead of .05.

Controller is a YRC1000micro and the robot is the yaskawa HC10DT. MotoRos2 Installation has been verified through the md5sum procedure. Version is Humble 0.1.2.

Debug log here
[20240617T160939_motoros2_debug_log.txt](https://github.com/user-attachments/files/15876679/20240617T160939_motoros2_debug_log.txt)

Config Yaml (txt format as yaml is not supported)
[motoros2configtxt.txt](https://github.com/user-attachments/files/15876832/motoros2configtxt.txt)

Controller Version Information:
SYSTEM: YBS4.02.00A(JP/EN)-00
PARAMETER: 1.69/1.69
MODEL: 1-06VXHC10-A1*(HC10DT)
APPLI: General

ted-miller, 2024-06-17:
Hi @luke-reframe 
This was fixed in https://github.com/Yaskawa-Global/motoros2/pull/168 and is included in the `0.1.2` release. Please update to the latest version.

If that release doesn't resolve the issue, please reopen this issue and include the full debug output, including the top part where the configuration data is parsed. (You'll need to start the python logger before booting the YRC.)

As a short-term fix, you can modify line 3 of the ROS_INIT job on the teach pendant.

`Ros_InformChecker_CheckInformLines: mismatch: line 3, expected: 'TIMER T=0.050', got: 'TIMER T=0.500'`

Change the TIMER to `T=0.050`.

ted-miller, 2024-06-17:
I just noticed that you did say you're using 0.1.2.

Please post the full output of the debug log and also the job which was generated.

luke-reframe, 2024-06-17:
Hi @ted-miller 

Here is the generated Job
[INIT_ROSJBIGenerated.txt](https://github.com/user-attachments/files/15877338/INIT_ROSJBIGenerated.txt)

Here is the full debug log
[20240617T170553_motoros2_debug_log.txt](https://github.com/user-attachments/files/15877341/20240617T170553_motoros2_debug_log.txt)

Thank you