Yaskawa-Global/motoros2 #268: Extend `motoRosAssert(..)` to output expected value

gavanderhoorn, 2024-06-24:
[Current implementation](https://github.com/Yaskawa-Global/motoros2/blob/d8b75ddcfe763db44f6f13ebc66c28469d154acd/src/ErrorHandling.c#L79-L82) posts an alarm and enters an infinite loop printing `motoRosAssert: MotoROS2: Fatal Error (subcode: N)`.

While that works, it would be more helpful to us (and users) if it printed the expected value (whatever that would mean for specific `motoRosAssert(..)` invocations).

Example:

https://github.com/Yaskawa-Global/motoros2/blob/d8b75ddcfe763db44f6f13ebc66c28469d154acd/src/PositionMonitor.c#L55-L62

an extended version of `motoRosAssert(..)` would output the values of `ret` and `RCL_RET_OK` as part of the message.

gavanderhoorn, 2024-06-24:
Context: #266 (among others).

jimmy-mcelwain, 2024-09-03:
What would the scope of this issue be? The conditions tested by `motoRosAssert` have a wide variety of operators. An initial search shows ==, !=, >=, <=, non-NULL, etc. are used. It could also cover several different types. Would we want a function for each of these checks? The implementation wouldn't be difficult, but I don't know where the best place to stop would be.

gavanderhoorn, 2024-09-03:
We should be able to use [macro stringification](https://gcc.gnu.org/onlinedocs/cpp/Stringizing.html) to avoid having to create `N` different versions of `motoRosAssert(..)`.

There are various C / C++ unit testing frameworks which use this approach to print more informative result/error messages.

In the end `motoRosAssert(1 < x, ...)` would result in the `1 < x` being available as a string itself, in addition to being a conditional. That string-ified version could then used as part of the message.

jimmy-mcelwain, 2024-09-30:
I have some work related to this [here](https://github.com/Yaskawa-Global/motoros2/tree/OutputExpectedValueOnAssertion). I am using macro stringification to print out variable names and values. The macros make it a bit ugly, but it adds a lot of information to the debug log without making calls to the assertion functions more difficult. I believe that I could get rid of the `motoRos_complete_ASSERT_XX_XXX` functions and just define everything with the macros, but I am not very familiar with the convention there. There's some code duplication that I could try to cut down on too. I haven't tested it very much, I just ran it once and it seemed to work. But is this the sort of thing that you are looking for @gavanderhoorn?

ted-miller, 2024-10-01:
Two points:

1) I really struggle to understand some of those naming conventions. My 'knee-jerk' reaction would be to make them more verbose. But they're already fairly long. So we're definitely going to need some form of 'summary block' for internal documentation:

```
/// <summary>
/// 
/// </summary>
/// <param name="subCodeIfFalse"></param>
```

2) Let me start by saying that I am not skilled with complex C macros or the 'stringification' (awesome word, btw). But I don't think [this](https://github.com/Yaskawa-Global/motoros2/blob/8148234024d75efcdf1fbdee06f02a920c951510/src/ErrorHandling.h#L227-L246) is the way to go. AFAICT, there is a one-to-one passthrough of the macro command with the function definition. If that's true, then the macros aren't actually saving you anything. It's just another layer. 

I believe that either the `rcl` or `rmw` layers offer some examples of some really gnarly macros that basically implement template-classes. I'll see if I can find some stuff and we can discuss together.

jimmy-mcelwain, 2024-10-01:
What the macros offer is the ability to read in the variable/expression names that are being tested without having to manually enter that information in as a string. E.g. if I call

```
motoRos_ASSERT_GE_INT_MESSAGE(g_nodeConfigSettings.ros_domain_id, 
    MIN_ROS_DOMAIN_ID_LINUX,
    SUBCODE_CONFIGURATION_INVALID_DOMAIN_ID,
    "Domain ID (%d) invalid",
    g_nodeConfigSettings.ros_domain_id);
```
If it fails, then as part of the print out, it will say 

```
Actual: g_nodeConfigSettings.ros_domain_id == -1
Expected: g_nodeConfigSettings.ros_domain_id >= 0 (MIN_ROS_DOMAIN_ID_LINUX)
```

whereas the non-macro version would say something along the lines of 

```
Actual: -1
Expected: value greater than or equal to 0
```

I can get rid of the "original" functions and just define it all as a macro, I just didn't do that because I already had the functions written when I decided to try the stringification.

ted-miller, 2024-10-01:
Ah. I definitely missed that fact

jimmy-mcelwain, 2024-10-01:
I did change the names to be more verbose, and changed some other naming conventions. If some of the names are too long, I can change them back