Yaskawa-Global/motoros2 #270: Tests_ActionServer_FJT.c causes compilation to fail if MOTOROS2_TESTING_ENABLE is defined

jimmy-mcelwain, 2024-06-26:
When I try to compile with `MOTOROS2_TESTING_ENABLE`, I get the following output

```
...
1>Compiling Tests_ActionServer_FJT.c
1>Tests_ActionServer_FJT.c : warning : In function 'Ros_Testing_Ros_ActionServer_FJT_Parse_GoalPosTolerances_null_args'
1>Tests_ActionServer_FJT.c(25): warning : implicit declaration of function 'Ros_ActionServer_FJT_Parse_GoalPosTolerances'
1>Tests_ActionServer_FJT.c : warning : In function 'Ros_Testing_Ros_ActionServer_FJT_Reorder_TrajPt_To_Internal_Order_null_args'
1>Tests_ActionServer_FJT.c(383): warning : implicit declaration of function 'Ros_ActionServer_FJT_Reorder_TrajPt_To_Internal_Order'
1>Compiling Tests_ControllerStatusIO.c
1>Compiling Tests_CtrlGroup.c
1>Compiling Tests_RosMotoPlusConversionUtils.c
1>Compiling Tests_TestUtils.c
1>Building mr2_yrc1_h.out
1>Tests_ActionServer_FJT.obj : warning : In function `Ros_Testing_ActionServer_FJT'
1>(.text+0x2a86) : warning : Tests_ActionServer_FJT.c (.text+0x2a86)
1>ActionServer_FJT.obj : warning : ActionServer_FJT.c
1>ld returned 1 exit status : warning : collect2  ld returned 1 exit status
1>mpBuilder : error : Problem with mpBuilder - Failed during build process - Build Failed
...

```

The problem appears to be that [`Ros_ActionServer_FJT_Parse_GoalPosTolerances`](https://github.com/Yaskawa-Global/motoros2/blob/d8b75ddcfe763db44f6f13ebc66c28469d154acd/src/ActionServer_FJT.c#L485-L488) and [`Ros_ActionServer_FJT_Reorder_TrajPt_To_Internal_Order`](https://github.com/Yaskawa-Global/motoros2/blob/d8b75ddcfe763db44f6f13ebc66c28469d154acd/src/ActionServer_FJT.c#L564-L568) are both static functions defined within `ActionServer_FJT.c`, but they are being referenced in [`Tests_ActionServer_FJT.c`](https://github.com/Yaskawa-Global/motoros2/blob/d8b75ddcfe763db44f6f13ebc66c28469d154acd/src/Tests_ActionServer_FJT.c).

gavanderhoorn, 2024-06-27:
Right. This is likely a result of the fact *all* `.c`s are included automatically in the `mpBuilder` build in VS.

`Tests_ActionServer_FJT.c` is not intended to be compiled stand-alone. It gets `#include`d at the end of `ActionServer_FJT.c` exactly because it needs access to those two `static` functions:

https://github.com/Yaskawa-Global/motoros2/blob/d8b75ddcfe763db44f6f13ebc66c28469d154acd/src/ActionServer_FJT.c#L866-L867

not too nice (`#include`-ing a `.c`), but there isn't really any other way I believe.

I'll see if I can guard against this problem by adding some preprocessor directives.

ted-miller, 2024-06-27:
> does mpBuilder honour VS project settings which exclude a file from the build?

Nope. 

What's the motivation to make those functions static?

gavanderhoorn, 2024-06-27:
> > does mpBuilder honour VS project settings which exclude a file from the build?
> 
> Nope.

Unfortunate, as that would've been an easy way to solve this.

> What's the motivation to make those functions static?

I've used it here to mark them as *private* functions: they are implementation details, and only exist to make the function that calls them easier to read.

PR #273 (2024-07-01): Prevent building FJT tests stand-alone
As per title.

Should fix #270 (well .. work around it).

See the commit message for additional comments.