Yaskawa-Global/motoros2 #305: Need to clearly indicate failure when configuring speed FB

ted-miller, 2024-09-09:
We had a user get into [this scenario](https://github.com/Yaskawa-Global/motoros2/blob/main/doc/troubleshooting.md#alarm-49974). But he reports that he never saw the alarm, `8001 Speed FB enabled, reboot now. [10]`

https://github.com/Yaskawa-Global/motoros2/discussions/294#discussioncomment-10595168

This results in MR2 appearing to freeze at startup and communication dropping to the agent.

The only way that I know it occurred was by looking through the alarm history.

I too believe that I have seen this scenario, but I just brushed it off and assumed that I wasn't paying attention. 

We need to try to reproduce with a "clean" robot controller that has FSU enabled.

ted-miller, 2024-09-09:
And assuming that we can reproduce, add some debug messaging to indicate what's wrong.

gavanderhoorn, 2024-09-10:
Isn't that all part of `GP_getFeedbackSpeedMRegisterAddresses(..)` (called [here](https://github.com/Yaskawa-Global/motoros2/blob/main/src/CtrlGroup.c#L135-L142))?

ted-miller, 2024-09-10:
Yes. 
If `S1CxG0335` is not set, then `GP_getFeedbackSpeedMRegisterAddresses` will attempt to directly modify the `SC.PRM` file in the controller. After that modification, it raises an alarm and sits in an infinite (blocking) loop to force the user to reboot.

But...
a) The user didn't see the alarm. Don't know why...
b) Since the FSU CRC-check was enabled, the modification to SC.PRM was never accepted. So MR2 gets stuck in the bootup routine.

If the pendant isn't properly displaying the alarm, then I don't know that we'll be able to fix that directly. (Will need to report to Japan.) But we can ensure that the debug message gets reported in a way that is visible to the user. 

Alternatively (or additionally), we can modify the setup instructions to have users manually enable the speed-feedback parameters.