Yaskawa-Global/motoros2 #342: Add ROS Answers link back once the archive is online

jimmy-mcelwain, 2024-12-12:
ROS Answers is offline, but we previously had a link with valuable discussion there. An archive of ROS Answers will be available in the future. Once it is available, simply add the new link to the README again.  

Old link was [here](https://github.com/Yaskawa-Global/motoros2/blob/9f5ccf6dcf86cb3be340c850cb3aa990aa9c4bd6/README.md?plain=1#L686)

>Please create an issue to track putting this link back once the [ROS Answers archive](https://github.com/ros-infrastructure/answers-archive) is online. The MoveIt issue does not capture the full discussion.

_Originally posted by @gavanderhoorn in https://github.com/Yaskawa-Global/motoros2/pull/337#discussion_r1881746228_

jimmy-mcelwain, 2025-01-03:
The archive is seemingly online already. We haven't actually merged #337, so we can just add the link back before we merge.

ted-miller, 2025-01-07:
The PR hasn't been merged yet. I'll link this issue to that PR so that this closes automatically.

jimmy-mcelwain, 2025-01-07:
>The PR hasn't been merged yet. I'll link this issue to that PR so that this closes automatically.

True but the removal of the link was never actually merged into main, it was only ever on the jazzy branch. This issue was created sort of prematurely, I guess since I assumed that the ROS Answers link would be dead for a long time.

ted-miller, 2025-01-07:
Oh. I understand now.

closed 2025-01-07 by ted-miller