Yaskawa-Global/motoros2 #361: 'start_point_queue_mode' fails if the robot has moved since being powered off

ted-miller, 2025-01-27:
This is basically the same problem as #37.

It can also happen if restoring an old CMOS backup. 

When you attempt to `start_point_queue_mode`, the pendant raises alarm `OUT OF RANGE (DROP-VALUE)`. But the service call just returns `The controller has an active Error: Can't turn on servo because: 'Servo power cannot be turned on' (0x3450)`. So, the message isn't necessarily _wrong_, it's just not helpful. 

In my case, I was able to just reset the alarm and try again. But usually, this alarm requires you to verify the home-position before you can proceed.

I'm not yet sure how to detect this scenario so that we can provide a better error code.

gavanderhoorn, 2025-02-06:
#182? Does `robot_status::error_codes` not list all active errors?

ted-miller, 2025-02-07:
> https://github.com/Yaskawa-Global/motoros2/pull/182? Does robot_status::error_codes not list all active errors?

This particular problem is not that 'type' of error. 

A traditional pendant `error` is caused by incorrect user behavior on the pendant. (Like hitting PLAY when servos are off.) This particular problem is more from the robot being in a bad programmatic state.