Yaskawa-Global/motoros2 #376: Port numbers with 5+ digits get truncated

jimmy-mcelwain, 2025-02-26:
See https://github.com/Yaskawa-Global/motoros2/discussions/374#discussioncomment-12332137

Port numbers with 5 or more digits get truncated when trying to connect. 

If I have a valid port in the config file 16510, I would have to connect to 1651 on the docker side.  If I specify port 88888 in the config file (I am aware that this is not a valid port), then I can still connect properly if I specify port 8888 when I start the docker container. 

This is a problem if a user puts a valid 5 digit port number in the config file. 

That is caused by this line upstream [here](https://github.com/micro-ROS/rmw_microxrcedds/blob/554445fe739e4c5bcb27753ffbe21e03bb2c3c1f/rmw_microxrcedds_c/src/rmw_microros/init_options.c#L101). `MAX_PORT_LEN` is set to 5, but in reality `snprintf` is truncating at 4 so it has space to put the null character. I haven't looked anywhere yet to see if this is an intentional design choice, but I doubt that it is. I can look more into it soon.

jimmy-mcelwain, 2025-02-27:
Opened an issue https://github.com/micro-ROS/rmw_microxrcedds/issues/314. 

For the time being should we validate the port manually? Even if it gets fixed upstream it might be a long time before it makes its way here.

Less important/maybe we shouldn't do this, but we could also set an alarm if the port is < 1024 since those are the well known ports and usually aren't recommended for developers.

jimmy-mcelwain, 2025-03-03:
A fix was merged upstream https://github.com/micro-ROS/rmw_microxrcedds/pull/315. It might be a while before it makes its way into a release, though. Not sure what the best option for us is, for now.

ted-miller, 2025-03-03:
I think we should just wait for the next release/tag on their end. It looks like it'll happen in May/June. (https://github.com/micro-ROS/rmw_microxrcedds/tags) The easy fix in the meantime is to use the suggested port that we provide in the yaml file.

gavanderhoorn, 2025-03-05:
It would not be much work bumping `rmw_microxrcedds` to include the fix and use it as part of `libmicroros`. We could also wait for a release upstream and then bump on our side.

We'd need another release of MotoROS2 though to get that fix included.

jimmy-mcelwain, 2025-07-18:
I agree, this should be closed

closed 2025-07-18 by jimmy-mcelwain