Yaskawa-Global/motoros2 #377: experiencing continuous lags without a telnet debugging output

legend-chen, 2025-02-27:
Dear Yaskawa Engineers
I’m facing an issue with Motoros2: everything works as expected when using Telnet debugger, but when I stop using it and run MoveIt independently with the default RViz control tool, I experience a series of stutters. Do you have any suggestions on what could be causing this or how to fix it?
Thanks for your help!

ted-miller, 2025-02-27:
Hello @legend-chen 

Can you clarify whether you're using the [Debug Logging client](https://github.com/Yaskawa-Global/motoros2/blob/main/doc/troubleshooting.md#debug-log-client) or using the [log_to_stdout](https://github.com/Yaskawa-Global/motoros2/blob/34bfdaf4c16a1381ffd4e0d03af5d05001f1eea4/config/motoros2_config.yaml#L187-L197) feature? If using the `log_to_stdout` feature, this will impact performance of the application. It would slow down execution.

I do not have any specific explanation of what you're experiencing. Though perhaps it is something similar to this report: https://github.com/Yaskawa-Global/motoros2/issues/43#issuecomment-2225660625

gavanderhoorn, 2025-03-01:
I'm not sure why I can't convert this into a Discussion. Github seems to try, then fails and re-opens this one.